isaac-sim/IsaacLab #988: [Bug Report] pip's dependency conflicts when using "./isaaclab.sh -i" in conda environment

gcMs1, 2024-09-16:
Hello, when i installed the IsaacLab following the guide [here](https://isaac-sim.github.io/IsaacLab/source/setup/installation/binaries_installation.html), I got a pip's dependency conflicts when I using "./isaaclab.sh -i" in conda environment.
[image]

I fixed the first and second conflict by using the command "pip install usd-core==23.11". When i tried solving the third one by using the command "pip install botocore==1.19.52", I got another conflict.
[image]

How can i solve this conflict? Could you give me some suggestion? 
Thank you!

Mayankm96, 2024-09-16:
This can be safely ignored. We are working with the Isaac Sim team to ensure these errors disappear in future releases.

gcMs1, 2024-09-17:
> This can be safely ignored. We are working with the Isaac Sim team to ensure these errors disappear in future releases.

Hello, I reinstalled IsaacSim of version 4.1.0, everything work well (I installed version 2023.1.1). What i need to do is to install the latest version

AntoineRichard, 2026-08-27:
Thanks @gcMs1 for writing this up with the full pip output.

Those conflicts (`usd-core`, `botocore`) came from the pinning inside the Isaac Sim wheels and were benign — you confirmed the install worked after moving from Isaac Sim 2023.1.1 to 4.1.0. Installation has since been reworked around the uv-based flow in `docs/source/setup/installation/`, so those pins are long gone. Closing as resolved; please reopen or open a new issue if you hit real dependency errors on a current release.

---
*🤖 This issue was reviewed as part of an AI-assisted triage of the Isaac Lab backlog, with automated verification against the current codebase. If we got this wrong, please reopen — we'd rather hear about it!*

closed 2026-08-27 by AntoineRichard