isaac-sim/IsaacLab #993: [Question] Does executing `robot.root_physx_view.get_jacobians()` make a forward simulation step?

ozhanozen, 2024-09-16:
Hi Isaac Lab Team,

I wanted to ask what happens on the simulation side when I execute `robot.root_physx_view.get_jacobians()` (or similar functions), especially if this steps the environment forward implicitly, i.e., `sim.step()`.

I am asking this because of an observation I recently had:

There is [this](https://github.com/isaac-sim/IsaacLab/blob/main/docs/source/refs/issues.rst) limitation of PhysX such that when we reset an environment/robot (e.g., with  `robot.reset()`), the immediate state values of the robot kinematic chain (e.g., ee position) are not yet updated. Until an explicit call to `sim.step()` is made, the state values stay as the last values before the `robot.reset()`. Consistent with this limitation, the output of the two print statements below is the same:

```
print(robot.root_physx_view.get_link_transforms())
robot.write_joint_state_to_sim(default_joint_pos, default_joint_vel) # fixed default_joint_pos and default_joint_vel values
robot.set_joint_effort_target(zero_joint_efforts) 
robot.write_data_to_sim()
robot.reset()
robot.update(sim_dt)
print(robot.root_physx_view.get_link_transforms())
```
What is confusing for me is, that if I insert a line that queries for Jacobian as the following (or pretty much anything that requires a forward kinematic chain calculation):
```
print(robot.root_physx_view.get_link_transforms())
robot.write_joint_state_to_sim(default_joint_pos, default_joint_vel) # fixed default_joint_pos and default_joint_vel values
robot.set_joint_effort_target(zero_joint_efforts) 
robot.write_data_to_sim()
robot.reset()
robot.update(sim_dt)
jacobians = robot.root_physx_view.get_jacobians()
print(robot.root_physx_view.get_link_transforms())
```
then the two print statements provide different values, which makes me believe that a forward pass to the kinematic chain has been made, without me explcility calling `sim.step()`. Is this what is happening? Or is it now possible to call a forward pass without calling sim update?

Thanks in advance!

Mayankm96, 2024-09-17:
Hi @ozhanozen ,

Yes, the `get_jacobians` API calls PhysX's `updateArticulationKinematics` implicitly under the hood to take into account changes in joint positions. This is why there is a different behavior depending on whether or not they have get_jacobians call in there.

In the coming release, we will expose this  `update_articulations_kinematic` from the tensor API more explicitly, which can be used for an explicit FK update before the step as needed.

ozhanozen, 2024-09-17:
Hi @Mayankm96, 

Thanks a lot for the answer. It makes more sense now.

closed 2024-09-17 by ozhanozen