isaac-sim/IsaacLab #996: [Question] Rendered as if the particles stopped in their initial state

NAMHAUK, 2024-09-18:
### Question
Hello, I've experienced a rendering issue in using Isaac Sim, and I'd like to ask you a question about this solution.

### Describe the issue
When Isaac Lab spawn Isaac Sim file (.usd), the object made with particle sampler does not move in the initial state. Other objects are properly simulated.

When i look at the simulation, it looks like the particles have stopped in their initial state. However, looking at the movement of rigid objects pushed out by particles, the particles are simulated, but only the rendering seems to be in the first state.

The screenshot below is a picture taken after a while after the simulation(not first scene).  It looks like the particles are not moving in the location they were first created. 
[image]

I compared the settings between the Isaac Sim running with the omniverse Launcher and the  Isaac Sim running with the Isaac Lab, but I couldn't find any difference.

I saw a similar problem in writing (https://github.com/isaac-sim/IsaacLab/discussions/509), but the solution to this writing was about Orbit, so I couldn't apply it to Isaac Lab. I wonder if there is other way to solve this issues.

### Steps to reproduce
I tried in the order below, and an issue occurred.

1. Make particle sampler in Isaac Sim
I created particle obect in the order below (Works well with isaac sim without the above error)
  - Created mesh in Isaac Sim running with Omniverse Launcher.
  - right click object created -> add -> physic -> particle sampler
  - also creates other rigid objects that are affected by particles
  - I saved this scene in titled "particles" (.usd)

2. Make Isaac Lab Code to import Isaac Sim objects created
I created Isaac Lab code in titled "test_usd" in path ("IsaacLab/source/standalone/tutorials/test")
This code is a little bit of a change from "spawn_prims.py" that is provided as a tutorial in the isaac lab.
```python
import argparse

from omni.isaac.lab.app import AppLauncher

# create argparser
parser = argparse.ArgumentParser(description="Tutorial on spawning prims into the scene.")
# append AppLauncher cli args
AppLauncher.add_app_launcher_args(parser)
# parse the arguments
args_cli = parser.parse_args()
# launch omniverse app
app_launcher = AppLauncher(args_cli)
simulation_app = app_launcher.app

"""Rest everything follows."""

import omni.isaac.lab.sim as sim_utils

def design_scene():
    """Designs the scene by spawning ground plane, light, objects and meshes from usd files."""
    # Ground-plane
    cfg_ground = sim_utils.GroundPlaneCfg()
    cfg_ground.func("/World/defaultGroundPlane", cfg_ground)

    # spawn distant light
    cfg_light_distant = sim_utils.DistantLightCfg(
        intensity=3000.0,
        color=(0.75, 0.75, 0.75),
    )
    cfg_light_distant.func("/World/lightDistant", cfg_light_distant, translation=(1, 0, 10))

    # spawn a usd file of a table into the scene
    cfg = sim_utils.UsdFileCfg(usd_path=f"omniverse://localhost/Projects/particles.usd")
    cfg.func("/Xform/pool", cfg, translation=(0.0, 0.0, 1.05))

def main():
    """Main function."""
    # Initialize the simulation context
    sim_cfg = sim_utils.SimulationCfg(dt=0.01)
    sim = sim_utils.SimulationContext(sim_cfg)
    
    # Set main camera
    sim.set_camera_view([5.0, 0.0, 5.5], [-0.5, 0.0, 0.5])

    # Design scene by adding assets to it
    design_scene()

    # Play the simulator
    sim.reset()
    # Now we are ready!
    print("[INFO]: Setup complete...")

    # Simulate physics
    while simulation_app.is_running():
        # perform step
        sim.step()

if __name__ == "__main__":
    # run the main function
    main()
    # close sim app
    simulation_app.close()
```

4. run file
```bash
./isaaclab.sh -p source/standalone/tutorials/test/test_usd.py
```

### System Info
- Isaac Sim Version: [4.1]
- OS: [Ubuntu 22.04]
- GPU: [RTX 4070 Super]
- CUDA: [12.1]

Thnaks for your time.

robegi, 2024-09-19:
Hi,

I am the guy who had the issue in [#509](https://github.com/isaac-sim/IsaacLab/discussions/509). Now I am using Isaac Lab and the same solution works. I solved it by using the cpu pipeline and disabling the fabric extension, which in the standard scripts (such as zero_agent.py) is done by calling the --cpu and --disable_fabric in the command line. In your custom script I think it might be sufficient to set them inside your code, the command you could modify might be:

```
sim_cfg = sim_utils.SimulationCfg(dt=0.01, use_fabric=False, device='cpu')
```
Have you already tried something similar?

By the way, I found that spawning the particles directly in the script usually works better than importing  a .usd file, as it allows better control over the simulation and avoids annoying problems when trying to modify the spawning position. But that is up to you.

Hope that this helps

NAMHAUK, 2024-09-20:
Thank you very very much! 
I solved the problem in this way. This is a problem I've found for a long time, thank you very much.

By the way, among the things you mentioned, how is it possible to spawn particles directly into the script instead of using .usd file? When I looked up the Isaac Lab document, I couldn't find a way to spawn particles. That's why I used the method of importing from the .usd file, so please share the method if possible.

Thank you for your kindness!

robegi, 2024-10-01:
Hi @NAMHAUK ,

an easy way to do it is to use the sample code provided inside the [physics demos](https://docs.omniverse.nvidia.com/extensions/latest/ext_physics/physics-particles.html) inside Isaac Lab. The isosurface one is the one I am currently using. 
How to integrate the code in the demos inside yours really depends on what you are planning to do and what kind of workflow you are using. An example was in the issue [#250](https://github.com/isaac-sim/IsaacLab/issues/250) in which the code sample to create cloth was integrated inside the code of this user.

Hope this helps

NAMHAUK, 2024-10-02:
Thank you for your answer.
I think it's a convenient way to do the spawning the particles directly in the script, but I thought the overhead was big for me to learn how to do it, so I decided to use it as a usd file for now.

However, I really appreciate your kind reply.

closed 2024-10-02 by NAMHAUK