isaac-sim/IsaacLab #1002: [Proposal] Relaxing gymnasium requirement to 0.29.1

BurakDmb, 2024-09-19:
### Proposal

Hi, for some other libraries, I am forced to use Gymnasium version 0.29.1.
However, Isaaclab has a 0.29.0 strict requirement.  
https://github.com/isaac-sim/IsaacLab/blob/52af899614318a25404677bd7b754d867e743c45/source/extensions/omni.isaac.lab/setup.py#L29 and https://github.com/isaac-sim/IsaacLab/blob/52af899614318a25404677bd7b754d867e743c45/source/extensions/omni.isaac.lab/config/extension.toml#L26

Can you bump Gymnasium to 0.29.1 or at least relax the requirement to cover both 0.29.0 and 0.29.1

0.29.1 is a minor release and it only adds some small warnings, so I don't think anything will break in IsaacLab.
https://github.com/Farama-Foundation/Gymnasium/releases/tag/v0.29.1

I can open a PR if help is needed.

Mayankm96, 2024-09-19:
The last time I checked, the video recorder in 0.29.1 was broken. So we reverted to keeping it at 0.29.0. We can relax it but then it becomes hard to ensure that things aren't breaking because other libraries made breaking changes.

BurakDmb, 2024-09-19:
I suggest a version specifier like this could cover your concerns and flex the version by a little bit. 
I tested in my local setup for pip installation and setup py seems to be working.
Source for specifying range in install_requires: https://packaging.python.org/en/latest/discussions/install-requires-vs-requirements/#install-requires

`"gymnasium >=0.29.0, <=0.29.1"`

BurakDmb, 2024-09-23:
Hi, I can create a PR for that, or is someone working on this now? 
Do you happen to see any problems with this version suggestion? @Mayankm96 

My suggestion is changing these lines with `"gymnasium >=0.29.0, <=0.29.1"` to support gymnasium versions 0.29.0 and 0.29.1:

https://github.com/isaac-sim/IsaacLab/blob/52af899614318a25404677bd7b754d867e743c45/source/extensions/omni.isaac.lab/setup.py#L29

https://github.com/isaac-sim/IsaacLab/blob/52af899614318a25404677bd7b754d867e743c45/source/extensions/omni.isaac.lab/config/extension.toml#L26

Mayankm96, 2024-10-09:
Gymnasium 1.0.0 is out. We would work on switching to it.

@Toni-SM for vis.

cyhunblr, 2024-11-27:
Are there any updates regarding this issue?

commit 64dc8745 (2024-12-02): Adds gym.Env to DirectMARLEnv for Gymnasium v1.0.0 (#1446)

# Description

Due to recent release of Gymnasium 1.0.0, this now becomes the Gymnasium
version that gets installed when running the installation process for
Isaac Lab. Gymnasium 1.0.0 requires all environments to be a subclass of
`gym.Env` when using the `make` interface. This change makes the
`DirectMARLEnv` a subclass of `gym.Env` to satisfy this requirement.

Fixes #1002 

## Type of change

- Bug fix (non-breaking change which fixes an issue)
- Breaking change (fix or feature that would cause existing
functionality to not work as [... cut]
closed 2024-12-02 by kellyguo11