isaac-sim/IsaacLab #1004: [Question] Support for AMP ?

MankaranSingh, 2024-09-20:
### Question

Hi, any timeline on when AMP environment would be ported to isaac Lab ?

kellyguo11, 2024-09-27:
Hello, AMP support is on our roadmap and will hopefully be available in the coming months.

closed 2024-09-27 by kellyguo11

OliEfr, 2025-01-27:
Hi, I have a working AMP implementation for IsaacLab. I would be willing to open a PR. Is the IsaacLab-Team interested?

kellyguo11, 2025-01-27:
Hi there, great to hear you have AMP implementation for Isaac Lab! We are planning to put out a few AMP environments in the next few days for SKRL. Would be interesting to see if we can integrate these implementations together to have more RL library support.

OliEfr, 2025-02-02:
@kellyguo11 I checked your code and saw you implemented it as a direct RL env. I guess because integration in manager based envs might be a bit messy. I, however, implemented in a [manager based env](https://github.com/OliEfr/IsaacLab/blob/5671a8a551cf8d4c8b2dfe751825952aeb26c2f5/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/manager_based_rl_env.py#L79). Also, I implemented AMP in rsl_rl directly and for a quadruped. Let me know if any of that is interesting.

Maybe the [reference state initialization](https://github.com/OliEfr/IsaacLab/blob/5671a8a551cf8d4c8b2dfe751825952aeb26c2f5/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/mdp/events.py#L36) I can be re-used? (I didnt check how you implemented it though)

kellyguo11, 2025-02-03:
@OliEfr absolutely, would love to have your contribution. Feel free to start a PR and we can discuss how best to integrate the implementations.

KingBright, 2025-08-12:
Hi there, is there a manager based amp support on the way?