isaac-sim/IsaacLab #1031: [Bug Report] Increasing the number of parallel environments results in severe degradation in TiledCamera rendering quality

trevormcinroe, 2024-09-24:
I am using the latest version (4.2.0) on a Ubuntu server while rendering headless.

Here is the quality of images rendered from one camera while only using a single environment:
[image]

Here is the quality of images rendered from one camera while using 32 parallel environments:
[image]

glvov-bdai, 2024-09-24:
Hi can you please share the script /conditions to recreate these results?

glvov-bdai, 2024-09-24:
In my preliminary tests I did not see any degradation with 32 or 64 envs in parallel. However, I only tried low resolution and static frames. If you provide system information and a way for us to recreate your experiment I'd be happy to look into this further!

32 envs

[image]

64 envs
[image]

I tried these with the manager based cartpole envs 

https://github.com/isaac-sim/IsaacLab/pull/995

glvov-bdai, 2024-09-24:
I did notice that the grid lines do seem to shift step to step 
[image]

trevormcinroe, 2024-09-24:
@glvov-bdai I can provide some minimal code tomorrow, but try with a more visually-complex model like Franka, and place the camera closer to the object. Perhaps that will show the issue?

StoneT2000, 2024-09-25:
I noticed that some environments (i guess the ones that spawn too far away?) dont show the floor anymore. My understanding is Isaac Lab essentially has one shared scene meaning objects can fly into the parallel views of others if they move there and technically there is only one floor. Is there a way to make the floor bigger so that all parallel cameras have a floor shown?

glvov-bdai, 2024-09-25:
@stonet2000 We are aware of the default ground plane not being visible beyond a certain contained size.
This is due to the ground plane being a finite USD that is not scaled. If you'd like, you could add more ground planes as assets, and just spawn them further away. Alternatively, you could use a larger USD asset for the ground plane, just having it span the desired sizes.

Although the default ground plane may not be visible beyond certain sizes, it should still present for collisions.

In terms of rendering performance, you can try my benchmark tool https://github.com/isaac-sim/IsaacLab/pull/976 with the different configurations I suggested and find empirically what works best.

If you run into any problems with the ground plane stuff I'd recommend opening an issue specifically for that for better visibility - doesn't seem like it's related to the rendering issues

kellyguo11, 2024-09-27:
The rendering quality difference could be coming from the denoiser in the renderer. One thing you could try is adding this to the code to change the denoiser mode to see if it'll help:
```
import omni.replicator.core as rep
rep.settings.set_render_rtx_realtime(antialiasing="DLAA")
```

trevormcinroe, 2024-10-01:
@rthaker01 @glvov-bdai apologies for the delay. ICLR deadline and all... I've been working with @elle-miller on some projects using IsaacLab. 

We've both been opening [BUG] reports, and we'll get back to you soon on this one with more information.

ArneKlages4444, 2024-10-08:
Hi, I tested the denoiser fix in a visually more complex environment. It seems that the denoiser helps a lot but does not solve the problem completely, see the following images (112x112 pixel).  Especially when it comes to lighting, the degeneration effect is still very visible.

2 envs:
[image]
1024 envs with denoiser:
[image]
1024 envs:
[image]

2 envs:
[image]
1024 envs with denoiser:
[image]
1024 envs:
[image]

System:
- IsaacLab version (last commit): https://github.com/isaac-sim/IsaacLab/discussions/132
- Isaac Sim Version: 4.2.0
- GPU: L40S (48 GiB)
- CUDA Version: 12.2
- GPU Driver Version: 535.161.08

elle-miller, 2024-10-09:
@glvov-bdai OK, while the denoiser helps, the fix still does not make images in a good state for training (84px images, 1024 envs). The strange trailing effect of the cart still persists. Has anyone else experienced this?

**Without denoiser:**
[image]

**With denoiser:**
[image]

glvov-bdai, 2024-10-09:
Hi @elle-miller @ArneKlages4444 @trevormcinroe 

Thank you for investigating this issue and providing replications steps.

@kellyguo11 @mpgussert still looks like the issue is persisting on isaac sim side?

Can we please add this to functionality to fix in the next isaac release?

SantiDiazC, 2025-01-06:
Hi, 
I have experienced a similar issue when modifying the pose of the camera to render two different views. 
In this case the image has some degradation when I do that, and it get worse when the number of environments are increased. Also, I changed the denoiser as @kellyguo11 suggested.  Is there any solution for this? Thank you!

I am using the latest isaaclab 1.4.0 and Isaacsim 4.2.

ArneKlages4444, 2025-01-31:
It seems that with Isaac Sim 4.5.0 and Isaac Lab 2.0.0, this problem is partially solved. I can't notice any image degradation when using a high number of environments. 

DLAA 10 envs:
[image]

DLAA 1020 envs:
[image]

DLAA 4095 envs:
[image]

Default config 10 envs:
[image]

Default config 1020 envs:
[image]

The issue with the ghosting (trailing effect) mentioned by @elle-miller still remains. I think this is a problem of DLSS and DLAA, because they rely on previous frames. If you turn antialiasing off or use FXAA (that does not use previous frames), the ghosting is gone. But the problem with this approach is that the image quality suffers. I still prefer using DLAA with low image resolutions.

Off
[image]

FXAA
[image]

kellyguo11, 2025-02-04:
Thanks for confirming! We are working with our rendering team on ways to reduce the ghosting effect, it is indeed likely due to DLSS.