isaac-sim/IsaacLab #1035: [Question] How to update Isaac Sim in a more efficient way

HzfFrank, 2024-09-25:
### Question

Hi, since my collaborator is using a newer version of Isaac Sim (4.1.0), I often encounter version mismatch issues when running Isaac Lab, so I need to update my Isaac Sim version. However, Isaac Lab is linked to Isaac Sim, and our project depends on but is independent of the Isaac Lab folder. As a result, every time I update Isaac Sim, I need to re-download Isaac Lab and go through the installation process to create the environment, etc. After that, I still need to reinstall my project to ensure it connects to Isaac Lab. As the developers suggested, we should frequently use the latest version of Isaac Sim. I just noticed that version 4.2.0 of Isaac Sim is also available now, which means both my collaborator and I would need to go through this tedious update process again. Therefore, I would like to ask if there is an easier way to update Isaac Sim, so that Isaac Lab and our project connected to it can be updated quickly.

WuDuidi, 2024-09-25:
Creating the Isaac Sim Symbolic Link
```
# enter the cloned repository
cd IsaacLab
# create a symbolic link
ln -s path_to_isaac_sim _isaac_sim
# For example: ln -s /home/nvidia/.local/share/ov/pkg/isaac-sim-4.1.0 _isaac_sim
```

Mayankm96, 2024-09-25:
@Toni-SM, Will the pip upgrade help automate this process?

Toni-SM, 2024-09-26:
Yes, in principle the installation as a Python package using a Python package manager, such as `pip`, should help automating the process