isaac-sim/IsaacLab #1041: [Proposal] "isaacsim/exts/omni.isaac.core/omni/isaac/core/utils/math.py" is overriding the stdlib module "math"

WYYAHYT, 2024-09-26:
### Proposal

There is a confilct between `isaacsim/exts/omni.isaac.core/omni/isaac/core/utils/math.py` and `stdlib math`

### Motivation
While using VScode, Pylince extension catches this conflict and produces a warning:
`"isaacsim/exts/omni.isaac.core/omni/isaac/core/utils/math.py" is overriding the stdlib module "math"`

### Alternatives
Rename `"isaacsim/exts/omni.isaac.core/omni/isaac/core/utils/math.py"` to `"isaacsim/exts/omni.isaac.core/omni/isaac/core/utils/math_{xx}.py"` or any ohter names

WYYAHYT, 2024-09-26:
same proposal for `"isaacsim/exts/omni.isaac.core/omni/isaac/core/utils/random.py"`

RandomOakForest, 2025-01-05:
Thank you for posting this. We will review.

AntoineRichard, 2026-08-27:
Thanks for flagging this, @WYYAHYT. The `math.py` in question ships with Isaac Sim's `omni.isaac.core` extension rather than with Isaac Lab, so the Isaac Sim tracker is the better home for it. That extension has since been superseded by `isaacsim.core`, which is what Isaac Lab builds against today, so the old `omni.isaac.core` is no longer in the picture on our side.

Closing here on that basis. If Pylance still flags this on a current Isaac Sim release it's worth raising there, and please reopen if we missed an Isaac Lab-side angle.

---
*🤖 This issue was reviewed as part of an AI-assisted triage of the Isaac Lab backlog, with automated verification against the current codebase. If we got this wrong, please reopen — we'd rather hear about it!*

closed 2026-08-27 by AntoineRichard