isaac-sim/IsaacLab #1050: [Question] Debugging in VS Code

robin-vetsch, 2024-09-27:
### Question

I have successfully installed Isaac Sim 4.2.0 and Isaac Lab and I can successfully run the following command.

```
python source/standalone/tutorials/00_sim/create_empty.py
```

But I want to debug the Python script in VSCode (incl. breakpoints, etc.) The debugging of a standalone example in Isaac Sim 4.2.0 works, the debugging of an Isaac Lab example unfortunately does not.

The .launch file looks like this:

```
// This file is a template and is automatically generated by the setup_vscode.py script.
// Do not edit this file directly.
// 
// Generated from: /home/robin/IsaacLab/.vscode/tools/launch.template.json
{
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [
        {
            "name": "Python: Current File",
            "type": "python",
            "request": "launch",
            "program": "${file}",
            "console": "integratedTerminal"
        },
        {
            "name": "Python: Attach (windows-x86_64/linux-x86_64)",
            "type": "python",
            "request": "attach",
            "port": 3000,
            "host": "localhost"
        },
        {
            "name": "Python: Train Environment",
            "type": "python",
            "request": "launch",
            "args" : ["--task", "Isaac-Reach-Franka-v0", "--headless"],
            "program": "${workspaceFolder}/source/standalone/workflows/rsl_rl/train.py",
            "console": "integratedTerminal"
        },
        {
            "name": "Python: Play Environment",
            "type": "python",
            "request": "launch",
            "args" : ["--task", "Isaac-Reach-Franka-v0", "--num_envs", "32"],
            "program": "${workspaceFolder}/source/standalone/workflows/rsl_rl/play.py",
            "console": "integratedTerminal"
        }
    ]
}

```

As soon as I start the script in VScode in debug mode I get the following error:

```
/usr/bin/env /home/robin/anaconda3/envs/IsaacLab/bin/python /home/robin/.vscode/extensions/ms-python.debugpy-2024.10.0-linux-x64/bundled/libs/debugpy/adapter/../../debugpy/launcher 36497 -- /home/robin/IsaacLab/source/standalone/tutorials/00_sim/create_empty.py 
bash: /usr/bin/env: Argument list too long
```

Do I have to make further configurations to perform debugging?

Many thanks for help!

Dhoeller19, 2024-10-08:
Hi @robin-vetsch,
This is very strange. For me it works properly, and the terminal prints a similar instruction (without failing though).
It seems like something is wrong with the vsode setup. I would look into the following:
1. Configure Isaac lab vscode with `./isaaclab.sh -v`
2. Make sure the python interpreter is correctly set to the `isaaclab` conda env. In vscode, Ctrl+Shift+P -> Python:Select Interpreter -> The conda env 
3. Reset all the vscode settings (I think this is probably the most likely to succeed)

robin-vetsch, 2024-10-08:
Hi @Dhoeller19 

I have completely reinstalled VSCode and also Isaac Sim with the Isaac Lab - unfortunately without success.

The points you suggested did not help either. I created a conda env with the name “IsaacLabEnv” during the installation of IsaacLab. In VSCode this env is defined as Interpreter Path. see image:

[image]

There is also no form of .python.env file:

[image]

Below you can find the settings.json file, where also the interpreter path is defined.

```
// This file is a template and is automatically generated by the setup_vscode.py script.
// Do not edit this file directly.
// 
// Generated from: /home/robin/IsaacLab/.vscode/tools/settings.template.json
{
    "files.associations": {
        "*.tpp": "cpp",
        "*.kit": "toml",
        "*.rst": "restructuredtext"
    },
    "editor.rulers": [120],

    // files to be ignored by the linter
    "files.watcherExclude": {
        "**/.git/objects/**": true,
        "**/.git/subtree-cache/**": true,
        "**/node_modules/**": true,
        "**/_isaac_sim/**": true,
        "**/_compiler/**": true
    },
    // Configuration for spelling checker
    "spellright.language": [
        "en-US-10-1."
    ],
    "spellright.documentTypes": [
        "markdown",
        "latex",
        "plaintext",
        "cpp",
        "asciidoc",
        "python",
        "restructuredtext"
    ],
    "cSpell.words": [
        "literalinclude",
        "linenos",
        "instanceable",
        "isaacSim",
        "jacobians",
        "pointcloud",
        "ridgeback",
        "rllib",
        "robomimic",
        "teleoperation",
        "xform",
        "numpy",
        "tensordict",
        "flatcache",
        "physx",
        "dpad",
        "gamepad",
        "linspace",
        "upsampled",
        "downsampled",
        "arange",
[... 601 lines cut ...]
        "${workspaceFolder}/_isaac_sim/kit/plugins/bindings-python",
        "${workspaceFolder}/_isaac_sim/kit/python/lib/python3.10/site-packages",
        "${workspaceFolder}/_isaac_sim/python_packages",
        "${workspaceFolder}/source/extensions/omni.isaac.lab_assets",
        "${workspaceFolder}/source/extensions/omni.isaac.lab_tasks",
        "${workspaceFolder}/source/extensions/omni.isaac.lab"
    ],
    "python.autoComplete.extraPaths": [
        "/opt/ros/noetic/lib/python3/dist-packages",
        "/home/robin/IsaacLab/_isaac_sim/../../..//opt/ros/noetic/lib/python3/dist-packages",
        "/home/robin/IsaacLab/_isaac_sim/kit/python/lib/python3.10/site-packages",
        "/home/robin/IsaacLab/_isaac_sim/python_packages",
        "/home/robin/IsaacLab/_isaac_sim/exts/omni.isaac.kit",
        "/home/robin/IsaacLab/_isaac_sim/kit/kernel/py",
        "/home/robin/IsaacLab/_isaac_sim/kit/plugins/bindings-python",
        "/home/robin/IsaacLab/_isaac_sim/exts/omni.isaac.lula/pip_prebundle",
        "/home/robin/IsaacLab/_isaac_sim/exts/omni.exporter.urdf/pip_prebundle",
        "/home/robin/IsaacLab/_isaac_sim/extscache/omni.kit.pip_archive-0.0.0+10a4b5c0.lx64.cp310/pip_prebundle",
        "/home/robin/IsaacLab/_isaac_sim/exts/omni.isaac.core_archive/pip_prebundle",
        "/home/robin/IsaacLab/_isaac_sim/exts/omni.isaac.ml_archive/pip_prebundle",
        "/home/robin/IsaacLab/_isaac_sim/exts/omni.pip.compute/pip_prebundle",
        "/home/robin/IsaacLab/_isaac_sim/exts/omni.pip.cloud/pip_prebundle"
    ]
}
```

robin-vetsch, 2024-10-09:
@Dhoeller19 

I have commented out some env paths regarding ROS in the bashrc file. Now the debugging works.

closed 2024-10-09 by robin-vetsch

LeZheng-x, 2024-12-23:
I wonder know how to debug in [viplanner](https://github.com/leggedrobotics/viplanner/tree/main/omniverse), e.g. the user should specify the python file. I'm confused as to how to debug this. 
I tried using Bash Debug in vscode, and although it worked fine, I still couldn't attach breakpoints.