isaac-sim/IsaacLab #1052: [Question] Wierd Movement in the custom Franka robot

chandramohanjagadeesan, 2024-09-27:
Hi, 

I created a custom usd file with the franka robot. I removed the hand and attached a fixed camera in the USD file. I copied the CFg file of Franka and edited it for my use. The robot spawns well and moves to the desired position as I want it to but when the robot is initialized in the environment the robot goes to a specific position and only then comes back to the initial position I gave in the config file and I have no idea how to remove this movement. Any help is appreciated.

mpgussert, 2024-10-02:
Hello @chandramohanjagadeesan!

From your description, it sounds like the simulation is returning to it's default state.  In Isaac Sim, this would by like hitting play, and then hitting stop. However, I can't help you more without additional information about your problem.  How are you commanding the robot? Can you share that script?

Thanks!

chandramohanjagadeesan, 2024-10-03:
Hi @mpgussert, 

thanks for the reply. I can provide you more information now. I tried to debug and I found that this issue does not happen if i import the same robot in Manager based environment but happens only in Direct RL environment. After making the environment when the env.step is called with some action even though the first action is given the robot goes to default state and robot goes to the position i gave during the second step. When i tried giving different action for the second step the tobot reached the position only after env.step was called thrice. To me it seems like there is a delay in the action given to the robot. 
[image]
This is after env.reset(). The robot is in the initial position given in the cfg file.
[image]
This is after env.step() is called with action [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0] 

The robot reached this position in step 2. 

If you need more information eg the cfg files etc.. please let me know...

chandramohanjagadeesan, 2024-10-04:
[image]
This is the script for applying actions. I have also tried to use set_joint_position_target which also gives me the same error.

I tried disabling gravity in the ArticulationCfg that also did not work

chandramohanjagadeesan, 2024-10-04:
# Copyright (c) 2022-2024, The Isaac Lab Project Developers.
# All rights reserved.
#
# SPDX-License-Identifier: BSD-3-Clause

"""Configuration for the Franka Emika robots with Zivid 2 Camera.

The following configurations are available:

* :obj:`FRANKA_PANDA_CFG`: Franka Emika Panda robot with Zivid 2 Camera
* :obj:`FRANKA_PANDA_HIGH_PD_CFG`: Franka Emika Panda robot with zivid 2 camera with stiffer PD control

Zivid 2 camera and the attachment for the camera with the robot were downloaded as .stl file and the models were attached together to form the camera assembly in auto desk  fusion 360. This model was exported as USDa file and this Usda file was imported in to Isaacsim. USD file of frnaka emika robot without the fingers and hand were combined with the camera assemble model. That USD file became Franka_ZIVID.USD . The following CFG file is the articulation and actuator config file for Franka_Zivid  
"""

import omni.isaac.lab.sim as sim_utils
from omni.isaac.lab.actuators import ImplicitActuatorCfg
from omni.isaac.lab.assets.articulation import ArticulationCfg
import os 

dir_path = os.path.dirname(os.path.realpath(__file__))

##
# Configuration
##

FRANKA_ZIVID_CFG = ArticulationCfg(
    spawn=sim_utils.UsdFileCfg(
        usd_path=f"{dir_path}/../USD/Franka_Zivid.usd",
        activate_contact_sensors=False,
        rigid_props=sim_utils.RigidBodyPropertiesCfg(
            disable_gravity=False,
            max_depenetration_velocity=5.0,
        ),
        articulation_props=sim_utils.ArticulationRootPropertiesCfg(
            enabled_self_collisions=True, solver_position_iteration_count=8, solver_velocity_iteration_count=0
        ),
        # collision_props=sim_utils.CollisionPropertiesCfg(contact_offset=0.005, rest_offset=0.0),
    ),
    init_state=ArticulationCfg.InitialStateCfg(
        joint_pos={
            "panda_joint1": 0.0,
            "panda_joint2": -0.569,
            "panda_joint3": 0.0,
            "panda_joint4": -2.810,
            "panda_joint5": 0.0,
            "panda_joint6": 3.037,
            "panda_joint7": 0.741
        },
        pos=(0.0,0.0,0.0),
    ),
    actuators={
        "panda_shoulder": ImplicitActuatorCfg(
            joint_names_expr=["panda_joint[1-4]"],
            effort_limit=87.0,
            velocity_limit=2.175,
            stiffness=80.0,
            damping=4.0,
        ),
        "panda_forearm": ImplicitActuatorCfg(
            joint_names_expr=["panda_joint[5-7]"],
            effort_limit=12.0,
            velocity_limit=2.61,
            stiffness=80.0,
            damping=4.0,
        ),
    },
    soft_joint_pos_limit_factor=1.0,
)
"""Configuration of Franka Emika Zivid robot."""

FRANKA_ZIVID_HIGH_PD_CFG = FRANKA_ZIVID_CFG.copy()
FRANKA_ZIVID_HIGH_PD_CFG.spawn.rigid_props.disable_gravity = True
FRANKA_ZIVID_HIGH_PD_CFG.actuators["panda_shoulder"].stiffness = 400.0
FRANKA_ZIVID_HIGH_PD_CFG.actuators["panda_shoulder"].damping = 80.0
FRANKA_ZIVID_HIGH_PD_CFG.actuators["panda_forearm"].stiffness = 400.0
FRANKA_ZIVID_HIGH_PD_CFG.actuators["panda_forearm"].damping = 80.0
"""Configuration of Franka Emika Zivid robot with stiffer PD control.

This configuration is useful for task-space control using differential IK.

This is my configuration file

chandramohanjagadeesan, 2024-10-17:
Hello @mpgussert 

I changed the target position values for each joint in the USD file and the initial position changed but not to the one given in the target position. Can you please let me know the tag for initial position in USD ?

closed 2024-10-23 by chandramohanjagadeesan