isaac-sim/IsaacLab #1061: [Question] How to check collision between two robot arms?

KevinGeng3, 2024-09-28:
### Question

Based on franka_lift_cube task, I add one more franka arm. So now I need to check collision between two arms, but I think there is no collision detection API right now. Does someone know how to implement this function? Thank you very much!

KevinGeng3, 2024-09-28:
One idea is using contact_sensor, but I do not know how to filter the contact between arm and cube. I think there is no clear tutorial about how to use contact_sensor.

Mayankm96, 2024-10-01:
Do the docs here clarify your issue?

> https://isaac-sim.github.io/IsaacLab/source/api/lab/omni.isaac.lab.sensors.html#omni.isaac.lab.sensors.ContactSensor

KevinGeng3, 2024-10-08:
> Do the docs here clarify your issue?
> 
> > https://isaac-sim.github.io/IsaacLab/source/api/lab/omni.isaac.lab.sensors.html#omni.isaac.lab.sensors.ContactSensor

Thank you for your advice! But actually, before I wrote this question, I have already read this docs. However, I met some problems when I put it into practice. But now I have cleared the problems.

closed 2024-10-08 by KevinGeng3

ZhangHan0301, 2025-03-12:
> Thank you for your advice! But actually, before I wrote this question, I have already read this docs. However, I met some problems when I put it into practice. But now I have cleared the problems.

"May I ask how you solved it?"