isaac-sim/IsaacLab #1070: [Bug Report] Multiple Tiled Cameras Override Each Other

ori-gadot, 2024-09-29:
### Bug Description

When using two tiled cameras in the scene, the last camera overrides all observations, resulting in both cameras capturing the same image.

### Steps to reproduce

1. Run the following script, which initializes the `CartpoleRGBCameraEnvCfg` environment with two environments and two tiled cameras.

```
from omni.isaac.lab.app import AppLauncher

app_launcher = AppLauncher(headless=False, enable_cameras=True)
simulation_app = app_launcher.app

"""Rest everything follows."""

import torch

from omni.isaac.lab.envs import ManagerBasedRLEnv
from omni.isaac.lab.sensors.camera.utils import save_images_to_file

from omni.isaac.lab_tasks.manager_based.classic.cartpole import CartpoleRGBCameraEnvCfg

def main():
    env_cfg = CartpoleRGBCameraEnvCfg()
    env_cfg.scene.num_envs = 2
    env = ManagerBasedRLEnv(cfg=env_cfg)

    while simulation_app.is_running():
        with torch.inference_mode():
            actions = torch.rand_like(env.action_manager.action)
            obs, _, _, _, _ = env.step(actions)
            save_images_to_file(obs["policy"]["image1"], "/tmp/image1.png")
            save_images_to_file(obs["policy"]["image2"], "/tmp/image2.png")
    env.close()

if __name__ == "__main__":
    main()
    simulation_app.close()
```

2. Modify the `CartpoleRGBCameraSceneCfg` class in the following file: `source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/manager_based/classic/cartpole/cartpole_camera_env_cfg.py`, adding two tiled cameras as shown below:

```
@configclass
class CartpoleRGBCameraSceneCfg(CartpoleSceneCfg):
    tiled_camera1: TiledCameraCfg = TiledCameraCfg(
        prim_path="{ENV_REGEX_NS}/Camera1",
        offset=TiledCameraCfg.OffsetCfg(pos=(-7.0, 0.0, 3.0), rot=(0.9945, 0.0, 0.1045, 0.0), convention="world"),
        data_types=["rgb"],
        spawn=sim_utils.PinholeCameraCfg(
            focal_length=24.0, focus_distance=400.0, horizontal_aperture=20.955, clipping_range=(0.1, 20.0)
        ),
        width=1280,
        height=720,
    )

    tiled_camera2: TiledCameraCfg = TiledCameraCfg(
        prim_path="{ENV_REGEX_NS}/Camera2",
        offset=TiledCameraCfg.OffsetCfg(pos=(-7.0, 0.0, 3.0), convention="world"),
        data_types=["rgb"],
        spawn=sim_utils.PinholeCameraCfg(
            focal_length=24.0, focus_distance=400.0, horizontal_aperture=20.955, clipping_range=(0.1, 20.0)
        ),
        width=1280,
        height=720,
    )
```

3. Modify `RGBObservationsCfg` to include observations for both cameras:

```
@configclass
class RGBObservationsCfg:
    """Observation specifications for the MDP."""

    @configclass
    class RGBCameraPolicyCfg(ObsGroup):
        """Observations for policy group."""

        image1 = ObsTerm(func=grab_images, params={"sensor_cfg": SceneEntityCfg("tiled_camera1"), "data_type": "rgb", "normalize": True})
        image2 = ObsTerm(func=grab_images, params={"sensor_cfg": SceneEntityCfg("tiled_camera2"), "data_type": "rgb", "normalize": True})

        def __post_init__(self) -> None:
            self.enable_corruption = False
            self.concatenate_terms = False

    policy: ObsGroup = RGBCameraPolicyCfg()
```

4. Modify the scene lights in `source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/manager_based/classic/cartpole/cartpole_env_cfg.py` for clearer image results:

```
@configclass
class CartpoleSceneCfg(InteractiveSceneCfg):
    """Configuration for a cart-pole scene."""

    # ground plane
    ground = AssetBaseCfg(
        prim_path="/World/ground",
        spawn=sim_utils.GroundPlaneCfg(size=(100.0, 100.0), color=(0.5, 0.5, 0.5)),
    )

    # cartpole
    robot: ArticulationCfg = CARTPOLE_CFG.replace(prim_path="{ENV_REGEX_NS}/Robot")

    # lights
    dome_light = AssetBaseCfg(
        prim_path="/World/DomeLight",
        spawn=sim_utils.DomeLightCfg(color=(0.9, 0.9, 0.9), intensity=10.0),
    )
    distant_light = AssetBaseCfg(
        prim_path="/World/DistantLight",
        spawn=sim_utils.DistantLightCfg(color=(0.9, 0.9, 0.9), intensity=1.0),
        init_state=AssetBaseCfg.InitialStateCfg(rot=(0.738, 0.477, 0.477, 0.0)),
    )

    sphere_light = AssetBaseCfg(
        prim_path="{ENV_REGEX_NS}/SphereLight",
        spawn=sim_utils.SphereLightCfg(
            color=(0.1, 0.1, 0.1), enable_color_temperature=True, color_temperature=5500, intensity=100, radius=2.0, exposure=10.0,
        ),
        init_state=AssetBaseCfg.InitialStateCfg(pos=(0, 0, 5)),
    )
```

### Additional context

Two identical tiled cameras with slightly different orientations (one capturing more of the ground, the other more of the sky) were used. Observation data was saved as images for inspection.

Running the following scenarios, only the first one reproduces the bug:

- Scenario 1: Running two tiled cameras in two environments causes the bug—both image files show the second camera’s view.

	- Camera 1	
		[image]

	- Camera 2
		[image]

- Scenario 2: Running two tiled cameras in a single environment works correctly—each camera renders its own FOV.

	- Camera 1	
		[image]

	- Camera 2
		[image]

- Scenario 3: Running two non-tiled cameras in two environments works as expected.

	- Camera 1
		[image]

	- Camera 2
		[image]

Note: Although it may be unrelated, there is a noticeable decline in image quality in the first scenario compared to the other two.

garylvov, 2024-09-29:
I can confirm that this issue already happens by the time that the image is read by grab_images in the observation group.

I've combined your repro steps in a single standalone file for easier troubleshooting here: 
https://gist.github.com/garylvov/d9cd169883838bd5b43d2f7b6d75c52c

garylvov, 2024-09-29:
@Dhoeller19 @Mayankm96 I believe that this may be on Isaac sim side, or maybe the buffers in `tiled_camera.py` aren't created/used correctly

WangYCheng23, 2024-09-30:
same issue.

kellyguo11, 2024-09-30:
Thanks for the report, I don't believe we can currently have multiple tiled cameras at different resolutions as the resolution settings are global. We'll look into adding support for this.

glvov-bdai, 2024-09-30:
@kellyguo11 I believe this issue still happens even if the multiple tiled cameras are the same resolution

kellyguo11, 2024-10-01:
Yes, you are right. It seems currently we can only have one tiled camera. The fix is in the works to support multiple tiled cameras, including at different resolutions. Hopefully it'll be made available in the next release.

glvov-bdai, 2024-10-01:
@elle-miller Has an interesting observation @kellyguo11 . I can confirm that when I set the camera views with `set_world_poses_from_view`, I am able to get different images from different tiled cameras in the same environment.

To me, this would suggest that there is a bug in our side of the interface with the tiled camera. I can also confirm, that even without setting world poses from view, that when viewing several tiled camera through the isaac viewport, that they look at the correct locations. 

Maybe we can add in some sort of hot fix @kellyguo11 where we use set_world_poses_from_view to put a band aid over multiple cameras not working for the time being. If you or @ori-gadot can run my reproduction script to confirm that this fixes the issue I can write a PR including this fix to unblock people prior to the next release

Using set_world_poses_from_view : https://gist.github.com/glvov-bdai/767ad9d76ea2a10b625fe7cbfbc21c20

[image]

kensukenk, 2024-10-02:
Hi, I have a similar issue. 

When I have a Franka environment with a front-facing and wrist camera, I can get the Tiled Images to render properly. However, the moment I have two environments, the TiledCamera on the wrist has an incorrect orientation:

[image]
Above is what my wrist camera is supposed to look like. It tracks the arm normally.

[image]
This is what the wrist camera looks like if I have multiple environments. However, my front camera renders properly (i.e., it doesn't seem like the wrist camera overwrites the front camera. I will sat that the front camera suffers from #1031 

[image]
Normal front camera view

[image]
Front Camera view if I have parallelized environments.

glvov-bdai, 2024-10-02:
@kensukenk Can you please be a little bit more explicit about what the problem is? Is it the camera not being in the desired orientation when using Tiled Cameras but it is in the correct orientation with USD Cameras? 

If you are using set_world_poses for your camera (or anything related to pose), I'd double check that the pose convention is consistent ("opengl" or "ros" or "world) for all cameras. From what I was able to piece together from your photos, I'd guess that the usd camera is in OpenGL format, whetheras the Tiled camera may be in ROS format, but I'm not sure.

from omni.isaac.lab.sensors.camera.camera.py

        Since different fields use different conventions for camera orientations, the method allows users to
        set the camera poses in the specified convention. Possible conventions are:

        - :obj:`"opengl"` - forward axis: -Z - up axis +Y - Offset is applied in the OpenGL (Usd.Camera) convention
        - :obj:`"ros"`    - forward axis: +Z - up axis -Y - Offset is applied in the ROS convention
        - :obj:`"world"`  - forward axis: +X - up axis +Z - Offset is applied in the World Frame convention

        See :meth:`omni.isaac.lab.sensors.camera.utils.convert_orientation_convention` for more details
        on the conventions.

Can you please share code to replicate your results? I'm surprised that the multiple cameras don't override each other and would be curious to investigate this further.

kensukenk, 2024-10-02:
Sure, my code is a modification of the Franka Cube Lift env. My issue is that when I use only one environment, my TiledCamera renders correctly. However, when I use two or more environments in parallel (and change nothing else), the TiledCamera on the franka wrist flips orientation.

`# All rights reserved.
#
# SPDX-License-Identifier: BSD-3-Clause

from dataclasses import MISSING

import omni.isaac.lab.sim as sim_utils
from omni.isaac.lab.assets import ArticulationCfg, AssetBaseCfg, RigidObjectCfg
from omni.isaac.lab.envs import ManagerBasedRLEnvCfg
from omni.isaac.lab.managers import CurriculumTermCfg as CurrTerm
from omni.isaac.lab.managers import EventTermCfg as EventTerm
from omni.isaac.lab.managers import ObservationGroupCfg as ObsGroup
from omni.isaac.lab.managers import ObservationTermCfg as ObsTerm
from omni.isaac.lab.managers import RewardTermCfg as RewTerm
from omni.isaac.lab.managers import SceneEntityCfg
from omni.isaac.lab.managers import TerminationTermCfg as DoneTerm
from omni.isaac.lab.scene import InteractiveSceneCfg
from omni.isaac.lab.sensors.frame_transformer.frame_transformer_cfg import FrameTransformerCfg
from omni.isaac.lab.sim.spawners.from_files.from_files_cfg import GroundPlaneCfg, UsdFileCfg
from omni.isaac.lab.utils import configclass
from omni.isaac.lab.utils.assets import ISAAC_NUCLEUS_DIR
from omni.isaac.lab.sensors import TiledCameraCfg, CameraCfg
from omni.isaac.lab.envs.mdp.observations import grab_images

from . import mdp

##
# Scene definition
##

@configclass
class ObjectTableSceneCfg(InteractiveSceneCfg):
    """Configuration for the lift scene with a robot and a object.
    This is the abstract base implementation, the exact scene is defined in the derived classes
    which need to set the target object, robot and end-effector frames
    """

    # robots: will be populated by agent env cfg
    robot: ArticulationCfg = MISSING
    # end-effector sensor: will be populated by agent env cfg
    ee_frame: FrameTransformerCfg = MISSING
    # target object: will be populated by agent env cfg
    object: RigidObjectCfg = MISSING
    #camera: TiledCameraCfg = MISSING
    wrist_cam: CameraCfg | TiledCameraCfg = MISSING
    front_cam: CameraCfg | TiledCameraCfg = MISSING

    # Table
    table = AssetBaseCfg(
        prim_path="{ENV_REGEX_NS}/Table",
        init_state=AssetBaseCfg.InitialStateCfg(pos=[0.5, 0, 0], rot=[0.707, 0, 0, 0.707]),
        spawn=UsdFileCfg(usd_path=f"{ISAAC_NUCLEUS_DIR}/Props/Mounts/SeattleLabTable/table_instanceable.usd"),
    )

    # plane
    plane = AssetBaseCfg(
        prim_path="/World/GroundPlane",
        init_state=AssetBaseCfg.InitialStateCfg(pos=[0, 0, -1.05]),
        spawn=GroundPlaneCfg(),
    )

    # lights
    light = AssetBaseCfg(
        prim_path="/World/light",
        spawn=sim_utils.DomeLightCfg(color=(0.75, 0.75, 0.75), intensity=3000.0),
    )

##
[... 327 lines cut ...]
            prim_path="{ENV_REGEX_NS}/Robot/panda_hand/wrist_cam",
            update_period=0.01,
            height=128,
            width=128,
            data_types=["rgb"],
            spawn=PinholeCameraCfg(
                focal_length=24.0, focus_distance=400.0, horizontal_aperture=20.955, clipping_range=(0.1, 20)
            ),
            offset=TiledCameraCfg.OffsetCfg(pos=(0.1, 0.0, 0.0), rot=(0.6848517, -0.1750107, -0.1748713, 0.6853973), convention="ros"),
        )
        
        self.scene.front_cam = TiledCameraCfg(
                prim_path="{ENV_REGEX_NS}/front_cam",
            update_period=0.01,
            height=128,
            width=128,
            data_types=["rgb"],
            spawn=PinholeCameraCfg(
                focal_length=24.0, focus_distance=400.0, horizontal_aperture=20.955, clipping_range=(0.1, 20)
            ),
            offset=TiledCameraCfg.OffsetCfg(pos=(0.75, 0.0, 0.75), rot=(-0.1259888, 0.6960601, 0.695506, -0.1260892 ), convention="ros"),
        )
`

In particular, I defined the TiledCameras in the last text chunks directly above (self.scene.front_cam, self.scene.wrist_cam)

glvov-bdai, 2024-10-02:
@kensukenk Thanks for the additional information. This looks like a separate issue to me to do with how poses are being set, do you mind opening a new issue with this information so we can keep this issue more focused to cameras overriding each other?

Also if you're able to provide the steps to reproduce as a standalone file in the style of this gist:
https://gist.github.com/garylvov/d9cd169883838bd5b43d2f7b6d75c52c

that would help the dev team trouble shoot this issue faster. Otherwise, we have to recreate/keep track of your modifications, which can slow things down.

Thanks!

Mayankm96, 2025-02-20:
This should be fixed in IsaacLab 2.0 and IsaacSim 4.5. The issue goes to the Omniverse side so previous versions of IsaacLab and IsaacSim do not support multi-resolutions tiled cameras.

Please open this issue if problem still persists.