isaac-sim/IsaacLab #1071: [Question] Tiled Cameras with Different Resolutions

ori-gadot, 2024-09-29:
### Question

Is it possible to run an environment with two tiled cameras that have different width and height settings? At the moment, I receive a warning when attempting this, and the smaller images are being tiled incorrectly.

garylvov, 2024-09-29:
I've only been able to do tiled cameras of the same resolution. As a messy workaround for the time being, you could use the highest resolution for all cameras and downsample one or more of the cameras to be of the smaller desired resolution, but of course, this has a performance overhead.

Maybe @kellyguo11 @pascal-roth  @Dhoeller19 have a more robust fix

garylvov, 2024-09-29:
I can confirm that this is not currently supported by Isaac sim

from omni.replicator.core.scripts.create 

def render_product_tiled(
    cameras: List[Union[str, ReplicatorItem]],
    tile_resolution: Tuple[int, int],
    force_new: bool = False,
    name: str = None,
) -> viewport_manager.HydraTexture:
    """Creates a single render product to render multiple sensor tiles.

    Creates a render product attached to a list of sensors and outputs a tiled buffer with the output of all attached sensors.
    Tiled rendering is most performant when rendering a large number of sensors at low resolution. Returns a ``HydraTexture`` object
    with resolution equal ``tile_resolution[0] * ceil(sqrt(len(cameras))), tile_resolution[1] * ceil(sqrt(len(cameras))``.

    Note: Only a single tiled resolution is allowed per session. Creating more render products with different tiled resolution
        will cause the previous tiled render products to render incorrectly.