isaac-sim/IsaacLab #1076: [Proposal] only positive reward

fan-ziqi, 2024-09-30:
### Proposal

Users can selectively calculate only positive rewards in reward manager

mpgussert, 2024-10-02:
Users have full control over the reward function when defining their own environment.  It's not clear to me what you are asking for here.  Do you want a toggle to invert the rewards on the stand alone samples?

closed 2024-10-10 by mpgussert

fan-ziqi, 2024-10-20:
Thank you for reply!

I mean to say, the `compute_reward` function in legged_gym uses `only_positive_rewards` to control whether the reward is reduced to 0. Do you plan to introduce this feature in `reward_manager` as well?

```python
    def compute_reward(self):
        """ Compute rewards
            Calls each reward function which had a non-zero scale (processed in self._prepare_reward_function())
            adds each terms to the episode sums and to the total reward
        """
        self.rew_buf[:] = 0.
        for i in range(len(self.reward_functions)):
            name = self.reward_names[i]
            rew = self.reward_functions[i]() * self.reward_scales[name]
            self.rew_buf += rew
            self.episode_sums[name] += rew
        if self.cfg.rewards.only_positive_rewards:
            self.rew_buf[:] = torch.clip(self.rew_buf[:], min=0.)
        # add termination reward after clipping
        if "termination" in self.reward_scales:
            rew = self._reward_termination() * self.reward_scales["termination"]
            self.rew_buf += rew
            self.episode_sums["termination"] += rew
```