isaac-sim/IsaacLab #1078: [Question] My manipulator shakes too much when I train rl in skrl.

Holiclife-KTH, 2024-09-30:
created a USD file for the manipulator (UR5e + 2F85 gripper) and trained it for reaching and pushing tasks using the skrl library. However, the manipulator shakes too much compared to when I trained it using rsl_rl. Is there a parameter to make the manipulator move more smoothly?
Please, help me!!!

https://github.com/user-attachments/assets/6633ecc8-8bfd-4c6e-947e-3693c965abdc

Mayankm96, 2024-10-01:
@Toni-SM I think we should always do "inference" with the mean of the distribution. Is there any particular reason why the default behavior is sampled?

Toni-SM, 2024-10-01:
> Is there any particular reason why the default behavior is sampled?

@Mayankm96  I have not done it before because I need to add an extra step to make it compatible with multi-agent algorithms.

mpgussert, 2024-10-02:
@Mayankm96 @Toni-SM this may be a consequence of the mode not being set. Checkout `set_mode` in https://skrl.readthedocs.io/en/latest/api/agents.html . Most modules that use distribution are sensitive to this pytorch mode, which sets the behavior of the forward call to returning the mean instead of a sample.

@Holiclife-KTH have you `set_mode('eval')` for the `runner.agent` ?

RandomOakForest, 2024-11-22:
Thanks for posting this. This is a great discussion if you still face this issue. Maybe you can try the latest release posted this week. Let us know may we be of further help.

KevinGeng3, 2024-12-17:
Great discussion!

celestialdr4g0n, 2025-02-24:
Hi, I see `actions = outputs[-1].get("mean_actions", outputs[0])` in play.py of skrl (Feb 24 2025). Does this mean the "mean_actions" is the default now? If this the default now then this method does not work for me.

mukundkk, 2025-11-03:
Hey, just wanted to check if there were any updates on whether using mean actions was supposed to resolve this, or if there was another solution? I'm using SKRL 1.4.3 where mean actions are being used during inference, but I'm still getting pretty significant jitters during both training and evaluation so I imagine there should be another cause, but not sure what it is.

AntoineRichard, 2026-08-27:
Thanks @Holiclife-KTH for the clear report (and the video), and thanks everyone for the discussion here — closing this out with a summary of where things stand and concrete tuning guidance for the follow-up questions.

**The original play-script issue is fixed.** The skrl play script now uses the deterministic distribution mean by default: `scripts/reinforcement_learning/skrl/play.py` does `actions = outputs[-1].get("mean_actions", outputs[0])` for single-agent runs, and multi-agent algorithms got the same treatment in https://github.com/isaac-sim/IsaacLab/pull/1972. So stochastic sampling at inference time is no longer the cause of shaking on a current release.

**@celestialdr4g0n, @mukundkk** — jitter that persists *with* mean actions (and during training) usually means the policy itself learned a jittery solution, and that is tuned through the reward and actuator config rather than the play script:

1. **Action-smoothness penalties.** Isaac Lab ships `action_rate_l2` (penalizes change in action between steps) and `joint_vel_l2` (penalizes joint velocities) in `source/isaaclab/isaaclab/envs/mdp/rewards.py`. The shipped reach task is a good reference for how to weight them: `source/isaaclab_tasks/isaaclab_tasks/manager_based/manipulation/reach/reach_env_cfg.py` starts both at a small weight (`-0.0001`) and uses curriculum terms (`mdp.modify_reward_weight`) to ramp them up to `-0.005` / `-0.001` after 4500 steps — so the policy first learns the task, then learns to do it smoothly. If your arm shakes, these weights are the first knob: too small and the policy has no incentive to be smooth; ramping too early can prevent it from learning the task at all. `joint_acc_l2` and `joint_torques_l2` in the same file are additional options for stubborn cases.

2. **Actuator gains.** Under-damped PD gains make the arm oscillate around every target no matter how smooth the actions are. For a UR arm, compare against `UR10_CFG` in `source/isaaclab_assets/isaaclab_assets/robots/universal_robots.py` (`stiffness=800.0`, `damping=40.0` on an `ImplicitActuatorCfg`). If you built your own USD, make sure your config's stiffness/damping actually apply to your joints — a zero or near-zero damping is a very common cause of exactly this symptom.

3. **Action scale.** With `JointPositionActionCfg`, a large `scale` lets a small policy-output change command a big joint-target jump. The shipped UR10 reach config uses `scale=0.5` (`.../reach/config/ur_10/joint_pos_env_cfg.py`); reducing scale is a quick way to trade responsiveness for smoothness.

Since rsl_rl training was smooth on the same robot for the original poster, comparing the reward terms and action scale between the two run configs is the most direct diagnostic.

**@joon725** — combining a UR5e with a 2F-85 gripper is an asset-composition question rather than this training question; the asset configs in `source/isaaclab_assets/isaaclab_assets/robots/universal_robots.py` (which include UR10e + Robotiq gripper variants you can adapt) and a new dedicated issue with your specific error are the best route there.

If shaking still reproduces on a current release with the smoothness penalties and gains tuned as above, please open a fresh issue with your reward config and actuator settings — that makes it much easier to pin down.

---
*🤖 This issue was reviewed as part of an AI-assisted triage of the Isaac Lab backlog, with automated verification against the current codebase. If we got this wrong, please reopen — we'd rather hear about it!*

closed 2026-08-27 by AntoineRichard