isaac-sim/IsaacLab #1082: [Bug Report] EMAJointPositionToLimitsAction bug

Msornerrrr, 2024-09-30:
### Describe the bug

For the EMAJointPositionToLimitsAction controller, during reset(), it's trying to reset history to current joint positions. The indexing part
```
self._asset.data.joint_pos[env_ids, self._joint_ids]
```
has error if EMA control is applied for just part of the robot (e.g. arm + hand setup)

### Steps to reproduce

I don't find any current robot that's using the arm + hand setup so reproducing the error would be hard. But I do have a one line fix for this issue which I will describe latter.

### Fixing the error

```
self._prev_applied_actions[env_ids, :] = self._asset.data.joint_pos[env_ids][:, self._joint_ids]
```

### System Info

Describe the characteristic of your environment:

- Commit: [e.g. 8f3b9ca]
- Isaac Sim Version: 2022.2.0
- OS: Ubuntu 22.04
- GPU: NVIDIA GeForce RTX 4090
- CUDA: 12.4
- GPU Driver: 550.107.02

Mayankm96, 2024-10-01:
This seems like a broadcasting issue. A better fix probably is:

```python
self._prev_applied_actions[env_ids, :] = self._asset.data.joint_pos[env_ids[:, None], , self._joint_ids]
```

Can you try if that works? If yes, can you make an MR? Thanks!

bearpaw, 2024-10-10:
I got the same error with the following setting:

### Describe the bug

In some cases, we want to move certain joints instead of all the joints when we train an agent. This is enable by passing argument `env.actions.joint_pos.joint_names`.  This will cause an indexing error in pytorch.

### Steps to reproduce

For example, we want to only move the index finger of the Allegro hand for the cube reposing task:

```bash
python source/standalone/workflows/rsl_rl/train.py  \
 --task Isaac-Repose-Cube-Allegro-v0   \
--num_env 12 \
env.actions.joint_pos.joint_names='["index_joint.*"]'
```

We will get the following error:
```bash
Error executing job with overrides: ['env.actions.joint_pos.joint_names=["index_joint.*"]']                                                                                                                   
Traceback (most recent call last):                                                                                                                                                                            
  File "/home/weiy/Projects/code/IsaacLab/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/utils/hydra.py", line 91, in hydra_main
    func(env_cfg, agent_cfg, *args, **kwargs)                                                                                                                                                                 
  File "/home/weiy/Projects/code/IsaacLab/source/standalone/workflows/rsl_rl/train.py", line 119, in main                    
    env = RslRlVecEnvWrapper(env)                                                                                                                                                                             
  File "/home/weiy/Projects/code/IsaacLab/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/utils/wrappers/rsl_rl/vecenv_wrapper.py", line 89, in __init__
    self.env.reset()                                                                                                                                                                                          
  File "/home/weiy/miniconda3/envs/isaaclab/lib/python3.10/site-packages/gymnasium/wrappers/order_enforcing.py", line 61, in reset
    return self.env.reset(**kwargs)                                                                                                                                                                   
  File "/home/weiy/Projects/code/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/manager_based_env.py", line 252, in reset                                                                      
    self._reset_idx(indices)                   
  File "/home/weiy/Projects/code/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/manager_based_rl_env.py", line 339, in _reset_idx                                                              
    info = self.action_manager.reset(env_ids)  
  File "/home/weiy/Projects/code/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/managers/action_manager.py", line 285, in reset                                                                     
    term.reset(env_ids=env_ids)                    
  File "/home/weiy/Projects/code/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/mdp/actions/joint_actions_to_limits.py", line 188, in reset                                                    
    self._prev_applied_actions[env_ids, :] = self._asset.data.joint_pos[env_ids, self._joint_ids]      
IndexError: shape mismatch: indexing tensors could not be broadcast together with shapes [12], [4] 
```

### System Info

Describe the characteristic of your environment:

- Commit: 225a8961f5b286af8684263f25e34f198c079f97
- Isaac Sim Version: 4.2
- OS: Ubuntu 24.04.1
- GPU: RTX 3090
- CUDA: 12.4
- GPU Driver: 550.54.15

### Additional context

NA

### Checklist

- [ ] I have checked that there is no similar issue in the repo (**required**)
- [x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Criteria 1
- [ ] Criteria 2

closed 2024-10-22 by Mayankm96

benrichardson28, 2025-08-22:
This bug seems to have resurfaced in Isaac Sim Version: 4.5 (now on line 206).

If I define EMAJointPositionToLimitsAction joints=[".*"] it works fine (self._joint_ids = slice(None,None,None)), but if the joints are explicitly set, then self._joint_ids = list[int], causing self._asset.data.joint_pos[env_ids, self._joint_ids] to have two dimensions instead of three.

benrichardson28, 2025-08-22:
This bug seems to have resurfaced in Isaac Sim Version: 4.5 (now on line 206).
206: self._prev_applied_actions[env_ids, :] = self._asset.data.joint_pos[env_ids, self._joint_ids]

### Steps to reproduce
If I define EMAJointPositionToLimitsAction joints=[".*"] it works fine (self._joint_ids = slice(None,None,None)), but if the joints are explicitly set, then self._joint_ids = list[int], causing self._asset.data.joint_pos[env_ids, self._joint_ids] to have two dimensions instead of three.

### Fixing the error
```self._prev_applied_actions[env_ids, :] = self._asset.data.joint_pos[env_ids][:,:,self._joint_ids]```
or
```self._prev_applied_actions[env_ids, :] = self._asset.data.joint_pos[env_ids][...,self._joint_ids]```

These seem to work for both cases at least.

### System Info
Describe the characteristic of your environment:
- Isaac Sim Version: 4.5
- OS: Ubuntu 22.04
- GPU: NVIDIA NVIDIA RTX 6000 Ada
- CUDA: 12.8
- GPU Driver: 570.169

Yasu31, 2025-10-17:
I submitted a bug report about the newly resurfaced error here: https://github.com/isaac-sim/IsaacLab/issues/3753
and a PR that fixes it here: https://github.com/isaac-sim/IsaacLab/pull/3754

commit 7e4d69b8 (2024-10-22): Fixes pytorch broadcasting issue in `EMAJointPositionToLimitsAction` (#1207)

# Fix the pytorch broadcasting error when joint_ids are not None

This PR fixes the broadcasting error in `EMAJointPositionToLimitsAction`
when `joint_ids` are not None.

Fixes #1082

## Type of change

- Bug fix (non-breaking change which fixes an issue)

## Checklist

- [x] I have run the [`pre-commit` checks](https://pre-commit.com/) with
`./isaaclab.sh --format`
- [ ] I have made corresponding changes to the documentation
- [x] My changes generate no new warnings
- [ ] I have added tests that prove my fix is effe [... cut]