isaac-sim/IsaacLab #1085: [Question] I get black images when I save images from TiledCamera sensors

waivivian, 2024-10-02:
I tried to follow https://isaac-sim.github.io/IsaacLab/source/how-to/save_camera_output.html to save sensor images.

I get black images if I save from TiledCamera sensors.
But I can successfully get coloured images from Camera sensors.

Does the way mentioned in  https://isaac-sim.github.io/IsaacLab/source/how-to/save_camera_output.html only apply to Camera sensors?
Is there any way to get coloured images from TiledCamera sensors?

Thanks a lot!!!

garylvov, 2024-10-02:
Saving images from Tiled Cameras should work. Try running the sim for a few steps before taking images as they sometimes show up blank at the start.

 If you'd like to experiment with TIled Cameras, you can also try:

https://isaac-sim.github.io/IsaacLab/source/how-to/estimate_how_many_cameras_can_run.html

If you'd like to save images, you can try

adding this top level import

from omni.isaac.lab.sensors.camera.utils import save_images_to_file

and then on roughly line 665 of source/standalone/tutorials/04_sensors/benchmark_cameras.py, you can call save images to file on the camera like 

```
 # Loop through all camera lists and their data_types
        for camera_list, data_types, label in zip(camera_lists, camera_data_types, labels):
            for cam_idx, camera in enumerate(camera_list):
                   save_image_to_file(camera.data.output["rgb"], FILE_PATH) # MODIFY HERE 
```

For more examples of saving tiled rendering you can see https://github.com/isaac-sim/IsaacLab/issues/1070 , (mentions how there is currently a bug for multiple cameras but does illustrate correctly how to save images from tiled cameras)

If none of the above helps, please share steps to reproduce your issue.

Dhoeller19, 2024-10-02:
Hi all,
Exactly, the run_usd_camera.py is meant to showcase normal cameras. However, you can make it work with Tiledcameras: replace everything in the `if args_cli.save` statement with `save_images_to_file(camera.data.output["rgb"]/ 255.0, f"out.png")` (you need to add `from omni.isaac.lab.sensors import save_images_to_file` at the beginning of the script.