isaac-sim/IsaacLab #1126: [Question] Transformation of deformable object

Entongsu, 2024-10-02:
Hi,

Is it possible to get the transformation matrix of the deformable object during the manipulation? Thank you!

mpgussert, 2024-10-02:
Hello @Entongsu!

It's not clear what you mean by this.  Are you looking for the Transformation Matrix (XForm) for the prim? or are you interested in some transformation that characterizes the deformation?  The former we can certainly do, but he latter I am not so sure...

Gus

mpgussert, 2024-10-02:
How did you get that view? is it from `root_physx_view` on `DeformableObject`?  Are you able to share your code?

masoudmoghani, 2024-10-03:
Hi @Entongsu you should be able to use "root_pos_w" to get the root pos of the deformable body, please see: https://github.com/isaac-sim/IsaacLab/blob/80cff1649bf0155621df255c8960ae345017c17c/source/standalone/tutorials/01_assets/run_deformable_object.py#L143

Mayankm96, 2024-10-03:
I made a small change in a separate branch. Now that `get_transforms` work again maybe we should use them?

> https://github.com/isaac-sim/IsaacLab/compare/main...fix/deformable-root-state

Entongsu, 2024-10-03:
Hi,
I have used my deformable object for the simulation. When I execute the function of [get_tranforms](https://github.com/isaac-sim/IsaacLab/blob/c1244323911e4686bb9eef00bb306b00720ad4e8/source/extensions/omni.isaac.lab/omni/isaac/lab/assets/deformable_object/deformable_object_data.py#L106), I got the error of  Exception: Failed to get soft body transforms from backend.

Mayankm96, 2024-10-04:
Which Isaac Sim version? 4.2 has it fixed from what I checked. Isaac Sim 4.1 no 😓

Entongsu, 2024-10-04:
I used the 4.2 version for that.

RandomOakForest, 2024-11-22:
Thanks for posting this. It would be great if you could try the latest release posted this week. Let us know if you still have this issue. Thanks.

AntoineRichard, 2026-08-27:
Thanks for the question, @Entongsu — this one comes up often enough that it's worth a clear answer on the record.

Deformable pose is exposed through the public data API rather than the `SoftBodyView.get_transforms()` backend call: use `root_pos_w` (derived from the nodal state) on the deformable object's data class — see `source/isaaclab_physx/isaaclab_physx/assets/deformable_object/deformable_object_data.py`. The same pattern applies on the Newton side under `source/isaaclab_newton/isaaclab_newton/assets/deformable_object`. Closing since there was no follow-up, but please reopen with a repro on a current release if the nodal/root state doesn't cover your case.

---
*🤖 This issue was reviewed as part of an AI-assisted triage of the Isaac Lab backlog, with automated verification against the current codebase. If we got this wrong, please reopen — we'd rather hear about it!*

closed 2026-08-27 by AntoineRichard