isaac-sim/IsaacLab #1162: [Bug Report] docker/container.py enter fails if `DISPLAY` env is not defined

GiulioRomualdi, 2024-10-05:
If you are submitting a bug report, please fill in the following details and use the tag [bug].

### Describe the bug
I'm trying to enter the container built into a server without display. The `enter` command fails  
as shown above

### Steps to reproduce

I started the container as written in the documentation `python3 docker/container.py start`

Then when I entered the container I got the following error:
```
python3 docker/container.py enter
[INFO] Using container profile: base
[INFO] X11 Forwarding is disabled from the settings in '.container.cfg'
[INFO] X11 forwarding is disabled. No action taken.
[INFO] Entering the existing 'isaac-lab-base' container in a bash session...

Traceback (most recent call last):
  File "/home/iit.local/gromualdi/robot-code/IsaacLab/docker/container.py", line 127, in <module>
    main(args_cli)
  File "/home/iit.local/gromualdi/robot-code/IsaacLab/docker/container.py", line 111, in main
    ci.enter()
  File "/home/iit.local/gromualdi/robot-code/IsaacLab/docker/utils/container_interface.py", line 146, in enter
    f"DISPLAY={os.environ['DISPLAY']}",
  File "/usr/lib/python3.10/os.py", line 680, in __getitem__
    raise KeyError(key) from None
KeyError: 'DISPLAY
```

### System Info

Describe the characteristics of your environment:

- Commit: e00d62561a1b4ab19985bc15b40ac77c10bd5e33 (v1.2.0)
- Isaac Sim Version: `4.2.0-rc.18+release.16044.3b2ed111.gl`
- OS: Ubuntu 22.04
- GPU: NVIDIA A100 80GB PCIe
- CUDA: 12.2
- GPU Driver: 535.183.01

### Additional context

Add any other context about the problem here.

### Checklist

- [ ] I have checked that there is no similar issue in the repo (**required**)
- [ ] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Make the script running even if the display is not defined

GiulioRomualdi, 2024-10-05:
https://github.com/isaac-sim/IsaacLab/issues/1163 attempts to solve the issue

Dhoeller19, 2024-10-07:
Thanks a lot @GiulioRomualdi for the report and the fix!
We'll get this merged asap.

PR #1163 (2024-10-10): Fixes running the docker container when the DISPLAY env variable is not defined
# Description

This PR attempts to fix #1162 by adding the possibility to do not pass the `DISPLAY` env to the docker exec command if the variable is not defined

Fixes #1162

## Type of change

- Bug fix (non-breaking change which fixes an issue)

## Screenshots

Please attach before and after screenshots of the change if applicable.

## Checklist

- [x] I have run the [`pre-commit` checks](https://pre-commit.com/) with `./isaaclab.sh --format`
- [ ] I have made corresponding changes to the documentation
- [x] My  [... cut]
commit 73c47e4f (2024-10-10): Fixes running the docker container when the DISPLAY env variable is not defined (#1163)

# Description

This PR attempts to fix #1162 by adding the possibility to do not pass
the `DISPLAY` env to the docker exec command if the variable is not
defined

Fixes #1162

## Type of change

- Bug fix (non-breaking change which fixes an issue)

## Checklist

- [x] I have run the [`pre-commit` checks](https://pre-commit.com/) with
`./isaaclab.sh --format`
- [ ] I have made corresponding changes to the documentation
- [x] My changes generate no new warnings
- [ ] I have added tests that prove my fix is e [... cut]
closed 2024-10-10 by Dhoeller19