isaac-sim/IsaacLab #1166: [Question] How to use ActuatorNet Models for Manipualtor Arms and dextrous hands

SakethTecholution, 2024-10-06:
### Question
@Mayankm96 @animesh-garg 
There is no documentation or demos provided for utilising the actuator net models of real world robot . Does this really improve or effect the training process and results in accurate motion . I have a dextrous hand but its very hard to tune or measure the physical parameters of hand to match simulation . In this case would actuatornet be helpful . If yes can you please guide me thorugh some of exisiting codes or docs to understand .

* A clear documentaiton is lacking in this part, It would be great if a sample demo with better documentation on real world robot control would be helpful

garylvov, 2024-10-07:
Some examples of training actuator networks: 

https://github.com/Improbable-AI/walk-these-ways/tree/master/scripts/actuator_net
https://github.com/garylvov/scraps/tree/main/actuator_net

as for the process of training the network, I also found https://arxiv.org/pdf/1901.08652 helpful 

@Mayankm96 can likely provide better context on when to use an actuator network - from what I remember it's designed for non-linear motor dynamics and does not provide a benefit on DC motors but I'm not sure

Dhoeller19, 2024-10-07:
Hi @SakethTecholution and thanks @garylvov for answering,
Indeed this actuator network is an expert topic that is described in the arxiv paper above. In the future,
we will release more documentation for sim-to-real transfer and we will describe the training process for the actuator networks more.