isaac-sim/IsaacLab #1171: Compatibility issues with Matlab (call matlab from python with Matlab engine)

mikelasa, 2024-10-07:
Hello,

I have a matlab script based on a manipulation toolbox that executes some computation that I will need to control my robots in the simulation. My idea is to call the matlab .m file from python using the matlab engine module to get some values from this script (I'm not translating it for now as the toolbox is quite large).

I'm having issues when importing the matlab engine module to the isaacLab framework. I followed 
the official API installation guide as follows:

[](https://es.mathworks.com/help/matlab/matlab_external/install-the-matlab-engine-for-python.html)

Since IsaacLab is using a python interpreter provided by omniverse, I tried executing the installation command as following:
`sudo /home/mikel/.local/share/ov/pkg/isaac-sim-4.2.0/python.sh -m pip install matlabengine==24.2.1`

Matlab engine installs with no issues and seems that when I launch my simulation in isaac lab, Matlab engine is imported too, but,  some compatibility issues arises after that are far beyond my knowledge and I dont know how to tackle this.

```
import argparse
from omni.isaac.lab.app import AppLauncher

# add argparse arguments
parser = argparse.ArgumentParser(description="Teleop with variable impedance control.")
parser.add_argument("--num_envs", type=int, default=128, help="Number of environments to spawn.")
parser.add_argument("--impedance_mode", type=int, default=1, help="Impedance mode: 1=fixed, 2=variable_K, 3=variable_K_D")
parser.add_argument("--command_mode", type=int, default=1, help="Command mode: 1=pose_abs, 2=pose_rel")
parser.add_argument("--sensitivity", type=float, default=1.0, help="Sensitivity factor.")
parser.add_argument("--teleop_device", type=str, default="keyboard", help="Device for interacting with environment")
parser.add_argument("--robot_init_mode", type=int, default=1, help="Start robot from grasping position")

# append AppLauncher cli args
AppLauncher.add_app_launcher_args(parser)
# parse the arguments
args_cli = parser.parse_args()

# launch omniverse app
app_launcher = AppLauncher(args_cli)
simulation_app = app_launcher.app

import torch
from omni.isaac.lab.envs import ManagerBasedEnv
from omni.isaac.lab.managers import SceneEntityCfg

import matlab.engine
```

```
Traceback (most recent call last):
  File "/home/mikel/IsaacOrbitLFD/IsaacLab/source/standalone/cartesian_impedance_control/env/bimanual_robot_no_ee.py", line 38, in <module>
    import matlab.engine
  File "/home/mikel/IsaacOrbitLFD/IsaacLab/_isaac_sim/kit/python/lib/python3.10/site-packages/matlab/__init__.py", line 48, in <module>
    from matlabmultidimarrayforpython import double, single, uint8, int8, uint16, \
ImportError: /lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /usr/local/MATLAB/R2024b/extern/bin/glnxa64/matlabmultidimarrayforpython.cpython-310-x86_64-linux-gnu.so)
```

I double checked that I have installed the correct version of matlab (R2024b) with the correct version of matlab engine too, but I'm unable to link IsaacLab with matlab.

Has anyone used matlab within issacLab? I'm not an expert programming and I will appreciate some help, I'm kind of lost here.

Toni-SM, 2024-10-07:
Kaixo @mikelasa 

This MTLAB post may help: https://www.mathworks.com/matlabcentral/answers/1907290-how-to-manually-select-the-libstdc-library-to-use-to-resolve-a-version-glibcxx_-not-found

mikelasa, 2024-10-08:
> Kaixo @mikelasa
> 
> This MATLAB post may help: https://www.mathworks.com/matlabcentral/answers/1907290-how-to-manually-select-the-libstdc-library-to-use-to-resolve-a-version-glibcxx_-not-found

Kaixo Antonio! Long time no see haha, how you doing? 

I solved my issue some hours ago, I was going to post the steps I took:

1. cd /usr/local/MATLAB/R2024b/extern/engines/python
2. sudo /home/mikel/.local/share/ov/pkg/isaac-sim-4.2.0/python.sh -m pip install matlabengine==24.2.1 

Then, the updated the library LIBSTDC 6.x and checked that GLIBCXX_3.4.29 was on the list:

1. sudo add-apt-repository ppa:ubuntu-toolchain-r/test 
2. sudo apt-get update
3. sudo apt-get upgrade
4. sudo apt-get dist-upgrade
5. strings /usr/lib/x86_64-linux-gnu/libstdc++.so.6 | grep GLIBCXX

Doing this, I can work now with matlab.

W.r.t to:

> As a final disclaimer that you are likely already aware of, I would strongly recommend that your port your matlab code to python regardless of whether you can get the engine to work. Using the matlab engine has a serious performance overhead both for RAM and CPU, and there likely will be bottlenecks for converting from GPU to matlab CPU back to GPU. In my past research, I found that just starting the engine (without any calculations) used several gb of ram. Regardless, It is much more performant to do calculations on the GPU directly then deal with many conversions.
> 
> As for the matlab manipulation toolbox being large/hard to port, at least for something like IK, you could use the existing task space controller support https://isaac-sim.github.io/IsaacLab/source/tutorials/05_controllers/run_diff_ik.html

Thank you for your advice, and yes, the performance when I use matlab is quite poor due to this reason. I'm using the toolbox to test something I have been working on as it is faster than starting to port all the code to python even if I have to sacrifice performance for now. I have in mind to port it to python once I see the results of the experiment.

The toolbox is about grasping theory based in the chapter 28 of springer handbook of robotics: 
[springer handbook of robotics](https://link.springer.com/book/10.1007/978-3-540-30301-5)

Thanks both for your answers!

closed 2024-10-08 by mikelasa