isaac-sim/IsaacLab #1181: [Proposal] Add TorchRL to IsaacLab workflows

fyu-bdai, 2024-10-08:
### Proposal

[TorchRL ](https://pytorch.org/rl/stable/index.html) is a RL framework produced by the PyTorch team that is growing in popularity within the research community. It offers fast optimization times, flexibility, and leverages tensordicts for data passing between environment and algorithm. It would be nice to add TorchRL as a new workflow option to IsaacLab. 

### Motivation

Please outline the motivation for the proposal. Summarize the core use cases and user problems and needs you are trying to solve.

Is your feature request related to a problem? e.g.,"I'm always frustrated when [...]".

If this is related to another GitHub issue, please link here too.

### Alternatives

A clear and concise description of any alternative solutions or features you've considered, if any.

### Additional context

Add any other context or screenshots about the feature request here.

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Complete TorchRL train and play pipeline integrated into IsaacLab.

mpgussert, 2024-10-10:
Good suggestion! I'll pass it along :)

Thank you!

fyu-bdai, 2024-10-10:
> Good suggestion! I'll pass it along :)
> 
> Thank you!

I've already created the PRs for this proposal :sweat_smile: