isaac-sim/IsaacLab #1182: [Bug Report] ModuleNotFoundError: No module named 'Semantics'

cidxb, 2024-10-08:
When running the tutorial script command :_**`./isaaclab.sh -p source/standalone/environments/list_envs.py`**_ 

It return the missing module Semantics.

`[INFO] Using python from: /home/xx/anaconda3/envs/RL/bin/python                                                                                                                                                                  
[INFO][AppLauncher]: Loading experience file: /home/xx/workspace/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/app/../../../../../../apps/isaaclab.python.headless.kit
[Warning] [omni.isaac.kit.simulation_app] Modules: ['omni.kit_app'] were loaded before SimulationApp was started and might not be loaded correctly.
[Warning] [omni.isaac.kit.simulation_app] Please check to make sure no extra omniverse or pxr modules are imported before the call to SimulationApp(...)
Starting kit application with the following args:  ['/home/xx/anaconda3/envs/RL/lib/python3.10/site-packages/isaacsim/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py', '/home/xx/workspace/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/app/../../../../../../apps/isaaclab.python.headless.kit', '--/app/tokens/exe-path=/home/xx/anaconda3/envs/RL/lib/python3.10/site-packages/omni', '--/persistent/app/viewport/displayOptions=3094', '--/rtx/materialDb/syncLoads=True', '--/rtx/hydra/materialSyncLoads=True', '--/omni.kit.plugin/syncUsdLoads=True', '--/app/renderer/resolution/width=1280', '--/app/renderer/resolution/height=720', '--/app/window/width=1440', '--/app/window/height=900', '--/renderer/multiGpu/enabled=True', '--/app/fastShutdown=True', '--ext-folder', '/home/xx/anaconda3/envs/RL/lib/python3.10/site-packages/isaacsim/exts', '--ext-folder', '/home/xx/anaconda3/envs/RL/lib/python3.10/site-packages/isaacsim/apps', '--/renderer/activeGpu=0', '--/physics/cudaDevice=0', '--portable', '--no-window', '--/app/window/hideUi=True']
Passing the following args to the base kit application:  []
Loading user config located at: '/home/xx/anaconda3/envs/RL/lib/python3.10/site-packages/omni/data/Kit/Isaac-Sim/4.2/user.config.json'
[Info] [carb] Logging to file: /home/xx/anaconda3/envs/RL/lib/python3.10/site-packages/omni/logs/Kit/Isaac-Sim/4.2/kit_20241008_154522.log
2024-10-08 07:45:22 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
2024-10-08 07:45:22 [1ms] [Warning] [omni.ext.plugin] [ext: omni.isaac.benchmark_environments] Extensions config 'extension.toml' doesn't exist '/home/xx/anaconda3/envs/RL/lib/python3.10/site-packages/isaacsim/exts/omni.isaac.benchmark_environments' or '/home/xx/anaconda3/envs/RL/lib/python3.10/site-packages/isaacsim/exts/omni.isaac.benchmark_environments/config'
2024-10-08 07:45:22 [1ms] [Warning] [omni.ext.plugin] [ext: omni.isaac.gym] Extensions config 'extension.toml' doesn't exist '/home/xx/anaconda3/envs/RL/lib/python3.10/site-packages/isaacsim/exts/omni.isaac.gym' or '/home/xx/anaconda3/envs/RL/lib/python3.10/site-packages/isaacsim/exts/omni.isaac.gym/config'
2024-10-08 07:45:22 [2ms] [Warning] [omni.ext.plugin] [ext: omni.isaac.robot_benchmark] Extensions config 'extension.toml' doesn't exist '/home/xx/anaconda3/envs/RL/lib/python3.10/site-packages/isaacsim/exts/omni.isaac.robot_benchmark' or '/home/xx/anaconda3/envs/RL/lib/python3.10/site-packages/isaacsim/exts/omni.isaac.robot_benchmark/config'
[0.092s] [ext: omni.kit.async_engine-0.0.1] startup
[0.125s] [ext: omni.stats-1.0.1] startup
[0.126s] [ext: omni.client-1.2.0] startup
[0.154s] [ext: omni.assets.plugins-0.0.0] startup
[0.156s] [ext: omni.gpu_foundation-0.0.0] startup
2024-10-08 07:45:23 [101ms] [Error] [carb] Interface: [carb::graphics::Graphics v3.4] requested but default plugin [carb::graphics::Graphics v3.2] cannot provide requested version
2024-10-08 07:45:23 [101ms] [Error] [carb] [Plugin: carb.profiler.gpu.plugin] Dependency: [carb::graphics::Graphics v3.4] failed to be resolved.
[0.166s] [ext: omni.kit.pipapi-0.0.0] startup
[0.167s] [ext: omni.kit.pip_archive-0.0.0] startup
[0.167s] [ext: omni.mtlx-0.1.0] startup
[0.167s] [ext: omni.usd.config-1.0.4] startup
[0.169s] [ext: omni.gpucompute.plugins-0.0.0] startup
[0.169s] [ext: omni.usd.libs-1.0.1] startup
[0.211s] [ext: omni.kit.telemetry-0.5.0] startup
2024-10-08 07:45:23 [150ms] [Error] [carb] Interface: [carb::graphics::Graphics v3.4] requested but default plugin [carb::graphics::Graphics v3.2] cannot provide requested version
2024-10-08 07:45:23 [150ms] [Error] [carb] [Plugin: carb.profiler.gpu.plugin] Dependency: [carb::graphics::Graphics v3.4] failed to be resolved.
2024-10-08 07:45:23 [150ms] [Error] [carb] Interface: [carb::graphics::Graphics v3.4] requested but default plugin [carb::graphics::Graphics v3.2] cannot provide requested version
2024-10-08 07:45:23 [150ms] [Error] [carb] [Plugin: carb.profiler.gpu.plugin] Dependency: [carb::graphics::Graphics v3.4] failed to be resolved.
[0.220s] [ext: omni.usd.schema.geospatial-0.0.0] startup
[0.266s] [ext: omni.usd.schema.semantics-0.0.0] startup
[0.275s] [ext: omni.usd.schema.physx-106.0.20] startup
[0.296s] [ext: omni.usd.schema.audio-0.0.0] startup
[0.298s] [ext: omni.usd_resolver-1.0.0] startup
[0.302s] [ext: omni.kit.actions.core-1.0.0] startup
[0.303s] [ext: omni.usd.core-1.2.11] startup
[0.305s] [ext: omni.kit.commands-1.4.9] startup
[0.308s] [ext: omni.hydra.scene_delegate-0.3.3] startup
[0.311s] [ext: omni.graph.exec-0.9.3] startup
[0.312s] [ext: omni.physx.foundation-106.0.20] startup
[0.312s] [ext: omni.kit.exec.core-0.13.2] startup
[0.313s] [ext: omni.kit.usd_undo-0.1.8] startup
[0.313s] [ext: omni.activity.core-1.0.1] startup
[0.315s] [ext: omni.resourcemonitor-105.0.1] startup
[0.316s] [ext: carb.audio-0.1.0] startup
[0.317s] [ext: usdrt.scenegraph-7.4.8] startup
[0.342s] [ext: omni.kvdb-106.0.20] startup
[0.343s] [ext: omni.timeline-1.0.10] startup
[0.344s] [ext: omni.kit.audiodeviceenum-1.0.1] startup
[0.345s] [ext: omni.hydra.usdrt_delegate-7.4.7] startup
[0.357s] [ext: omni.localcache-106.0.20] startup
[0.358s] [ext: omni.usd-1.11.2] startup
[0.382s] [ext: omni.usdphysics-106.0.20] startup
[0.383s] [ext: omni.convexdecomposition-106.0.20] startup
[0.386s] [ext: omni.kit.usd.layers-2.1.31] startup
[0.391s] [ext: omni.physx.cooking-106.0.20] startup
[0.396s] [ext: omni.datastore-0.0.0] startup
[0.396s] [ext: omni.blobkey-1.1.0] startup
[0.396s] [ext: omni.ujitso.default-1.0.0] startup
[0.398s] [ext: omni.hsscclient-0.0.0] startup
[0.399s] [ext: omni.kit.stage_template.core-1.1.21] startup
[0.400s] [ext: omni.kit.primitive.mesh-1.0.16] startup
[0.403s] [ext: omni.physx-106.0.20] startup
[0.410s] [ext: omni.kit.stage_templates-1.2.3] startup
[0.411s] [ext: omni.physx.stageupdate-106.0.20] startup
[... 53 lines cut ...]
    app_launcher = AppLauncher(headless=True)
  File "/home/xx/workspace/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/app/app_launcher.py", line 115, in __init__
    self._load_extensions()
  File "/home/xx/workspace/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/app/app_launcher.py", line 595, in _load_extensions
    from omni.isaac.core.utils.extensions import enable_extension
  File "/home/xx/anaconda3/envs/RL/lib/python3.10/site-packages/isaacsim/exts/omni.isaac.core/omni/isaac/core/__init__.py", line 11, in <module>
    from omni.isaac.core.physics_context.physics_context import PhysicsContext
  File "/home/xx/anaconda3/envs/RL/lib/python3.10/site-packages/isaacsim/exts/omni.isaac.core/omni/isaac/core/physics_context/__init__.py", line 9, in <module>
    from omni.isaac.core.physics_context.physics_context import PhysicsContext
  File "/home/xx/anaconda3/envs/RL/lib/python3.10/site-packages/isaacsim/exts/omni.isaac.core/omni/isaac/core/physics_context/physics_context.py", line 15, in <module>
    from omni.isaac.core.utils.prims import get_prim_at_path, get_prim_path, is_prim_path_valid
  File "/home/xx/anaconda3/envs/RL/lib/python3.10/site-packages/isaacsim/exts/omni.isaac.core/omni/isaac/core/utils/prims.py", line 19, in <module>
    from omni.isaac.core.utils.semantics import add_update_semantics
  File "/home/xx/anaconda3/envs/RL/lib/python3.10/site-packages/isaacsim/exts/omni.isaac.core/omni/isaac/core/utils/semantics.py", line 13, in <module>
    import Semantics
ModuleNotFoundError: No module named 'Semantics'
2024-10-08 07:45:24 [1,841ms] [Warning] [carb] Recursive unloadAllPlugins() detected!
`

It is strange, since I ran this script before and worked well.
So I follow the error report to the script, and search for solution online.
Then I found a old post https://forums.developer.nvidia.com/t/how-to-use-semantics-utils-in-isaacsim/241275
It shows Semantics module is imported form pxr;
So have I change and problem fix.
But it is strange, why it happened.

Mayankm96, 2024-10-08:
Which version of Isaac Sim are you using? Did you install it from Pip or the binaries?

Toni-SM, 2024-10-08:
From the extension versions It seems to be you are using Isaac Sim version 4.1.0].
Please, update your Isaac Sim to the latest 4.2.0 version

cidxb, 2024-10-09:
> Which version of Isaac Sim are you using? Did you install it from Pip or the binaries?

> From the extension versions It seems to be you are using Isaac Sim version 4.1.0. Please, update your Isaac Sim to the latest 4.2.0 version.

Atctually, I was using  Isaac Sim version 4.2. I have updated right before the error pop out, so I remember it quite well.
I did use  PIP installed it.

cidxb, 2024-10-11:
Actually after reinstall isaac sim, the problem gone, just feeling strange, since the problem just come out of nowhere :)

closed 2024-10-11 by cidxb