isaac-sim/IsaacLab #1189: [Question] Error in reset_idx

chandramohanjagadeesan, 2024-10-08:
I have created a custom environment in isaaclab. i am having an empty point cloud initialized at the beginning in the __init__ function. This point cloud is building up based on the working of the agent. Once the environment is reset I would like to reset the point cloud of the specific environment. I would like to know how the value will be initialised in isaaclab and how should i reset the values?

Thanks

mpgussert, 2024-10-10:
Heya @chandramohanjagadeesan!

Resetting a single environment is done within the specific environment, all of this resetting is usually managed in that environment definition thanks to the scene object.  I'm not sure what you're asking.  Can you please be more specific?  Do you have an example that shows what you're trying to do?

Thanks!

chandramohanjagadeesan, 2024-10-10:
Hi Thanks for the reply,

I have a variable for making a point cloud of the object from each step. Where can i initialize it if I want the variable to be separate for each environment. According to my understanding if I initialize it in __init__ function the same variable will be shared by all the environments and also if I want to reset the environment then the values will not be reset. I am i correct in my understanding ? Is there a place where if i initialize some variables it will make copies of it for all the environments?

kellyguo11, 2024-10-11:
Hi there, can you please clarify if you are following the manager-based or direct workflows? Assuming it's direct, initializing a variable in the task class init will make it a class variable. We generally allocate buffers that hold data for all environments, and when needed for reset, we can index the buffer based on the environment indices that need reset. It's less computationally efficient to have a variable for each environment.