isaac-sim/IsaacLab #1190: [Question] Multi-run/Sweep support for Hydra/Wandb

Chris321sirhc, 2024-10-08:
### Question
Hello everyone,

this may be very easy but is there currently a way to use the hydra functionality for multirun using different configs or even automatically sweep? The problem I encounter when running the `train.py` script from rsl_rl is that the simulator closes after finishing a single task/environment. Ideally, I'd like to use any built-in functionality instead of manually writing a bash script or the likes.

If there is an intended way to achieve this, I'd be delighted to hear about it, thank you in advance!

Kind Regards,
Chris :)

garylvov, 2024-10-09:
Hi Chris,

I am currently working on adding functionality for several runs and hyperparameter tuning with Ray.

You can see my progress here: https://github.com/glvov-bdai/IsaacLab/tree/feature/hyperparam_tune/source/standalone/workflows/ray

I anticipate that by the end of this week I will have a fully functioning prototype, there is a small bug in that link / some missing functionality I am working on.

If you'd like to get a head start, I believe that everything up to and including step 4 currently works.

I plan to PR this when I finish it and hopefully it will become part of core functionality soon ;)

RandomOakForest, 2024-11-22:
Thank you for submitting this. Please let us know may we need to follow up, and if we can close this issue. Thanks.

closed 2024-12-14 by Chris321sirhc