isaac-sim/IsaacLab #1218: [Question] Collision and Instancers

AntoineRichard, 2024-10-11:
### Question

Hello there,

I'm looking into implementing scenes with many objects per environments, and usually in the USD pipeline, the best way to do this is to use a `UsdGeom PointInstancer`. I know that the visualization markers are using this mechanism to instantiate many visual objects.

I modified it so that collision properties only could be passed to the prototypes being instanced, but that led to the physics going bonkers in the sim. Is it possible I did something wrong, and can a point instancer work when only collisions are enabled?

I know that an alternative is to use an `Sdf Change Block` to replicate the behavior of the instancer, but I was hoping there was some other way to do this.

Let me know,

Cheers,

Antoine

mpgussert, 2024-10-17:
Heya Antoine!

Can you provide a short clip of the physics "going bonkers"?  are things just exploding and flying away?  or does it crash with a specific error?  I like the idea of using the point instancer with certain meshes for vectorization

AntoineRichard, 2024-10-18:
Hi Michael!

Here is a small video:
[Dance.webm](https://github.com/user-attachments/assets/e23a31de-d441-4a4e-bf26-fbaffae80f9b)

I did try some extra things by directly tapping into Fabric through USDRT, but couldn't get it to output to the scene.

mpgussert, 2024-10-18:
Is the car is using the point instancer?  Or is it the gates? I think the car should be represented as an articulation https://github.com/isaac-sim/IsaacLab/tree/main/source/extensions/omni.isaac.lab/omni/isaac/lab/assets/articulation

AntoineRichard, 2024-10-21:
The car is using the "Articulation" class! Only the gates were using the instancer. 
I'll see if I can send a small script to reproduce the issue this week!

RandomOakForest, 2024-11-22:
Thanks for posting this. It would be great if you can follow up so the team can address this.

AntoineRichard, 2026-08-27:
Thanks for the report and video, @AntoineRichard, and sorry for the delay. We weren't able to get to the bottom of this at the time, and a lot has changed since. Isaac Lab's `PointInstancer` usage today is visualization-only, and instancers with collision aren't a supported configuration — for many-object scenes, `RigidObjectCollection` (`isaaclab.assets.RigidObjectCollectionCfg`) or cloned rigid objects are the intended route.

Closing as stale — if you still see this on a current release, please reopen with a small repro.

---
*🤖 This issue was reviewed as part of an AI-assisted triage of the Isaac Lab backlog, with automated verification against the current codebase. If we got this wrong, please reopen — we'd rather hear about it!*

closed 2026-08-27 by AntoineRichard