isaac-sim/IsaacLab #1219: [Bug Report] Bug title

StephenWelch, 2024-10-11:
If you are submitting a bug report, please fill in the following details and use the tag [bug].

### Describe the bug

Stateful observation modifiers cannot be added to the observation term config

### Steps to reproduce

Please try to provide a minimal example to reproduce the bug. Error messages and stack traces are also helpful.
Add a `DigitalFilterCfg` to the modifiers list for one of the observation terms in `source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/manager_based/locomotion/velocity/velocity_env_cfg.py`:
```
joint_pos = ObsTerm(
    func=mdp.joint_pos_rel, 
    modifiers=[DigitalFilterCfg(A=[0.0], B=[0.0, 1.0])],
    noise=Unoise(n_min=-0.01, n_max=0.01))
```
Then run training:
```
./isaaclab.sh -p source/standalone/workflows/rsl_rl/train.py --task Isaac-Velocity-Flat-G1-v0
```
This results in the following error:
```
[8.363s] Simulation App Startup Complete
[INFO]: Parsing configuration from: <class 'omni.isaac.lab_tasks.manager_based.locomotion.velocity.config.g1.flat_env_cfg.G1FlatEnvCfg'>
[INFO]: Parsing configuration from: omni.isaac.lab_tasks.manager_based.locomotion.velocity.config.g1.agents.rsl_rl_ppo_cfg:G1FlatPPORunnerCfg
Traceback (most recent call last):
  File "/home/swelch/isaaclab/source/standalone/workflows/rsl_rl/train.py", line 148, in <module>
    main()
  File "/home/swelch/isaaclab/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/utils/hydra.py", line 75, in wrapper
    env_cfg, agent_cfg = register_task_to_hydra(task_name, agent_cfg_entry_point)
  File "/home/swelch/isaaclab/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/utils/hydra.py", line 53, in register_task_to_hydra
    ConfigStore.instance().store(name=task_name, node=cfg_dict)
  File "/home/swelch/miniconda3/envs/isaacsim/lib/python3.10/site-packages/hydra/core/config_store.py", line 85, in store
    cfg = OmegaConf.structured(node)
  File "/home/swelch/miniconda3/envs/isaacsim/lib/python3.10/site-packages/omegaconf/omegaconf.py", line 125, in structured
    return OmegaConf.create(obj, parent, flags)
  File "/home/swelch/miniconda3/envs/isaacsim/lib/python3.10/site-packages/omegaconf/omegaconf.py", line 178, in create
    return OmegaConf._create_impl(
  File "/home/swelch/miniconda3/envs/isaacsim/lib/python3.10/site-packages/omegaconf/omegaconf.py", line 900, in _create_impl
    format_and_raise(node=None, key=None, value=None, msg=str(e), cause=e)
  File "/home/swelch/miniconda3/envs/isaacsim/lib/python3.10/site-packages/omegaconf/_utils.py", line 819, in format_and_raise
    _raise(ex, cause)
  File "/home/swelch/miniconda3/envs/isaacsim/lib/python3.10/site-packages/omegaconf/_utils.py", line 797, in _raise
    raise ex.with_traceback(sys.exc_info()[2])  # set env var OC_CAUSE=1 for full trace
  File "/home/swelch/miniconda3/envs/isaacsim/lib/python3.10/site-packages/omegaconf/omegaconf.py", line 861, in _create_impl
    return DictConfig(
  File "/home/swelch/miniconda3/envs/isaacsim/lib/python3.10/site-packages/omegaconf/dictconfig.py", line 111, in __init__
    format_and_raise(node=None, key=key, value=None, cause=ex, msg=str(ex))
  File "/home/swelch/miniconda3/envs/isaacsim/lib/python3.10/site-packages/omegaconf/_utils.py", line 819, in format_and_raise
    _raise(ex, cause)
  File "/home/swelch/miniconda3/envs/isaacsim/lib/python3.10/site-packages/omegaconf/_utils.py", line 797, in _raise
    raise ex.with_traceback(sys.exc_info()[2])  # set env var OC_CAUSE=1 for full trace
  File "/home/swelch/miniconda3/envs/isaacsim/lib/python3.10/site-packages/omegaconf/dictconfig.py", line 109, in __init__
    self._set_value(content, flags=flags)
  File "/home/swelch/miniconda3/envs/isaacsim/lib/python3.10/site-packages/omegaconf/dictconfig.py", line 647, in _set_value
    raise e
  File "/home/swelch/miniconda3/envs/isaacsim/lib/python3.10/site-packages/omegaconf/dictconfig.py", line 644, in _set_value
    self._set_value_impl(value, flags)
  File "/home/swelch/miniconda3/envs/isaacsim/lib/python3.10/site-packages/omegaconf/dictconfig.py", line 690, in _set_value_impl
    self.__setitem__(k, v)
  File "/home/swelch/miniconda3/envs/isaacsim/lib/python3.10/site-packages/omegaconf/dictconfig.py", line 314, in __setitem__
    self._format_and_raise(key=key, value=value, cause=e)
  File "/home/swelch/miniconda3/envs/isaacsim/lib/python3.10/site-packages/omegaconf/base.py", line 231, in _format_and_raise
    format_and_raise(
  File "/home/swelch/miniconda3/envs/isaacsim/lib/python3.10/site-packages/omegaconf/_utils.py", line 819, in format_and_raise
    _raise(ex, cause)
  File "/home/swelch/miniconda3/envs/isaacsim/lib/python3.10/site-packages/omegaconf/_utils.py", line 797, in _raise
    raise ex.with_traceback(sys.exc_info()[2])  # set env var OC_CAUSE=1 for full trace
  File "/home/swelch/miniconda3/envs/isaacsim/lib/python3.10/site-packages/omegaconf/dictconfig.py", line 308, in __setitem__
    self.__set_impl(key=key, value=value)
[... 259 lines cut ...]
### System Info

Describe the characteristic of your environment:

- Commit: `92dcb8c05642aa890979a5e40a7b5cc9255799d2`
- Isaac Sim Version: `4.1.0-rc.7+4.1.14801.71533b68.gl`
- OS: `Ubuntu 20.04`
- GPU: `NVIDIA GeForce RTX 4060 Laptop GPU`
- CUDA: `12.2`
- GPU Driver: `535.183.01`

### Additional context

N/A

### Checklist

- [ X] I have checked that there is no similar issue in the repo (**required**)
- [ X] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

TBD

StephenWelch, 2024-10-12:
After some digging, I think I identified the root cause of the issue - `configclass` `to_dict()` does not properly convert lists of conficlasses to lists of dicts.

StephenWelch, 2024-10-12:
Here is my solution. I think it could use some cleanup from a more experienced dev, but it works:
```
    # check that input data is class instance
    if not hasattr(obj, "__class__"):
        raise ValueError(f"Expected a class instance. Received: {type(obj)}.")
    # convert object to dictionary
    obj_dict = None
    if isinstance(obj, dict):
        obj_dict = obj
    elif hasattr(obj, '__dict__'):
        obj_dict = obj.__dict__

    if obj_dict is None:
        return obj

    # convert to dictionary
    data = dict()
    for key, value in obj_dict.items():
        # disregard builtin attributes
        if key.startswith("__"):
            continue
        # check if attribute is callable -- function
        if callable(value):
            data[key] = callable_to_string(value)
        # check if attribute is a dictionary
        elif hasattr(value, "__dict__") or isinstance(value, dict):
            data[key] = class_to_dict(value)
        elif isinstance(value, list):
            data[key] = [class_to_dict(v) for v in value]
        else:
            data[key] = value

    return data
```

jtigue-bdai, 2024-10-12:
I have recreated the bug and I am currently investigating. What leads you to believe the `configclass.todict()` is the culprit?

jtigue-bdai, 2024-10-12:
Actually I think I have followed the trail to the same conclusion.

closed 2024-10-24 by Dhoeller19

jtigue-bdai, 2024-10-24:
@StephenWelch, a fix has been merged for this. Thanks for finding the source of the error.

PR #1227 (2024-10-24): Allows configclass `to_dict` operation to handle a list of configclasses
# Description

This PR add in the ability to properly convert configclass to dict if a configclass instance contains a list of configclasses.

Fixes #1219 

## Type of change

- Bug fix (non-breaking change which fixes an issue)

## Checklist

- [x] I have run the [`pre-commit` checks](https://pre-commit.com/) with `./isaaclab.sh --format`
- [x] I have made corresponding changes to the documentation
- [x] My changes generate no new warnings
- [x] I have added tests that prove my fix is effective or that my feature works
- [... cut]
commit 41a9dd44 (2024-10-24): Allows configclass `to_dict` operation to handle a list of configclasses (#1227)

# Description

This PR add in the ability to properly convert configclass to dict if a
configclass instance contains a list of configclasses.

Fixes #1219 

## Type of change

- Bug fix (non-breaking change which fixes an issue)

## Checklist

- [x] I have run the [`pre-commit` checks](https://pre-commit.com/) with
`./isaaclab.sh --format`
- [x] I have made corresponding changes to the documentation
- [x] My changes generate no new warnings
- [x] I have added tests that prove my fix is effective or that my
featur [... cut]