isaac-sim/IsaacLab #1224: [Question] How to import USD files into InteractiveSceneCfg, which attaches the rigid body and deformable body with attachment function in Isaac Sim?

ycsh15151, 2024-10-12:
### Question

Hello there,

I'm working on a project simulating shaking apples. I firstly built the apple along with its stalk and branch in the Isaac Sim, the apple (sphere shape) and the branch (cylinder shape) are rigid body, and the stalk (cylinder mesh) is deformable body. Then, we use attachment function to attach the branch and the stalk, as well as the apple and the stalk. Since the stalk is deformable, the apple can swing after hitting. 

The problem is, when I try to import this usd file into InteractiveSceneCfg from isaac lab for reinforcement learning, I got stuck. The attachment didn't work when I import it into AssetBaseCfg, DeformableObjectCfg or ArticulationCfg, which means the whole structure of usd file fell apart in the simulator.

If there is a way to achieve this, I'd be delighted to hear about it, thank you in advance!

Kind Regards,

Zhian Su :)

mpgussert, 2024-10-18:
Hello @ycsh15151!

Have you tried importing the individual sub components appropriately?  That is, instead of importing the entire asset as one thing, what if you imported it as multiple small things, like the branch, apple, and stalk separately.  This way you could import the apple and branch as RigidObjects  and the Stem as a DeformableObject.  You probably only need Articulation for your robot. 

I hope this helps!
Gus

ycsh15151, 2024-10-28:
> Hello @ycsh15151!
>
> Have you tried importing the individual sub components appropriately? That is, instead of importing the entire asset as one thing, what if you imported it as multiple small [...]

Thanks for your generous reply. 
But if importing them separately, we can not attach the apple and branch with the stem in python, as it seems no function available for attachment between RigidObjects and DeformableObject, this function might only available in GUI of Isaac Sim.

closed 2024-12-02 by ycsh15151