isaac-sim/IsaacLab #1225: [Question] UR10e + RG2 Gripper is not training end-to-end

SabaGhorbani, 2024-10-12:
### Question
Hi everyone,

I've recently started working with IsaacLab, focusing on the management-based code. I created an End-to-End RL solution similar to the Panda robot for object lifting. The robot reaches the object, but the grasping part isn't training properly. I tested the Panda end effector on the UR10e, but it hasn't been successful. I’ve adjusted nearly all the parameters, but it's still not working as effectively as it does with the Panda.

Can anyone help me troubleshoot this issue?

[image]

WangYCheng23, 2024-10-13:
you should adjust ee frame offset.

SabaGhorbani, 2024-10-13:
I have adjusted the end-effector frame, but interestingly, it touches the object and then moves above it, and the end-effector is not closing when it should to function properly. I even replaced the end-effector with another and ensured it was working correctly, but during the best training outcome, the gripper passes through the object and rises above it. In other cases, the gripper closes, touches the cube, and then attempts to move the cube higher.

It's unusual—unlike the training process with the Panda robot, this behavior seems off.

WangYCheng23, 2024-10-14:
In the figure, it appears that your end-effector (EE) frame is positioned above the end-effector itself. You should consider reducing the Z-offset value of the EE frame.

SabaGhorbani, 2024-10-14:
Yes, I understand your point. I've uploaded a video here to help clarify the issue. When I decrease the value, instead of grasping the wall, it just passes through the ball, which is quite puzzling for me. I’ve modified the following code (EE):

        marker_cfg = FRAME_MARKER_CFG.copy()
        marker_cfg.markers["frame"].scale = (0.1, 0.1, 0.1)
        marker_cfg.prim_path = "/Visuals/FrameTransformer"
        self.scene.ee_frame = FrameTransformerCfg(
            prim_path="{ENV_REGEX_NS}/Robot/base_link",
            debug_vis=False,
            visualizer_cfg=marker_cfg,
            target_frames=[
                FrameTransformerCfg.FrameCfg(
                    prim_path="{ENV_REGEX_NS}/Robot/wrist_3_link",
                    name="end_effector",
                    offset=OffsetCfg(
                        pos=[0.0, 0.0, 0.10],
                    ),

[Screencast from 2024-10-14 20-13-13.webm](https://github.com/user-attachments/assets/3a3a817a-7495-40a9-acc0-542fbe82fa2b)

SabaGhorbani, 2024-10-15:
I changed the offset to [0.0, 0.0, -0.50], but the gripper stay very far from cube.

[Screencast from 2024-10-15 21-58-05.webm](https://github.com/user-attachments/assets/cb242906-61e1-4cf1-9a7a-c7a5d69f4557)

WangYCheng23, 2024-10-15:
try tuning the value of the offset gently. However, the best way to find a accurate offset value is to grab the ee frame to a suitable place in isaasim app(click stop button during training)

SabaGhorbani, 2024-10-15:
Thank you very much for the idea. So, you think the main issue is tuning the offset, which should be adjusted very precisely. Actually, I have changed it several times, but I’m not sure how accurate it needs to be. Can it be negative based on your last recommendation? I thought the offset is the distance from the wrist to the middle of the fingers, which I assumed couldn’t be negative.

My second question is: to find this value in IsaacSim, should I remove --headless during training and then stop the simulation? Can I move the robot manually afterward and measure the distances?

WangYCheng23, 2024-10-16:
first, check your 'ee_frame' setting, and then set 'debug vis' to True.

second, remove --headless during training and then stop the simulation. After that, drag and drop '/Visuals/EndEffectorFrameTransformer' to the place you want and remember the offset showing in the simulation.

SabaGhorbani, 2024-10-17:
I think I am doing something wrong because when I drag and drop, nothing happens.
[Screencast from 2024-10-18 10-46-50.webm](https://github.com/user-attachments/assets/eb1409a9-1382-40e4-901f-f859a563760d)

WangYCheng23, 2024-10-18:
1、stop simulation 2、click the frame under the ENdffectorFrameTransformer （carefully check your setting in code）
[image]

your ee frame is above the robot ee. try to low down the frame to the place of gripper.

[image]

SabaGhorbani, 2024-10-18:
I updated ee fram. It seems I only can move the markers:

        marker_cfg = FRAME_MARKER_CFG.copy()
        marker_cfg.markers["frame"].scale = (0.1, 0.1, 0.1)
        marker_cfg.prim_path = "/Visuals/FrameTransformer"
        self.scene.ee_frame = FrameTransformerCfg(
            prim_path="{ENV_REGEX_NS}/Robot/base_link",
            debug_vis=True,
            visualizer_cfg=marker_cfg,
            target_frames=[
                FrameTransformerCfg.FrameCfg(
                    prim_path="{ENV_REGEX_NS}/Robot/wrist_3_link",
                    name="end_effector",
                    offset=OffsetCfg(
                        pos=[0.0, 0.0, 0.1],
                    ),
                ),

[Screencast from 2024-10-18 13-42-55.webm](https://github.com/user-attachments/assets/33e4ef28-49c6-447f-8181-d71d56aec73f)

WangYCheng23, 2024-10-22:
> I updated ee fram. It seems I only can move the markers:
>
>
>
>         marker_cfg = FRAME_MARKER_CFG.copy()
>
>         marker_cfg.markers["frame"].scale = (0.1, 0.1, 0.1)
>
>         marker_cfg.prim_path = [...]

stop the simulation and move it by changing properties

SabaGhorbani, 2024-10-23:
Thank you. I have tried many different options in the properties, but the end-effector does not move. I’m not sure how to solve this issue. I also tested various offsets from -0.5 to 0.5, but it has not improved the situation.

TerenceXue-tech, 2024-10-31:
How did you obtain the UR with gripper resources? Could you please share them?

SabaGhorbani, 2024-10-31:
I made it, I found RG2 urdf and then connected it to the universal robot

TerenceXue-tech, 2024-10-31:
> I made it, I found RG2 urdf and then connected it to the universal robot

Could you please share the USD file via email? I want to try the training. Thanks a lot!

TerenceXue-tech, 2024-10-31:
If convenient, you  could share the code and the USD ifle.

> I made it, I found RG2 urdf and then connected it to the universal robot

RandomOakForest, 2024-11-22:
This is a great discussion. Thanks for posting this. Will close for now. Feel free to begin a new post in our _Discussions_ may we need to follow up on this. Thanks.

closed 2024-11-22 by RandomOakForest