isaac-sim/IsaacLab #1231: [Question] Clarification on Joint Dimensions in Observation Space for Unitree G1 in IsaacLab

XinyuSong123, 2024-10-14:
Hi,

I plan to use a motion dataset to train the Unitree G1 in IsaacLab. I noticed that the observation space for the joints contains 37 dimensions, but I couldn't find in the documentation which dimensions correspond to specific joints. Could you kindly clarify this? Thank you!

mpgussert, 2024-10-18:
Hello @XinyuSong123!

I believe the answers you seek can be found here! https://github.com/isaac-sim/IsaacLab/blob/main/source/extensions/omni.isaac.lab_assets/omni/isaac/lab_assets/unitree.py

specifically line 273 is where the joints are listed for the robot. If you count up the joints (2x for legs, arms, and feet) you get 38... I can't seem to find an explicit mapping anywhere, so I labeled this thread for the dev team.  maybe they can offer insight :D

AsawareeBhide, 2024-11-18:
Hi! Please check the [joint names property](https://isaac-sim.github.io/IsaacLab/main/source/api/lab/omni.isaac.lab.actuators.html#omni.isaac.lab.actuators.ActuatorBase.joint_names).

XinyuSong123, 2024-11-22:
> Hi! Please check the [joint names property](https://isaac-sim.github.io/IsaacLab/main/source/api/lab/omni.isaac.lab.actuators.html#omni.isaac.lab.actuators.ActuatorBase.joint_names).

Hi, I tried to print the relevant attributes in the definition of ```parse_env_cfg``` as shown below:
```
def parse_env_cfg(
    task_name: str, device: str = "cuda:0", num_envs: int | None = None, use_fabric: bool | None = None
) -> ManagerBasedRLEnvCfg | DirectRLEnvCfg:
    """Parse configuration for an environment and override based on inputs.

    Args:
        task_name: The name of the environment.
        device: The device to run the simulation on. Defaults to "cuda:0".
        num_envs: Number of environments to create. Defaults to None, in which case it is left unchanged.
        use_fabric: Whether to enable/disable fabric interface. If false, all read/write operations go through USD.
            This slows down the simulation but allows seeing the changes in the USD through the USD stage.
            Defaults to None, in which case it is left unchanged.

    Returns:
        The parsed configuration object.

    Raises:
        RuntimeError: If the configuration for the task is not a class. We assume users always use a class for the
            environment configuration.
    """
    # load the default configuration
    cfg = load_cfg_from_registry(task_name, "env_cfg_entry_point")

    print(cfg)

    # Access and print joint_names
    print("Joint names in actions.joint_pos:")
    print(cfg.actions.joint_pos.joint_names)

    print("Joint names in robot actuators:")
    for actuator_name, actuator_cfg in cfg.scene.robot.actuators.items():
        print(f"Actuator '{actuator_name}' has joint names expression: {actuator_cfg.joint_names_expr}")

    print("Joint names in robot init_state.joint_pos:")
    print(cfg.scene.robot.init_state.joint_pos.keys())

    # check that it is not a dict
    # we assume users always use a class for the configuration
    if isinstance(cfg, dict):
        raise RuntimeError(f"Configuration for the task: '{task_name}' is not a class. Please provide a class.")

    # simulation device
    cfg.sim.device = device
    # disable fabric to read/write through USD
    if use_fabric is not None:
        cfg.sim.use_fabric = use_fabric
    # number of environments
    if num_envs is not None:
        cfg.scene.num_envs = num_envs

    return cfg
```
The output is as follows:
```
Joint names in actions.joint_pos:
['.*']
Joint names in robot actuators:
Actuator 'legs' has joint names expression: ['.*_hip_yaw_joint', '.*_hip_roll_joint', '.*_hip_pitch_joint', '.*_knee_joint', 'torso_joint']
Actuator 'feet' has joint names expression: ['.*_ankle_pitch_joint', '.*_ankle_roll_joint']
Actuator 'arms' has joint names expression: ['.*_shoulder_pitch_joint', '.*_shoulder_roll_joint', '.*_shoulder_yaw_joint', '.*_elbow_pitch_joint', '.*_elbow_roll_joint', '.*_five_joint', '.*_three_joint', '.*_six_joint', '.*_four_joint', '.*_zero_joint', '.*_one_joint', '.*_two_joint']
Joint names in robot init_state.joint_pos:
dict_keys(['.*_hip_pitch_joint', '.*_knee_joint', '.*_ankle_pitch_joint', '.*_elbow_pitch_joint', 'left_shoulder_roll_joint', 'left_shoulder_pitch_joint', 'right_shoulder_roll_joint', 'right_shoulder_pitch_joint', 'left_one_joint', 'right_one_joint', 'left_two_joint', 'right_two_joint'])
```
It seems I still cannot get the correct order of the joints. Can you give me any clues about it? Thank you!!!

XinyuSong123, 2024-12-09:
I can manually set the order of joints in ObservationsCfg and ActionsCfg

closed 2024-12-09 by XinyuSong123