isaac-sim/IsaacLab #1233: [Bug Report] After switches from `carb` to `omni.log` for logging, using `make gettext` to build the document will report errors

fan-ziqi, 2024-10-15:
If you are submitting a bug report, please fill in the following details and use the tag [bug].

### Describe the bug

After https://github.com/isaac-sim/IsaacLab/pull/1215 , using `make gettext` to build the document will report errors

```
Failed to import omni.isaac.lab.managers.
Possible hints:
* ModuleNotFoundError: No module named 'omni.log'
* AttributeError: module 'omni.isaac.lab' has no attribute 'managers'
Failed to import omni.isaac.lab.markers.
Possible hints:
* AttributeError: module 'omni.isaac.lab' has no attribute 'markers'
* ModuleNotFoundError: No module named 'omni.log'
Failed to import omni.isaac.lab.envs.ui.
Possible hints:
* ModuleNotFoundError: No module named 'omni.log'
* AttributeError: module 'omni.isaac.lab' has no attribute 'envs'
Failed to import omni.isaac.lab.sim.
Possible hints:
* AttributeError: module 'omni.isaac.lab' has no attribute 'sim'
* ModuleNotFoundError: No module named 'omni.log'
Failed to import omni.isaac.lab.sim.utils.
Possible hints:
* AttributeError: module 'omni.isaac.lab' has no attribute 'sim'
* ModuleNotFoundError: No module named 'omni.log'
Failed to import omni.isaac.lab.terrains.
Possible hints:
* ModuleNotFoundError: No module named 'omni.log'
* AttributeError: module 'omni.isaac.lab' has no attribute 'terrains'
Failed to import omni.isaac.lab.terrains.trimesh.
Possible hints:
* ModuleNotFoundError: No module named 'omni.log'
* AttributeError: module 'omni.isaac.lab' has no attribute 'terrains'
Failed to import omni.isaac.lab.terrains.utils.
Possible hints:
* ModuleNotFoundError: No module named 'omni.log'
* AttributeError: module 'omni.isaac.lab' has no attribute 'terrains'
Failed to import omni.isaac.lab.sensors.
Possible hints:
* ModuleNotFoundError: No module named 'omni.log'
* AttributeError: module 'omni.isaac.lab' has no attribute 'sensors'
Failed to import omni.isaac.lab.assets.
Possible hints:
* ModuleNotFoundError: No module named 'omni.log'
* AttributeError: module 'omni.isaac.lab' has no attribute 'assets'
Failed to import omni.isaac.lab.sim.converters.
Possible hints:
* AttributeError: module 'omni.isaac.lab' has no attribute 'sim'
* ModuleNotFoundError: No module named 'omni.log'
Failed to import omni.isaac.lab.sensors.patterns.
Possible hints:
* ModuleNotFoundError: No module named 'omni.log'
* AttributeError: module 'omni.isaac.lab' has no attribute 'sensors'
Failed to import omni.isaac.lab.sim.spawners.
Possible hints:
* AttributeError: module 'omni.isaac.lab' has no attribute 'sim'
* ModuleNotFoundError: No module named 'omni.log'
Failed to import omni.isaac.lab.sim.spawners.shapes.
Possible hints:
* AttributeError: module 'omni.isaac.lab' has no attribute 'sim'
* ModuleNotFoundError: No module named 'omni.log'
Failed to import omni.isaac.lab.sim.spawners.meshes.
Possible hints:
* AttributeError: module 'omni.isaac.lab' has no attribute 'sim'
* ModuleNotFoundError: No module named 'omni.log'
Failed to import omni.isaac.lab.sim.spawners.lights.
Possible hints:
* AttributeError: module 'omni.isaac.lab' has no attribute 'sim'
* ModuleNotFoundError: No module named 'omni.log'
Failed to import omni.isaac.lab.sim.spawners.sensors.
Possible hints:
* AttributeError: module 'omni.isaac.lab' has no attribute 'sim'
* ModuleNotFoundError: No module named 'omni.log'
Failed to import omni.isaac.lab.sim.spawners.from_files.
Possible hints:
* AttributeError: module 'omni.isaac.lab' has no attribute 'sim'
* ModuleNotFoundError: No module named 'omni.log'
Failed to import omni.isaac.lab.sim.spawners.materials.
Possible hints:
* AttributeError: module 'omni.isaac.lab' has no attribute 'sim'
* ModuleNotFoundError: No module named 'omni.log'
Failed to import omni.isaac.lab.sim.spawners.wrappers.
Possible hints:
* AttributeError: module 'omni.isaac.lab' has no attribute 'sim'
* ModuleNotFoundError: No module named 'omni.log'
Failed to import omni.isaac.lab.envs.mdp.
Possible hints:
* ModuleNotFoundError: No module named 'omni.log'
* AttributeError: module 'omni.isaac.lab' has no attribute 'envs'
Failed to import omni.isaac.lab.envs.mdp.observations.
Possible hints:
* ModuleNotFoundError: No module named 'omni.log'
* AttributeError: module 'omni.isaac.lab' has no attribute 'envs'
Failed to import omni.isaac.lab.envs.mdp.actions.
Possible hints:
* ModuleNotFoundError: No module named 'omni.log'
* AttributeError: module 'omni.isaac.lab' has no attribute 'envs'
Failed to import omni.isaac.lab.envs.mdp.actions.actions_cfg.
Possible hints:
* ModuleNotFoundError: No module named 'omni.log'
* AttributeError: module 'omni.isaac.lab' has no attribute 'envs'
Failed to import omni.isaac.lab.envs.mdp.events.
Possible hints:
* ModuleNotFoundError: No module named 'omni.log'
* AttributeError: module 'omni.isaac.lab' has no attribute 'envs'
Failed to import omni.isaac.lab.envs.mdp.commands.
Possible hints:
* ModuleNotFoundError: No module named 'omni.log'
* AttributeError: module 'omni.isaac.lab' has no attribute 'envs'
Failed to import omni.isaac.lab.envs.mdp.commands.commands_cfg.
Possible hints:
* ModuleNotFoundError: No module named 'omni.log'
* AttributeError: module 'omni.isaac.lab' has no attribute 'envs'
Failed to import omni.isaac.lab.envs.mdp.rewards.
Possible hints:
* ModuleNotFoundError: No module named 'omni.log'
* AttributeError: module 'omni.isaac.lab' has no attribute 'envs'
Failed to import omni.isaac.lab.envs.mdp.terminations.
Possible hints:
* ModuleNotFoundError: No module named 'omni.log'
* AttributeError: module 'omni.isaac.lab' has no attribute 'envs'
Failed to import omni.isaac.lab.envs.mdp.curriculums.
Possible hints:
* ModuleNotFoundError: No module named 'omni.log'
* AttributeError: module 'omni.isaac.lab' has no attribute 'envs'
RuntimeError: _ARRAY_API is not PyCapsule object
Failed to import omni.isaac.lab_tasks.utils.
Possible hints:
* ModuleNotFoundError: No module named 'omni.log'
* AttributeError: module 'omni.isaac' has no attribute 'lab_tasks'
Failed to import omni.isaac.lab_tasks.utils.wrappers.
Possible hints:
* ModuleNotFoundError: No module named 'omni.log'
* AttributeError: module 'omni.isaac' has no attribute 'lab_tasks'
Failed to import omni.isaac.lab_tasks.utils.wrappers.rl_games.
Possible hints:
* ModuleNotFoundError: No module named 'omni.log'
* AttributeError: module 'omni.isaac' has no attribute 'lab_tasks'
Failed to import omni.isaac.lab_tasks.utils.wrappers.rsl_rl.
Possible hints:
* ModuleNotFoundError: No module named 'omni.log'
* AttributeError: module 'omni.isaac' has no attribute 'lab_tasks'
Failed to import omni.isaac.lab_tasks.utils.wrappers.skrl.
Possible hints:
* ModuleNotFoundError: No module named 'omni.log'
* AttributeError: module 'omni.isaac' has no attribute 'lab_tasks'
Failed to import omni.isaac.lab_tasks.utils.wrappers.sb3.
Possible hints:
* ModuleNotFoundError: No module named 'omni.log'
* AttributeError: module 'omni.isaac' has no attribute 'lab_tasks'
Failed to import omni.isaac.lab.scene.
Possible hints:
* ModuleNotFoundError: No module named 'omni.log'
* AttributeError: module 'omni.isaac.lab' has no attribute 'scene'
Failed to import omni.isaac.lab_tasks.utils.data_collector.
Possible hints:
* ModuleNotFoundError: No module named 'omni.log'
* AttributeError: module 'omni.isaac' has no attribute 'lab_tasks'
Failed to import omni.isaac.lab.sim.schemas.
Possible hints:
* AttributeError: module 'omni.isaac.lab' has no attribute 'sim'
* ModuleNotFoundError: No module named 'omni.log'
```

### System Info

Describe the characteristic of your environment:

- Commit: [e.g. 8f3b9ca]
- Isaac Sim Version: [e.g. 2022.2.0, this can be obtained by `cat ${ISAACSIM_PATH}/VERSION`]
- OS: [e.g. Ubuntu 20.04]
- GPU: [e.g. RTX 2060 Super]
- CUDA: [e.g. 11.4]
- GPU Driver: [e.g. 470.82.01, this can be seen by using `nvidia-smi` command.]

### Additional context

Add any other context about the problem here.

### Checklist

- [ ] I have checked that there is no similar issue in the repo (**required**)
- [ ] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Criteria 1
- [ ] Criteria 2

Mayankm96, 2024-10-15:
Should have been fixed by #1234

closed 2024-10-15 by Mayankm96

fan-ziqi, 2024-10-15:
Thanks a lot!