isaac-sim/IsaacLab #1242: [Bug Report] Pylance report many errors in vscode, but the code can run properly.

zst1406217, 2024-10-16:
### Describe the bug

Pylance reports many errors in vscode, but the code can run properly. I have done "setup_python_env" to configure vscode, but there is still the problem:
[image]
When I open settings.json in .vscode folder, I find it says: "'python.analysis.extraPaths' cannot be set when a pyrightconfig.json or pyproject.toml is being used." Is this the reason and how can I deal with it?
### Steps to reproduce
Install following the instruction

### System Info

Describe the characteristic of your environment:

- Commit: [e.g. 8f3b9ca]
- Isaac Sim Version: 4.2.0
- OS: Ubuntu 22.04
- GPU: RTX 4090D 
- CUDA: 12.6
- GPU Driver: 560.35.03

### Additional context
None

### Checklist

- I have checked that there is no similar issue in the repo (**required**)
- I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- The way to resolve it.

RandomOakForest, 2024-11-22:
Thanks for posting this. Apologies for the delayed response. This may be resolved in the latest release posted this week. We hope this is something you could try. Let us know if you still face this issue. Thanks.

closed 2025-01-14 by RandomOakForest

Marvelousp4, 2025-03-07:
I also met this issue in the latest version

Marvelousp4, 2025-03-07:
You can add this to pyproject.toml

[tool.pyright]
extraPaths = [
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.app.about",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.app.selector",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.app.setup",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.asset.browser",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.asset.exporter.urdf",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.asset.exporter.urdf/pip_prebundle",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.asset.gen.conveyor",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.asset.gen.conveyor.ui",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.asset.gen.omap",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.asset.gen.omap.ui",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.asset.importer.heightmap",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.benchmark.examples",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.benchmark.services",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.code_editor.jupyter",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.code_editor.vscode",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.core.api",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.core.cloner",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.core.deprecation_manager",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.core.includes",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.core.nodes",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.core.prims",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.core.simulation_manager",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.core.throttling",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.core.utils",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.core.version",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.cortex.behaviors",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.cortex.framework",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.examples.browser",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.examples.extension",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.examples.interactive",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.examples.ui",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.gui.components",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.gui.menu",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.gui.property",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.replicator.behavior",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.replicator.behavior.ui",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.replicator.domain_randomization",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.replicator.examples",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.replicator.scene_blox",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.replicator.synthetic_recorder",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.replicator.writers",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.robot.manipulators",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.robot.manipulators.examples",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.robot.manipulators.ui",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.robot.policy.examples",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.robot.schema",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.robot.surface_gripper",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.robot.surface_gripper.ui",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.robot.wheeled_robots",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.robot.wheeled_robots.ui",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.robot_motion.lula",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.robot_motion.lula/pip_prebundle",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.robot_motion.lula_test_widget",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.robot_motion.motion_generation",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.robot_setup.assembler",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.robot_setup.gain_tuner",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.robot_setup.grasp_editor",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.robot_setup.import_wizard",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.robot_setup.xrdf_editor",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.ros1.bridge",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.ros2.bridge",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.ros2.tf_viewer",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.ros2.urdf",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.sensors.camera",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.sensors.camera.ui",
        "${workspaceFolder}/_isaac_sim/exts/isaacsim.sensors.physics",
[... 552 lines cut ...]
        "${workspaceFolder}/_isaac_sim/extscache/omni.usd.schema.semantics-0.0.0+d02c707b.lx64.r.cp310",
        "${workspaceFolder}/_isaac_sim/extscache/omni.usd.schema.sequence-2.3.1+106.5.0.lx64.r.cp310",
        "${workspaceFolder}/_isaac_sim/extscache/omni.usd_resolver-1.0.0+d02c707b.lx64.r.cp310",
        "${workspaceFolder}/_isaac_sim/extscache/omni.vdb_timesample_editor-0.1.11",
        "${workspaceFolder}/_isaac_sim/extscache/omni.videoencoding-0.1.1+d02c707b.lx64.r.cp310",
[... cut ...]
        "${workspaceFolder}/_isaac_sim/kit/kernel/py",
        "${workspaceFolder}/_isaac_sim/kit/plugins/bindings-python",
        "${workspaceFolder}/_isaac_sim/kit/python/lib/python3.10",
        "${workspaceFolder}/_isaac_sim/kit/python/lib/python3.10/site-packages",
        "${workspaceFolder}/source/isaaclab_mimic",
        "${workspaceFolder}/source/isaaclab",
        "${workspaceFolder}/source/isaaclab_rl",
        "${workspaceFolder}/source/isaaclab_assets",
        "${workspaceFolder}/source/isaaclab_tasks"
    ]

qiaoyo, 2025-05-29:
this is too much.  believe me . we can just add 
`[tool.pyright]
reportMissingImports = true  # 保留其他配置
reportUnusedVariable = true  # 保留其他配置

# 添加这一行来禁用参数类型检查
reportArgumentTypeMismatch = false`
in the file "pyproject.toml".
then run the basic setup again in vscode:
1.  tasks:run tasks,
2. setup_python_env