isaac-sim/IsaacLab #1243: [Question] How to develop isacc sim demo

MinnaZhong, 2024-10-16:
Hi Team,

I am Minna from UFactory. Recently we wanted to implement the same demo with our robotic arm(xArm6), but we didn't know how to configure it. We can do simple movement, and already generated USD file, you can download the file in the link below.
Do you need to configure this from the background? Kindly give us some guidance, thank you in advance.

USD file:
https://drive.google.com/drive/folders/1TEAFwETfudnYOz773uX2a3_cVOpQBkdp?usp=sharing

https://docs.omniverse.nvidia.com/isaacsim/latest/features/environment_setup/assets/usd_assets_robots.html
[image]

Best regards,
Minna

mpgussert, 2024-10-18:
Hello @MinnaZhong !

Sorry for the delayed response in getting back to you! Unfortunately there's little I can do to help with the information you have provided me, other than to assure you that we will better document this process.

When you say that you can do "simple movement and already generated a USD" do you mean that your USD has an articulation setup and you can control it? Are you trying to just tune the joints appropriately?  Or is the issue with controlling it programmatically or with a trained policy like in the demo? What is "simple movement" specifically? Please provide more information so we can help!

Thanks!
Gus

penglongxiang, 2024-10-18:
Hi @mpgussert, thanks for your kind reply. I am also a developer from [UFACTORY](https://github.com/xArm-Developer), we are a robotics product provider and have a large customer base, as requested by our users, as well as part of our development plan, we would like to ask is there a way to request an **official integration support** for our robot models (provided as usd files in links from the initial post), just like Franka, Kinova and Universal Robots, already integrated in Isaac Sim [robot assets](https://docs.omniverse.nvidia.com/isaacsim/latest/features/environment_setup/assets/usd_assets_robots.html). 

We can do the development by ourselves, just curious about how to make our models officially integrated in Isaac sim assets, demos and docs, making it easy for our users who are interested in RL to simulate our manipulators in your great platform. Thank you in advance, looking forward to your reply.

RandomOakForest, 2024-11-22:
Thank you for posting this. Apologies for the delayed response. We'll move this into our _Discussions_ section for the team to follow up. Thanks.