isaac-sim/IsaacLab #1245: [Question] Robot Assemble and Spawn

SonYeongGwang, 2024-10-16:
Hi recently, I'm trying to attach robotiq-2f 140 with UR10 robot.

First, I've try it with 'robot assembler' on the isaac sim workflow. However, I does not work raising error "Missing references found." as figure below. I used both robots from nvidia asset.

[image]
[image]
[image]
[image]

Later I found that, I can replicate the robot assembler by creating fixed joint between the end-effector and gripper base with proper position/orientation. It just looks like the figure below

[image]

After saving the robot in a USD format, I tried to spawn it from environment using isaac lab.

I used 'UsdFileCfg' method inside 'ArticulationCfg'.

However, I got an error:

[image]
[image]

Does anyone knows how I need to do??

Thanks

SantiDiazC, 2024-10-22:
Hi @SonYeongGwang ,
I did it in the past and the solution for this is in #405 (especially when using the interactivescene class)
I hope it help you.

bearpaw, 2024-10-22:
> Hi @SonYeongGwang , I did it in the past and the solution for this is in #405 (especially when using the interactivescene class) I hope it help you.

Hi @SantiDiazC , I am having the same issue. I tried to add dependency to `isaaclab.python.kit` by
```
[dependencies]
# Isaac Sim extensions
"omni.isaac.robot_assembler" = {}
```

but I got the following error:
```bash
2024-10-22 19:31:43 [1,606ms] [Error] [omni.ext.plugin] Failed to resolve extension dependencies. Failure hints:
	[isaaclab.python-1.2.0] dependency: 'omni.isaac.robot_assembler' = { version='^' } can't be satisfied. Available versions:
	 (none found)
 Synced registries:
	 - kit/default         : found 389 packages 
	 - kit/sdk             : found 317 packages 
	 - kit/community       : found 384 packages 
```

I installed isaaclab through [conda and Isaac Sim pip](https://isaac-sim.github.io/IsaacLab/main/source/setup/installation/pip_installation.html). I wasn't able to find `robot_assembler` in extension GUI either. 
[image]

Do you have any suggestions? Thank you.

SantiDiazC, 2024-11-11:
Hi @bearpaw ,

Sorry for late reply. I think I had the same problem when updating to the new version of Isaaclab.
I don't remember well but I think this may resolve your issue:

```python
from omni.isaac.core.utils.extensions import enable_extension
enable_extension("omni.isaac.robot_assembler")
from omni.isaac.robot_assembler import RobotAssembler, AssembledRobot
```
I hope it helps

SonYeongGwang, 2024-11-11:
Thanks @SantiDiazC . Unfortunately I could not solve the problem by with #405. However, it helped me a lot!

I attached two articulated objects using the GUI in Isaac Sim. By importing both into the simulation and connecting them with a fixed joint, I was able to attach the gripper to the manipulator.

@bearpaw, the method mentioned by @SantiDiazC worked for me as well! 

> Hi @bearpaw ,
>
> Sorry for late reply. I think I had the same problem when updating to the new version of Isaaclab. I don't remember well but I think this may resolve your issue:
>
> ```python
> from [...]

RandomOakForest, 2024-11-16:
Thanks for posting this. Could you provide the versions of Isaac Lab that you are working with and gave you this trouble?

SonYeongGwang, 2024-11-19:
@RandomOakForest I'm using Isaac Lab 1.2.0 with Isaac Sim 4.2.0. Ran in Ubuntu 20.04

RandomOakForest, 2024-11-22:
Thank you. We will review and follow up.

RandomOakForest, 2024-12-16:
This may have been fixed with Isaac Lab 1.3.0. If you still see this issue, let us know. Thanks.

closed 2025-01-14 by RandomOakForest