isaac-sim/IsaacLab #1250: [Question] Reset Default Joint Position of Robot with Closed Loop Linkage

evanzijianhe, 2024-10-17:
Hi,

I am trying to include a close loop linkage to a Digit-v3 mode. I’ve built a linkage between the motor and knee joint, and everything works fine in the Isaac Sim GUI. However, when I load the model into the simulation via code and attempt to reset its root state and joint positions after it falls down, the robot ends up flying away.

Does anyone have any suggestions to resolve this issue? Thank you so much. I have attached two videos and code for your reference: 
 
- I have moved the `Articulation Root` to the torso_base link of the robot
- I have marked `Exclude From Ariculation` for the closed-loop linkage joint

Robot works fine in the IsaacSim GUI:

https://github.com/user-attachments/assets/a6e807a6-0619-484f-b63e-f10ad5bdc4e1

Robot falls down, reset and flys away:

https://github.com/user-attachments/assets/03abbff9-5792-4af0-87c1-64481134a7b9

```
def main():
    """Main function."""
    sim = SimulationContext(
        sim_utils.SimulationCfg(
            dt=1/480, 
            device="cuda", 
            gravity=(0.0, 0.0, -9.81),
            physx=PhysxCfg(
                solver_type=0,
                max_position_iteration_count=8,
                max_velocity_iteration_count=1,
                min_position_iteration_count=4,
                min_velocity_iteration_count=1,
                bounce_threshold_velocity=0.2,
            ),
        )
    )
    sim.set_camera_view(eye=[4.0, 2.0, 2.25], target=[0.0, 0.0, 1.0])
    
    cfg = sim_utils.GroundPlaneCfg()
    cfg.func("/World/defaultGroundPlane", cfg)
    cfg = sim_utils.DomeLightCfg(intensity=2000.0, color=(0.75, 0.75, 0.75))
    cfg.func("/World/Light", cfg)
    
    origins = torch.tensor([
        [0.0, 0.0, 0.0],
        [0.0, 2.0, 0.0],
    ]).to(device=sim.device)

    # Robots
    # cassie = Articulation(CASSIE_CFG.replace(prim_path="/World/Cassie"))
    h1 = Articulation(H1_CFG.replace(prim_path="/World/H1"))
    # g1 = Articulation(G1_CFG.replace(prim_path="/World/G1"))
    digit = Articulation(DIGIT_CFG.copy().replace(prim_path="/World/Digit"))
    # digit_urdf = Articulation(DIGIT_URDF_CFG.copy().replace(prim_path="/World/Digit"))
    # digit_v4 = Articulation(DIGITV4_CFG.replace(prim_path="/World/DigitV4"))
    # robots = [cassie, h1, g1]
    robots = [ digit, h1]

    # Play the simulator
    sim.reset()

    # Now we are ready!
    print("[INFO]: Setup complete...")

    # Define simulation stepping
    sim_dt = sim.get_physics_dt()
    sim_time = 0.0
    count = 0
    # Simulate physics
    while simulation_app.is_running():
        # reset
        if count % 500 == 0:
            # reset counters
            sim_time = 0.0
            count = 0
            for index, robot in enumerate(robots):
                # reset dof state
                joint_pos, joint_vel = robot.data.default_joint_pos, robot.data.default_joint_vel
                joint_stiffness, joint_damping = robot.data.default_joint_stiffness, robot.data.default_joint_damping
                robot.write_joint_state_to_sim(joint_pos, joint_vel)
                robot.write_joint_damping_to_sim(joint_damping)
                robot.write_joint_stiffness_to_sim(joint_stiffness)
                root_state = robot.data.default_root_state.clone()
                root_state[:, :3] += origins[index]
                # print(robot.joint_names)
                robot.write_root_state_to_sim(root_state)
                robot.reset()
                
            # reset command
            print(">>>>>>>> Reset!")
        # # apply action to the robot
        for robot in robots:
            # joint_pos_target = robot.data.default_joint_pos + torch.randn_like(robot.data.joint_pos) * 0.1
            joint_pos_target = robot.data.default_joint_pos 
            robot.set_joint_position_target(joint_pos_target)
            robot.write_data_to_sim()
        # perform step
        sim.step()
        # update sim-time
        sim_time += sim_dt
        count += 1
        # update buffers
        for robot in robots:
            robot.update(sim_dt)
```

mpgussert, 2024-10-18:
Heya @evanzijianhe !

Thank you for the question! I always like seeing a break-dancing robot :3 

How do you know this is being caused by the loop linkage?  Is there a specific error you see that makes you think that's the reason? 

I notice in your manual reset you do not reset forces!  When the robot "breaks" does it always tip horizontal like that?  I wonder if there's a lingering normal force or something that is carried through the reset due to the change in orientation. Checkout the reset function in rigid body https://github.com/isaac-sim/IsaacLab/blob/a4118d7f9be827f8755ee71ea922928bb6c4a314/source/extensions/omni.isaac.lab/omni/isaac/lab/assets/rigid_object/rigid_object.py#L98 maybe you're missing this step in the reset?

hope this helps!
Gus

evanzijianhe, 2024-10-18:
Hi Gus,

Thanks so much for your suggestions. I went ahead and check the `reset` function and mu .usd file with the close-loop linkage, as well as referred to the post in the IsaacSIm forun: [Closed Articulation Simulation Problem](https://forums.developer.nvidia.com/t/closed-articulation-simulation-problem/258431?page=2).

I have fixerd the problem. The major issues are from the definition in the close linkage. Here are what I fixed: 
- Applied `Exclude From Articulation` only to one of the joints in the closed-loop linkage (previously, I had applied it to two joints).
- Removed the `Exclude From Articulation` tag from a fixed joint where it was incorrectly applied.
- Tune the initial pose and PD gains. 

Zijian

closed 2024-10-18 by evanzijianhe