isaac-sim/IsaacLab #1252: [Bug Report] Inconsistency detected by ld.so ... Assertion `listp != NULL' failed!

elle-miller, 2024-10-17:
Hi,

Over the past week, I am encountering the following error after simulation start up:

```
[1.207s] [ext: omni.physx.tensors-106.1.9] startup
[1.261s] [ext: omni.isaac.nucleus-0.3.1] startup
[1.262s] [ext: omni.kit.loop-isaac-1.2.0] startup
[1.262s] [ext: omni.isaac.core-3.19.5] startup
[2.478s] [ext: omni.physx.fabric-106.1.9] startup
[2.483s] [ext: omni.isaac.kit-1.13.2] startup
[2.483s] [ext: omni.isaac.cloner-0.8.1] startup
[2.484s] [ext: isaaclab.python.headless-1.2.0] startup
[2.485s] Simulation App Starting
Inconsistency detected by ld.so: ../elf/dl-tls.c: 618: _dl_allocate_tls_init: Assertion `listp != NULL' failed!
```
I have no fix, except to keep entering the same command until it eventually works. Not sure if this is an Isaac Sim or Lab issue.

Thanks!

### System Info

- Commit: 225a8961f5b286af8684263f25e34f198c079f97
- Isaac Sim Version: 4.2.0
- OS: Ubuntu 22.04
- GPU: GeForce RTX 2080
- CUDA: 12.2
- GPU Driver: 535.183.01

RandomOakForest, 2024-11-16:
Thank you for posting this. This is related to an issue currently being addressed by the team.

delenius, 2024-12-23:
I am also getting this, but from some of the isaaclab scripts. For example,
```
$ ./isaaclab.sh -p source/standalone/workflows/rl_games/train.py --task Isaac-Ant-v0 --headless
[INFO] Using python from: /homes/elenius/miniconda3/envs/isaaclab/bin/python                                                         
[INFO][AppLauncher]: Loading experience file: /homes/elenius/repos/IsaacLab/source/apps/isaaclab.python.headless.kit
[Warning] [omni.isaac.kit.simulation_app] Modules: ['omni.kit_app'] were loaded before SimulationApp was started and might not be loaded correctly.
[Warning] [omni.isaac.kit.simulation_app] Please check to make sure no extra omniverse or pxr modules are imported before the call to SimulationApp(...)
Loading user config located at: '/homes/elenius/miniconda3/envs/isaaclab/lib/python3.10/site-packages/omni/data/Kit/Isaac-Sim/4.2/user.config.json'
[Info] [carb] Logging to file: /homes/elenius/miniconda3/envs/isaaclab/lib/python3.10/site-packages/omni/logs/Kit/Isaac-Sim/4.2/kit_20241223_120254.log
2024-12-23 20:02:54 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
Inconsistency detected by ld.so: ../elf/dl-tls.c: 618: _dl_allocate_tls_init: Assertion `listp != NULL' failed!
```
I have isaaclab installed via pip, following the steps in the documentation.