isaac-sim/IsaacLab #1253: [Question] for omni.isaac.sensor, the "position" obtained from get_contact_sensor_raw_data is (0., 0., 0.)

waivivian, 2024-10-17:
[image]
Here is what I get, might I ask what does it particularly mean by "position" in this case?

Another question is I want to obtain the position of the point where the two objects contact, is that possible either in isaac lab or isaac sim, or is this "position" from get_contact_sensor_raw_data is serving for this need? 

Thank you in advance!

mpgussert, 2024-10-18:
Hello @waivivian!

There's a lot going on in this question, least of all the fact that it's an Isaac Sim API question, and not an Isaac lab api question.  However, I think the position returned is the position of the sensor origin in the world frame. I don't know if I can help much more beyond that though.  You may have better luck asking here https://forums.developer.nvidia.com/c/omniverse/simulation

good luck!

closed 2024-10-18 by mpgussert

waivivian, 2024-10-18:
Thanks a lot for your reply! For isaac lab is there anyway to obtain the position of the point where the two objects contact?

AravindhPadmanabhan, 2024-10-25:
@waivivian, I had a similar problem and was helped by others here: [Filtering contacts between bodies](https://github.com/isaac-sim/IsaacLab/issues/364#issuecomment-2107420371) Hope this helps!