isaac-sim/IsaacLab #1254: [Bug Report] RL Games not working in debug mode

robin-vetsch, 2024-10-17:
### Describe the bug

Training of “Isaac-Reach-UR10-v0” environment in debug mode from VS Code with rl_games leads to the error: “exception: no description”. It only works when starting the file over the terminal.

Adjustments in the __init__.py file according [#541 ](https://github.com/isaac-sim/IsaacLab/issues/541) to have not been successful.

### Steps to reproduce

I used the file: 
```
/home/robin/IsaacLab/source/standalone/workflows/rl_games/train.py
```

and I get an error in line:
```
runner.run({"train": True, "play": False, "sigma": train_sigma})
```

here the error:
```
Exception has occurred: AssertionError
exception: no description
  File "/home/robin/IsaacLab/source/standalone/workflows/rl_games/train.py", line 170, in main
    runner.run({"train": True, "play": False, "sigma": train_sigma})
  File "/home/robin/IsaacLab/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/utils/hydra.py", line 91, in hydra_main
    func(env_cfg, agent_cfg, *args, **kwargs)
  File "/home/robin/IsaacLab/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/utils/hydra.py", line 94, in wrapper
    hydra_main()
  File "/home/robin/IsaacLab/source/standalone/workflows/rl_games/train.py", line 178, in <module>
    main()
AssertionError: 
```

-->

### System Info

Describe the characteristic of your environment:

- Isaac Sim Version: 4.2.0
- OS: Ubuntu 20.04
- GPU: RTX A4000
- CUDA: 12.2
- GPU Driver: 535.183.01

### Additional context

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [ ] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Training of envs with rl_games is also working in debug mode.

RandomOakForest, 2024-11-16:
Thanks for submitting this. Can you confirm the Isaac Lab version you used is that of the latest release? or provide the commit you used? Thanks.

robinvetsch, 2024-11-21:
Hi,

I have installed IsaacLab with the newest version on a new Workstation, but there is still the same issue.

Bdl-1989, 2025-03-12:
[image]
i met this as well once i change the num_envs ( InteractiveSceneCfg(num_envs=1, env_spacing) from 4096 to small int.

sim is 4.5  rl_game Isaac-Cart-Double-Pendulum-Direct-v0

Toni-SM, 2025-03-12:
> i met this as well once i change the num_envs ( InteractiveSceneCfg(num_envs=1, env_spacing) from 4096 to small int.

The `num_envs` in rl_games cannot have an arbitrary value. It needs to be changed following a specific rule (see https://github.com/isaac-sim/OmniIsaacGymEnvs/blob/main/docs/troubleshoot.md#rl-training)

StafaH, 2026-09-20:
Thanks @robinvetsch for reporting the RL-Games failure under the VS Code debugger.

The original `source/standalone/workflows/rl_games/train.py` workflow has since been replaced, and debug attachment is now supported through `debugpy` following https://github.com/isaac-sim/IsaacLab/pull/4863. We verified the current `Isaac-Reach-UR10` task under `debugpy`: RL-Games completed a training epoch and saved its checkpoint successfully.

The empty `AssertionError` reported later with a small `num_envs` value is a separate RL-Games batching constraint: `horizon_length × num_envs × num_agents` must be divisible by `minibatch_size`.

While validating this, we found that the current documented attach command is missing the `debugpy` dependency after the packaging migration. That follow-up is tracked separately in #7915.

We are closing this original debugger-specific report as resolved. Please reopen it if RL-Games still fails under the current `debugpy` workflow with the default task configuration.

---
*🤖 This issue was reviewed as part of an AI-assisted triage of the Isaac Lab backlog, with automated verification against the current codebase. If we got this wrong, please reopen — we'd rather hear about it!*

closed 2026-09-20 by StafaH