isaac-sim/IsaacLab #1255: HTTP or TCP/IP connection crash

Samael0666, 2024-10-17:
I am trying to connect the isaac sim where i have created a simulation for warehouse and i want the lights in there by either HTTP or TCP/IP connection. But whenever i run the code in the script editor in isaac sim for litening to ports the isaac sim itself crashes and leaving me with no choice but to restart the application.

Samael0666, 2024-10-17:
https://github.com/user-attachments/assets/3b225f20-809d-4715-9fe3-54d4f0090aef

Mayankm96, 2024-10-17:
This issue is Isaac Sim specific. Please report this on NVIDIA forums instead: https://forums.developer.nvidia.com/c/omniverse/simulation/69

This repository is mainly for Isaac Lab related issues only.

closed 2024-10-17 by Mayankm96