isaac-sim/IsaacLab #1258: [Bug Report] Error using MultiAssetSpawnerCfg

chandramohanjagadeesan, 2024-10-17:
### Describe the bug

I used MultiAssetSpawnerCfg to import objects from usd file. When the environment gets reset in super()._reset_idx(env_ids). there is a following error. This error occurs in reset in directRL environment

### Steps to reproduce

Cfg in the interactive sceneCfg 

object: RigidObjectCfg =  RigidObjectCfg(
        prim_path="/World/envs/env_.*/Object",
        spawn=sim_utils.MultiAssetSpawnerCfg(
            assets_cfg=[
                sim_utils.UsdFileCfg(
                    usd_path=f"{dir_path}/Assets/Objects/USD/object_0.usd",
                    scale = [1.0, 1.0, 1.0],
                ),
                sim_utils.UsdFileCfg(
                    usd_path=f"{dir_path}/Assets/Objects/USD/object_1.usd",
                    scale = [1.0, 1.0, 1.0],
                ),
                sim_utils.UsdFileCfg(
                    usd_path=f"{dir_path}/Assets/Objects/USD/object_2.usd",
                    scale = [1.0, 1.0, 1.0],
                ),
                sim_utils.UsdFileCfg(
                    usd_path=f"{dir_path}/Assets/Objects/USD/object_3.usd",
                    scale = [1.0, 1.0, 1.0],
                )
            ],
            random_choice=True,
            rigid_props=sim_utils.RigidBodyPropertiesCfg(
                solver_position_iteration_count=4, solver_velocity_iteration_count=0
            ),
            mass_props=sim_utils.MassPropertiesCfg(mass=1.0),
            collision_props=sim_utils.CollisionPropertiesCfg(),
        ),
        init_state=RigidObjectCfg.InitialStateCfg(pos=(0.5, 0.0, 0.0), rot=(0.0, -0.0, 0.0, -0.0)),
        )

TuluhanAkbulut, 2024-10-19:
I am getting the same error with the ManagerbasedRL environments too. Here, if I spawn blue
usd_path=[

    # self.asset_config_path + "Blocks/DexCube/dex_cube_instanceable.usd",
    # self.asset_config_path + "Blocks/blue_block.usd",
    # self.asset_config_path + "Blocks/green_block.usd",
    # self.asset_config_path + "Blocks/red_block.usd",
    # self.asset_config_path + "Blocks/yellow_block.usd",
    # self.asset_config_path + "Mugs/SM_Mug_A2.usd",
    # self.asset_config_path + "Mugs/SM_Mug_B1.usd",
],

If usd path consists of red, blue, green, and yellow blocks, it works fine. If dex_cube is alone in the list, it works fine too. However, if it consists dex_cube with other blocks it throws the CUDA error. For SM_Mugs it is saying that AttributeError: 'RigidObject' object has no attribute '_external_force_b' which may be unrelated.

chandramohanjagadeesan, 2024-10-19:
Hi @TuluhanAkbulut 

The issue is in reset method in RigidObject.py file. 

It is caused by how self._external_force_b is indexed with env_ids to set value 0.0. I found a work around to make it work. You can change the lines 
        # reset external wrench
        self._external_force_b[env_ids] = 0.0
        self._external_torque_b[env_ids] = 0.0
to 
# reset external wrench
        self._external_force_b[0,0,env_ids] = 0.0
        self._external_torque_b[0,0,env_ids] = 0.0
This makes it work but I have to test if this change has any implications on other functions. 

Error occurs because env_ids is of shape tensor([3]) and self._external_force_b[env_ids] is of shape tensor[1,1,3]. Hope this helps till the actual solution comes up.

For note : @Mayankm96. I am tagging you because I found that you developed this feature from the past issues and I would like to know if this solution would work. Thanks in advance for your inputs.

TuluhanAkbulut, 2024-10-21:
Thanks for the reply, it did not work for me. It actually made it worse by throwing the same error even for only blue_block in the list, which was not happening before. Here, the related code and error message:

I am trying this method with Omniverse-nucleus objects at the moment:
```
self.asset_config_path = http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/4.2/Isaac/Props/

self.scene.object2 = RigidObjectCfg(
    prim_path="{ENV_REGEX_NS}/Object2",
    spawn=sim_utils.MultiAssetSpawnerCfg(
    assets_cfg=[
        # sim_utils.UsdFileCfg(
        # usd_path=self.asset_config_path + "Blocks/DexCube/dex_cube_instanceable.usd",
        # scale = [0.8, 0.8, 0.8],
        # ),
        sim_utils.UsdFileCfg(
        usd_path=self.asset_config_path + "Blocks/blue_block.usd",
        scale = [1.0, 1.0, 1.0],
        ),
        # sim_utils.UsdFileCfg(
        # usd_path=self.asset_config_path + "Blocks/red_block.usd",
        # scale = [0.7, 0.7, 0.7],
        # ),
    ],
    random_choice=False,
    rigid_props=sim_utils.RigidBodyPropertiesCfg(
        solver_position_iteration_count=4, solver_velocity_iteration_count=0
    ),
    mass_props=sim_utils.MassPropertiesCfg(mass=1.0),
    collision_props=sim_utils.CollisionPropertiesCfg(),
),
init_state=RigidObjectCfg.InitialStateCfg(pos=[0.8, 0.2, 0.1], rot=[1, 0, 0, 0]),
)
```

Error message:

../aten/src/ATen/native/cuda/IndexKernel.cu:92: operator(): block: [0,0,0], thread: [3,0,0] Assertion `-sizes[i] <= index && index < sizes[i] && "index out of bounds"` failed.
../aten/src/ATen/native/cuda/IndexKernel.cu:92: operator(): block: [0,0,0], thread: [4,0,0] Assertion `-sizes[i] <= index && index < sizes[i] && "index out of bounds"` failed.
../aten/src/ATen/native/cuda/IndexKernel.cu:92: operator(): block: [0,0,0], thread: [5,0,0] Assertion `-sizes[i] <= index && index < sizes[i] && "index out of bounds"` failed.
../aten/src/ATen/native/cuda/IndexKernel.cu:92: operator(): block: [0,0,0], thread: [6,0,0] Assertion `-sizes[i] <= index && index < sizes[i] && "index out of bounds"` failed.
../aten/src/ATen/native/cuda/IndexKernel.cu:92: operator(): block: [0,0,0], thread: [7,0,0] Assertion `-sizes[i] <= index && index < sizes[i] && "index out of bounds"` failed.
../aten/src/ATen/native/cuda/IndexKernel.cu:92: operator(): block: [0,0,0], thread: [8,0,0] Assertion `-sizes[i] <= index && index < sizes[i] && "index out of bounds"` failed.
../aten/src/ATen/native/cuda/IndexKernel.cu:92: operator(): block: [0,0,0], thread: [9,0,0] Assertion `-sizes[i] <= index && index < sizes[i] && "index out of bounds"` failed.
../aten/src/ATen/native/cuda/IndexKernel.cu:92: operator(): block: [0,0,0], thread: [10,0,0] Assertion `-sizes[i] <= index && index < sizes[i] && "index out of bounds"` failed.
../aten/src/ATen/native/cuda/IndexKernel.cu:92: operator(): block: [0,0,0], thread: [11,0,0] Assertion `-sizes[i] <= index && index < sizes[i] && "index out of bounds"` failed.
../aten/src/ATen/native/cuda/IndexKernel.cu:92: operator(): block: [0,0,0], thread: [12,0,0] Assertion `-sizes[i] <= index && index < sizes[i] && "index out of bounds"` failed.
../aten/src/ATen/native/cuda/IndexKernel.cu:92: operator(): block: [0,0,0], thread: [13,0,0] Assertion `-sizes[i] <= index && index < sizes[i] && "index out of bounds"` failed.
../aten/src/ATen/native/cuda/IndexKernel.cu:92: operator(): block: [0,0,0], thread: [14,0,0] Assertion `-sizes[i] <= index && index < sizes[i] && "index out of bounds"` failed.
../aten/src/ATen/native/cuda/IndexKernel.cu:92: operator(): block: [0,0,0], thread: [15,0,0] Assertion `-sizes[i] <= index && index < sizes[i] && "index out of bounds"` failed.
../aten/src/ATen/native/cuda/IndexKernel.cu:92: operator(): block: [0,0,0], thread: [16,0,0] Assertion `-sizes[i] <= index && index < sizes[i] && "index out of bounds"` failed.
../aten/src/ATen/native/cuda/IndexKernel.cu:92: operator(): block: [0,0,0], thread: [17,0,0] Assertion `-sizes[i] <= index && index < sizes[i] && "index out of bounds"` failed.
../aten/src/ATen/native/cuda/IndexKernel.cu:92: operator(): block: [0,0,0], thread: [18,0,0] Assertion `-sizes[i] <= index && index < sizes[i] && "index out of bounds"` failed.
../aten/src/ATen/native/cuda/IndexKernel.cu:92: operator(): block: [0,0,0], thread: [19,0,0] Assertion `-sizes[i] <= index && index < sizes[i] && "index out of bounds"` failed.
../aten/src/ATen/native/cuda/IndexKernel.cu:92: operator(): block: [0,0,0], thread: [20,0,0] Assertion `-sizes[i] <= index && index < sizes[i] && "index out of bounds"` failed.
../aten/src/ATen/native/cuda/IndexKernel.cu:92: operator(): block: [0,0,0], thread: [21,0,0] Assertion `-sizes[i] <= index && index < sizes[i] && "index out of bounds"` failed.
../aten/src/ATen/native/cuda/IndexKernel.cu:92: operator(): block: [0,0,0], thread: [22,0,0] Assertion `-sizes[i] <= index && index < sizes[i] && "index out of bounds"` failed.
../aten/src/ATen/native/cuda/IndexKernel.cu:92: operator(): block: [0,0,0], thread: [23,0,0] Assertion `-sizes[i] <= index && index < sizes[i] && "index out of bounds"` failed.
../aten/src/ATen/native/cuda/IndexKernel.cu:92: operator(): block: [0,0,0], thread: [24,0,0] Assertion `-sizes[i] <= index && index < sizes[i] && "index out of bounds"` failed.
../aten/src/ATen/native/cuda/IndexKernel.cu:92: operator(): block: [0,0,0], thread: [25,0,0] Assertion `-sizes[i] <= index && index < sizes[i] && "index out of bounds"` failed.
[... cut ...]
2024-10-21 01:51:00 [6,288ms] [Error] [omni.physx.plugin] PhysX error: Failed to unload CUDA module data, returned 710., FILE /builds/omniverse/physics/physx/source/cudamanager/src/CudaContextManager.cpp, LINE 817
2024-10-21 01:51:00 [6,289ms] [Error] [omni.physx.plugin] PhysX error: Failed to unload CUDA module data, returned 710., FILE /builds/omniverse/physics/physx/source/cudamanager/src/CudaContextManager.cpp, LINE 817
2024-10-21 01:51:00 [6,289ms] [Error] [omni.physx.plugin] PhysX error: Failed to unload CUDA module data, returned 710., FILE /builds/omniverse/physics/physx/source/cudamanager/src/CudaContextManager.cpp, LINE 817

RandomOakForest, 2024-11-16:
Thanks for submitting this. If you are still facing this issue, could you provide the versions of Isaac Sim and Isaac Lab you are using? Thanks.

matthewtrepte, 2026-10-19:
This is from a whileback: Isaac Lab 1.x. Please retry on develop with replicate_physics=False and a {ENV_REGEX_NS}/Object prim path. Closing as stale for now; reopen with if it still fails.

closed 2026-10-19 by matthewtrepte