isaac-sim/IsaacLab #1259: [Question] How to create static objects for ray caster sensor?

adrian-sch, 2024-10-17:
### Question

Im trying to configure a robot with a Lidar. For this I use the ray caster sensor. The Documentation states that currently only static objects are supported, but I dont understand how to define an static object for it to work. 

I modyfied the `04_sensors/run_ray_caster.py` example to add an static object `/World/cube` and replaced the groundplane in the `mesh_prim_paths` of the RayCasterCfg with the prim to the new object.

design_scene():
```
def design_scene() -> dict:
    """Design the scene."""
    # Populate scene
    # -- Rough terrain
    cfg = sim_utils.UsdFileCfg(usd_path=f"{ISAAC_NUCLEUS_DIR}/Environments/Terrains/rough_plane.usd")
    cfg.func("/World/ground", cfg)
    # -- Light
    cfg = sim_utils.DistantLightCfg(intensity=2000)
    cfg.func("/World/light", cfg)

    cone_spawn_cfg = sim_utils.CuboidCfg(
        size=(10.0, 10.0, 0.1),
        collision_props=sim_utils.CollisionPropertiesCfg(),
        visual_material=sim_utils.PreviewSurfaceCfg(diffuse_color=(0.0, 1.0, 0.0)),
    )
    cone_spawn_cfg.func(
        "/World/cube", cone_spawn_cfg, translation=(5.0, 5.0, 0.2), orientation=(1.0, 0.0, 0.0, 0.0)
    )

    # Create separate groups called "Origin1", "Origin2", "Origin3"
    # Each group will have a robot in it
    origins = [[0.25, 0.25, 0.0], [-0.25, 0.25, 0.0], [0.25, -0.25, 0.0], [-0.25, -0.25, 0.0]]
    for i, origin in enumerate(origins):
        prim_utils.create_prim(f"/World/Origin{i}", "Xform", translation=origin)
    # -- Balls
    cfg = RigidObjectCfg(
        prim_path="/World/Origin.*/ball",
        spawn=sim_utils.SphereCfg(
            radius=0.25,
            rigid_props=sim_utils.RigidBodyPropertiesCfg(),
            mass_props=sim_utils.MassPropertiesCfg(mass=0.5),
            collision_props=sim_utils.CollisionPropertiesCfg(),
            visual_material=sim_utils.PreviewSurfaceCfg(diffuse_color=(0.0, 0.0, 1.0)),
        ),
    )
    balls = RigidObject(cfg)
    # -- Sensors
    ray_caster = define_sensor()

    # return the scene information
    scene_entities = {"balls": balls, "ray_caster": ray_caster}
    return scene_entities
```
    
define_sensor():
```
def define_sensor() -> RayCaster:
    """Defines the ray-caster sensor to add to the scene."""
    # Create a ray-caster sensor
    ray_caster_cfg = RayCasterCfg(
        prim_path="/World/Origin.*/ball",
        mesh_prim_paths=["/World/cube"],
        pattern_cfg=patterns.GridPatternCfg(resolution=0.1, size=(2.0, 2.0)),
        attach_yaw_only=True,
        debug_vis=not args_cli.headless,
    )
    ray_caster = RayCaster(cfg=ray_caster_cfg)

    return ray_caster
```

The script chrashes on startup with this stacktrace:

```
Traceback (most recent call last):
  File "/home/adrian/Projects/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/sensors/sensor_base.py", line 64, in <lambda>
    lambda event, obj=weakref.proxy(self): obj._initialize_callback(event),
  File "/home/adrian/Projects/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/sensors/sensor_base.py", line 271, in _initialize_callback
    self._initialize_impl()
  File "/home/adrian/Projects/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/sensors/ray_caster/ray_caster.py", line 153, in _initialize_impl
    self._initialize_warp_meshes()
  File "/home/adrian/Projects/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/sensors/ray_caster/ray_caster.py", line 183, in _initialize_warp_meshes
    raise RuntimeError(f"Invalid mesh prim path: {mesh_prim_path}")
RuntimeError: Invalid mesh prim path: /World/cube
2024-10-17 15:15:42 [9,671ms] [Warning] [omni.hydra.scene_delegate.plugin] Calling getBypassRenderSkelMeshProcessing for prim /Visuals/RayCaster.proto0_mesh_id0 that has not been populated
Traceback (most recent call last):
  File "/home/adrian/Projects/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/sensors/sensor_base.py", line 170, in <lambda>
    lambda event, obj=weakref.proxy(self): obj._debug_vis_callback(event)
  File "/home/adrian/Projects/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/sensors/ray_caster/ray_caster.py", line 283, in _debug_vis_callback
    self.ray_visualizer.visualize(self._data.ray_hits_w.view(-1, 3))
AttributeError: 'NoneType' object has no attribute 'view'
Traceback (most recent call last):
  File "/home/adrian/Projects/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/sensors/sensor_base.py", line 170, in <lambda>
    lambda event, obj=weakref.proxy(self): obj._debug_vis_callback(event)
  File "/home/adrian/Projects/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/sensors/ray_caster/ray_caster.py", line 283, in _debug_vis_callback
    self.ray_visualizer.visualize(self._data.ray_hits_w.view(-1, 3))
AttributeError: 'NoneType' object has no attribute 'view'
Traceback (most recent call last):
  File "/home/adrian/Projects/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/sensors/sensor_base.py", line 170, in <lambda>
    lambda event, obj=weakref.proxy(self): obj._debug_vis_callback(event)
  File "/home/adrian/Projects/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/sensors/ray_caster/ray_caster.py", line 283, in _debug_vis_callback
    self.ray_visualizer.visualize(self._data.ray_hits_w.view(-1, 3))
AttributeError: 'NoneType' object has no attribute 'view'
Traceback (most recent call last):
  File "/home/adrian/Projects/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/sensors/sensor_base.py", line 170, in <lambda>
    lambda event, obj=weakref.proxy(self): obj._debug_vis_callback(event)
  File "/home/adrian/Projects/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/sensors/ray_caster/ray_caster.py", line 283, in _debug_vis_callback
    self.ray_visualizer.visualize(self._data.ray_hits_w.view(-1, 3))
AttributeError: 'NoneType' object has no attribute 'view'
[INFO]: Setup complete...
Traceback (most recent call last):
  File "/home/adrian/Projects/IsaacLab/source/standalone/tutorials/04_sensors/run_ray_caster.py", line 157, in <module>
    main()
  File "/home/adrian/Projects/IsaacLab/source/standalone/tutorials/04_sensors/run_ray_caster.py", line 152, in main
    run_simulator(sim=sim, scene_entities=scene_entities)
  File "/home/adrian/Projects/IsaacLab/source/standalone/tutorials/04_sensors/run_ray_caster.py", line 123, in run_simulator
    ray_caster.reset()
  File "/home/adrian/Projects/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/sensors/ray_caster/ray_caster.py", line 119, in reset
    self.drift[env_ids].uniform_(*self.cfg.drift_range)
AttributeError: 'RayCaster' object has no attribute 'drift'
```

mpgussert, 2024-10-17:
Hello Adrian! 

I think this is happening because of your use of the "cuboid" which is a shape, not a mesh.

`  File "/home/adrian/Projects/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/sensors/ray_caster/ray_caster.py", line 183, in _initialize_warp_meshes
    raise RuntimeError(f"Invalid mesh prim path: {mesh_prim_path}")
RuntimeError: Invalid mesh prim path: /World/cube`

If you use a mesh (that you aren't going to move), does the error change?

thanks!
Gus

adrian-sch, 2024-10-18:
I guess I found the issue. The ray caster initializes before the Cuboid is moved by the spawn command. As seen in the videos. Is it possible to move the mesh once before the initlization of the sensor? 

Cuboid tilted to move the ball:

[Bildschirmaufzeichnung vom 18.10.2024, 11:10:58.webm](https://github.com/user-attachments/assets/2ea7a09b-c8ab-46b3-94cc-70fc267034a0)

When i add a second Cuboid and dont move it while spawning you can see the edge where the scans go to infinity.

```
    cone_spawn_cfg = sim_utils.MeshCuboidCfg(
        size=(10.0, 10.0, 0.1),
        collision_props=sim_utils.CollisionPropertiesCfg(),
        visual_material=sim_utils.PreviewSurfaceCfg(diffuse_color=(0.0, 1.0, 0.0)),
    )
    cone_spawn_cfg.func(
        "/World/cube", cone_spawn_cfg, translation=(5.0, 5.0, 0.5), orientation=(0, 0.0436194, 0, 0.9990482)
    )
    cone2_spawn_cfg = sim_utils.MeshCuboidCfg(
        size=(10.0, 10.0, 0.1),
        collision_props=sim_utils.CollisionPropertiesCfg(),
        visual_material=sim_utils.PreviewSurfaceCfg(diffuse_color=(0.0, 0.0, 1.0)),
    )
    cone2_spawn_cfg.func(
        "/World/cube2", cone2_spawn_cfg
    )
```
[Bildschirmaufzeichnung vom 18.10.2024, 11:24:03.webm](https://github.com/user-attachments/assets/4b83f678-e476-4a8b-a44d-d1a3eb3ce4b7)

mpgussert, 2024-10-18:
I don't think that's possible currently (to move the sensor post initialization and have it function correctly), but I'm also still learning the lab codebase myself.

closed 2024-10-18 by mpgussert