isaac-sim/IsaacLab #1261: [Question] How to use Spot High-Level API in IsaacSim/Lab?

Jaraxxus-Me, 2024-10-17:
### Question

Thanks for the great example about velocity control and low-level joint control for Spot.
Since we don't do low-level control on real Spot, may I ask if it is possible to control the Spot in IsaacSim with High Level Spot API used in BD [examples](https://github.com/boston-dynamics/spot-sdk/tree/master/python/examples)?

Do we have that already somewhere in IsaacLab or we will need to implement custom equivalent low-level controllers?

Thanks!

mpgussert, 2024-10-17:
Thank you for the suggestion!  We will take this into consideration :) We want to avoid adding too much support for specific robots because we want the platform itself to motivate the creation of that kind of thing. However, It would be super useful if there were some way for users to "pass through" commands from higher APIs automagically. Maybe if we created some kind of base "translator" class that users could then customize to do that conversion? just some thoughts!

Jaraxxus-Me, 2024-10-18:
Thanks for the quick reply! Yes, a higher-level command interface would be very helpful, especially for Spot, as we usually can't do low-level control. In a real deployment, we currently have to directly use the BD examples instead of verifying that in simulation first.

Mayankm96, 2024-10-18:
I believe Boston Dynamics provide a research kit with the deployment code: https://bostondynamics.com/reinforcement-learning-researcher-kit/

I advise you to reach out to them to gain access to writing and deploying a learned controller.