isaac-sim/IsaacLab #1263: [Bug Report] Potential implementation error in `quat_inv()`

ozhanozen, 2024-10-17:
Hello everyone,

The current implementation of the `quat_inv()` is as the following:

https://github.com/isaac-sim/IsaacLab/blob/a4118d7f9be827f8755ee71ea922928bb6c4a314/source/extensions/omni.isaac.lab/omni/isaac/lab/utils/math.py#L238-L247
where `normalize()` is:
https://github.com/isaac-sim/IsaacLab/blob/a4118d7f9be827f8755ee71ea922928bb6c4a314/source/extensions/omni.isaac.lab/omni/isaac/lab/utils/math.py#L81-L91

I believe the correct quaternion inverse should include a division with the norm-squared, not the norm. If the input quaternion is not a unit quaternion, the current implementation would give a wrong result. If it is assumed to be a unit quaternion, then we could simply return `quat_conjugate(q)` without normalizing it again. 

I suggest we change the implementation to something like:

```
def quat_inv(q: torch.Tensor, eps: float = 1e-9) -> torch.Tensor:
    """Computes the inverse of a quaternion.

    Args:
        q (torch.Tensor): The quaternion orientation in (w, x, y, z). Shape is (N, 4).
        eps (float): A small value to avoid division by zero. Defaults to 1e-9.

    Returns:
        torch.Tensor: Inverse quaternion in (w, x, y, z). Shape is (N, 4).
    """
    return quat_conjugate(q) / q.pow(2).sum(dim=-1, keepdim=True).clamp(min=eps)
```

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

ozhanozen, 2024-12-17:
@mpgussert Shall I create a PR to address this until the global definitions are adopted?

Mayankm96, 2025-03-25:
From what I see here: https://ch.mathworks.com/help/nav/ref/quaternion.normalize.html

The current implementation seems correct.

ozhanozen, 2025-03-25:
Hi @Mayankm96, `normalize()` is fine, but `quat_inv()` is incorrect. 

The quarternion inverse should be as the following (see [MathWorks reference](https://ch.mathworks.com/help/aeroblks/quaternioninverse.html)):

`q^{-1} = [w, -x, -y, -z] / ( w^2 + x^2 + y^2 + z^2 )`,

whereas the Isaac Lab implementation effectively does:

`q^{-1} = [w, -x, -y, -z] / sqrt( w^2 + x^2 + y^2 + z^2 )`,

placing the denominator inside the square root. That’s why it’s wrong — you need the squared norm in the denominator, not its square root.

jtigue-bdai, 2025-06-27:
A couple of other sources:
https://en.wikipedia.org/wiki/Quaternion
https://graphics.stanford.edu/courses/cs348a-17-winter/Papers/quaternion.pdf

PR #2797 (2025-06-28): Fixes the implementation of `quat_inv()`
# Description

Corrects `quat_inv()` in utils/math.py: the inverse is now computed as:

`quat_conjugate(q) / q.pow(2).sum(dim=-1, keepdim=True).clamp(min=eps)`

ensuring correct results for **non-unit** quaternions.  

Fixes #1263 (see discussion for details). Also updated `CHANGELOG.rst` and corrected a few minor typos within.

## Type of change

- Bug fix (non-breaking change which fixes an issue)
- This change requires a documentation update

## Checklist

- [x] I have run the [`pre-commit` checks](https://pre-commit.com/) with `./isaaclab.sh --forma [... cut]
commit 3d6f55b9 (2025-06-28): Fixes the implementation of `quat_inv()` (#2797)

# Description

Corrects `quat_inv()` in utils/math.py: the inverse is now computed as:

`quat_conjugate(q) / q.pow(2).sum(dim=-1, keepdim=True).clamp(min=eps)`

ensuring correct results for **non-unit** quaternions.  

Fixes #1263 (see discussion for details). Also updated `CHANGELOG.rst`
and corrected a few minor typos within.

## Type of change

- Bug fix (non-breaking change which fixes an issue)
- This change requires a documentation update

## Checklist

- [x] I have run the [`pre-commit` checks](https://pre-commit.com/) with
`./isaaclab.s [... cut]
closed 2025-06-28 by kellyguo11