isaac-sim/IsaacLab #1265: [Bug Report] Error and Black Render Window When Running Isaac Lab Installation Guide

louislelay, 2024-10-17:
### Describe the bug

While following the installation guide for Isaac Lab binaries, running the command to launch the simulator with a tutorial script (`isaaclab.bat -p source\standalone\tutorials\00_sim\create_empty.py`) results in the simulator launching, but instead of displaying a black ground plane, the render window only shows a completely black screen. Additionally, there are several warnings and errors in the terminal, including issues with non-NVIDIA GPUs, NGX initialization, and a `UnicodeDecodeError`.

### Steps to reproduce

1. Follow the Isaac Lab installation guide to set up the environment using Isaac Sim binaries on Windows.
2. Run the command:  
   ```
   isaaclab.bat -p source\standalone\tutorials\00_sim\create_empty.py
   ```
3. Observe that the simulator launches but the render window is completely black and doesn't show a ground plane as expected.
4. Check the terminal logs for error messages.

```
C:\Users\Cortex\Documents\IsaacLab>isaaclab.bat -p source\standalone\tutorials\00_sim\create_empty.py
[INFO] Using python from: C:\Users\Cortex\Documents\IsaacLab\\_isaac_sim\python.bat
[WARN][AppLauncher]: There are no arguments attached to the ArgumentParser object. If you have your own arguments, please load your own arguments before calling the `AppLauncher.add_app_launcher_args` method. This allows the method to check the validity of the arguments and perform checks for argument names.
[INFO][AppLauncher]: Loading experience file: c:\users\cortex\documents\isaaclab\source\apps\isaaclab.python.kit
[Info] [carb] Logging to file: c:/users/cortex/documents/isaaclab/_isaac_sim/kit/logs/Kit/Isaac-Sim/4.2/kit_20241018_071601.log
2024-10-17 22:16:01 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
2024-10-17 22:16:04 [3,358ms] [Warning] [gpu.foundation.plugin] Skipping unsupported non-NVIDIA GPU: Intel(R) UHD Graphics 770
2024-10-17 22:16:04 [3,359ms] [Warning] [gpu.foundation.plugin] Skipping unsupported non-NVIDIA GPU: Intel(R) UHD Graphics 770

|---------------------------------------------------------------------------------------------|
| Driver Version: 552.55        | Graphics API: Vulkan
|=============================================================================================|
| GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
|     |                                  |        |     |            | Device-ID | UUID       |
|     |                                  |        |     |            | Bus-ID    |            |
|---------------------------------------------------------------------------------------------|
| 0   | NVIDIA T1000 8GB                 | Yes: 0 |     | 8229    MB | 10de      | 9f030100.. |
|     |                                  |        |     |            | 1ff0      | a4cc6c37.. |
|     |                                  |        |     |            | 6e        |            |
|---------------------------------------------------------------------------------------------|
| 1   | Intel(R) UHD Graphics 770        |        |     | 32587   MB | 8086      | 68130100.. |
|     |                                  |        |     |            | 4680      | 86808046.. |
|     |                                  |        |     |            | N/A       |            |
|=============================================================================================|
| OS: Windows 10 Pro, Version: 10.0 (22H2), Build: 19045, Kernel: 10.0.19041.5007
| Processor: 12th Gen Intel(R) Core(TM) i7-12700 | Cores: 12 | Logical: 20
|---------------------------------------------------------------------------------------------|
| Total Memory (MB): 65174 | Free Memory: 47402
| Total Page/Swap (MB): 74902 | Free Page/Swap: 56124
|---------------------------------------------------------------------------------------------|
2024-10-17 22:16:18 [16,575ms] [Warning] [omni.replicator.core.scripts.annotators] Annotator PostProcessDispatch is already registered, overwriting annotator template
2024-10-17 22:16:19 [18,125ms] [Error] [carb.scripting-python.plugin] UnicodeDecodeError: 'cp932' codec can't decode byte 0x83 in position 1424: illegal multibyte sequence

At:
  C:\Users\Cortex\Documents\IsaacLab\_isaac_sim\kit\python\lib\json\__init__.py(293): load
  c:/users/cortex/documents/isaaclab/_isaac_sim/exts/omni.isaac.sensor/omni/isaac/sensor/scripts/menu.py(232): __init__
  c:/users/cortex/documents/isaaclab/_isaac_sim/exts/omni.isaac.sensor/omni/isaac/sensor/scripts/extension.py(37): on_startup
  C:\Users\Cortex\Documents\IsaacLab\_isaac_sim/kit/kernel/py\omni\ext\_impl\_internal.py(164): _startup_ext
  C:\Users\Cortex\Documents\IsaacLab\_isaac_sim/kit/kernel/py\omni\ext\_impl\_internal.py(224): startup
  C:\Users\Cortex\Documents\IsaacLab\_isaac_sim/kit/kernel/py\omni\ext\_impl\_internal.py(328): startup_extension
  PythonExtension.cpp::startup()(2): <module>
  C:\Users\Cortex\Documents\IsaacLab\_isaac_sim/exts/omni.isaac.kit\omni\isaac\kit\simulation_app.py(411): _start_app
  C:\Users\Cortex\Documents\IsaacLab\_isaac_sim/exts/omni.isaac.kit\omni\isaac\kit\simulation_app.py(200): __init__
  c:\users\cortex\documents\isaaclab\source\extensions\omni.isaac.lab\omni\isaac\lab\app\app_launcher.py(588): _create_app
  c:\users\cortex\documents\isaaclab\source\extensions\omni.isaac.lab\omni\isaac\lab\app\app_launcher.py(113): __init__
  C:\Users\Cortex\Documents\IsaacLab\source\standalone\tutorials\00_sim\create_empty.py(29): <module>

2024-10-17 22:16:19 [18,126ms] [Error] [omni.ext.plugin] [ext: omni.isaac.sensor-12.9.1] Failed to startup python extension.
2024-10-17 22:16:22 [21,232ms] [Warning] [carb.graphics-vulkan.plugin] Trying to initialize NGX without vkGetInstanceProcAddr and vkGetDeviceProcAddr pointer-to-functions.
2024-10-17 22:16:22 [21,243ms] [Error] [carb.graphics-vulkan.plugin] Could not get NGX parameters block because NGX isn't enabled.
2024-10-17 22:16:22 [21,244ms] [Error] [carb.graphics-vulkan.plugin] Failed to create NGX context.
2024-10-17 22:16:22 [21,244ms] [Warning] [carb.scenerenderer-rtx.plugin] Failed to create NGX context.
[INFO]: Setup complete…
```

### System Info

Describe the characteristic of your environment:

- Commit: [1117 ?]
- Isaac Sim Version: [4.2.0]
- OS: [Windows 10 Pro, Version 10.0 (22H2)]
- GPU: [NVIDIA T1000 8GB, Intel(R) UHD Graphics 770]
- CUDA: [-]
- GPU Driver: [552.55 (NVIDIA T1000)]

### Additional context

- The system has two GPUs, but only the NVIDIA T1000 is active for IsaacLab. The Intel UHD Graphics 770 triggers warnings about unsupported non-NVIDIA GPUs.
- The error `UnicodeDecodeError: 'cp932' codec can't decode byte 0x83` is present in the logs and seems to be related to file encoding.
- Additionally, there are Vulkan API and NGX initialization issues.

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

- [ ] A proper ground plane is displayed in the render window.
- [ ] Errors and warnings related to NGX, Vulkan, and encoding are resolved.

Bowie1026, 2024-10-19:
I have the same question,
[image]

And there is no errors in my log.
`(isaaclab) G:\2024_Tennis_Robot\IsaacLab-main>isaaclab.bat -p source\standalone\tutorials\00_sim\create_empty.py
[INFO] Using python from: E:\anaconda\envs\isaaclab\python.exe
[WARN][AppLauncher]: There are no arguments attached to the ArgumentParser object. If you have your own arguments, please load your own arguments before calling the `AppLauncher.add_app_launcher_args` method. This allows the method to check the validity of the arguments and perform checks for argument names.
[INFO][AppLauncher]: Loading experience file: G:\2024_Tennis_Robot\IsaacLab-main\source\apps\isaaclab.python.kit
[Warning] [omni.isaac.kit.simulation_app] Modules: ['omni.kit_app'] were loaded before SimulationApp was started and might not be loaded correctly.
[Warning] [omni.isaac.kit.simulation_app] Please check to make sure no extra omniverse or pxr modules are imported before the call to SimulationApp(...)
Loading user config located at: 'e:/anaconda/envs/isaaclab/lib/site-packages/omni/data/Kit/Isaac-Sim/4.2/user.config.json'
[Info] [carb] Logging to file: e:/anaconda/envs/isaaclab/lib/site-packages/omni/logs/Kit/Isaac-Sim/4.2/kit_20241019_172004.log
2024-10-19 21:20:04 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.

|---------------------------------------------------------------------------------------------|
| Driver Version: 565.90        | Graphics API: Vulkan
|=============================================================================================|
| GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
|     |                                  |        |     |            | Device-ID | UUID       |
|     |                                  |        |     |            | Bus-ID    |            |
|---------------------------------------------------------------------------------------------|
| 0   | NVIDIA GeForce RTX 2070 Super .. | Yes: 0 |     | 8203    MB | 10de      | 6aa60100.. |
|     |                                  |        |     |            | 1ed1      | 75a9c30a.. |
|     |                                  |        |     |            | 1         |            |
|=============================================================================================|
| OS: Windows 10 Home China, Version: 10.0 (22H2), Build: 19045, Kernel: 10.0.19041.5007
| Processor: Intel(R) Core(TM) i7-10750H CPU @ 2.60GHz | Cores: 6 | Logical: 12
|---------------------------------------------------------------------------------------------|
| Total Memory (MB): 16222 | Free Memory: 7733
| Total Page/Swap (MB): 86702 | Free Page/Swap: 75501
|---------------------------------------------------------------------------------------------|
2024-10-19 21:20:11 [7,642ms] [Warning] [omni.replicator.core.scripts.annotators] Annotator PostProcessDispatch is already registered, overwriting annotator template
2024-10-19 21:20:12 [8,221ms] [Warning] [omni.kit.widget.cache_indicator.cache_state_menu] Unable to detect Omniverse Cache Server. File C:/Users/hp/.nvidia-omniverse/config/omniverse.toml is not found. Consider installing it for better IO performance.
[INFO]: Setup complete...`

Bowie1026, 2024-10-19:
I have the same problem, and there is no errors in my log.
'''
(isaaclab) G:\2024_Tennis_Robot\IsaacLab-main>isaaclab.bat -p source\standalone\tutorials\00_sim\create_empty.py
[INFO] Using python from: E:\anaconda\envs\isaaclab\python.exe
[WARN][AppLauncher]: There are no arguments attached to the ArgumentParser object. If you have your own arguments, please load your own arguments before calling the `AppLauncher.add_app_launcher_args` method. This allows the method to check the validity of the arguments and perform checks for argument names.
[INFO][AppLauncher]: Loading experience file: G:\2024_Tennis_Robot\IsaacLab-main\source\apps\isaaclab.python.kit
[Warning] [omni.isaac.kit.simulation_app] Modules: ['omni.kit_app'] were loaded before SimulationApp was started and might not be loaded correctly.
[Warning] [omni.isaac.kit.simulation_app] Please check to make sure no extra omniverse or pxr modules are imported before the call to SimulationApp(...)
Loading user config located at: 'e:/anaconda/envs/isaaclab/lib/site-packages/omni/data/Kit/Isaac-Sim/4.2/user.config.json'
[Info] [carb] Logging to file: e:/anaconda/envs/isaaclab/lib/site-packages/omni/logs/Kit/Isaac-Sim/4.2/kit_20241019_172004.log
2024-10-19 21:20:04 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.

|---------------------------------------------------------------------------------------------|
| Driver Version: 565.90        | Graphics API: Vulkan
|=============================================================================================|
| GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
|     |                                  |        |     |            | Device-ID | UUID       |
|     |                                  |        |     |            | Bus-ID    |            |
|---------------------------------------------------------------------------------------------|
| 0   | NVIDIA GeForce RTX 2070 Super .. | Yes: 0 |     | 8203    MB | 10de      | 6aa60100.. |
|     |                                  |        |     |            | 1ed1      | 75a9c30a.. |
|     |                                  |        |     |            | 1         |            |
|=============================================================================================|
| OS: Windows 10 Home China, Version: 10.0 (22H2), Build: 19045, Kernel: 10.0.19041.5007
| Processor: Intel(R) Core(TM) i7-10750H CPU @ 2.60GHz | Cores: 6 | Logical: 12
|---------------------------------------------------------------------------------------------|
| Total Memory (MB): 16222 | Free Memory: 7733
| Total Page/Swap (MB): 86702 | Free Page/Swap: 75501
|---------------------------------------------------------------------------------------------|
2024-10-19 21:20:11 [7,642ms] [Warning] [omni.replicator.core.scripts.annotators] Annotator PostProcessDispatch is already registered, overwriting annotator template
2024-10-19 21:20:12 [8,221ms] [Warning] [omni.kit.widget.cache_indicator.cache_state_menu] Unable to detect Omniverse Cache Server. File C:/Users/hp/.nvidia-omniverse/config/omniverse.toml is not found. Consider installing it for better IO performance.
[INFO]: Setup complete...
'''

Bowie1026, 2024-10-19:
I have the same question(black ground plane) and there is no errors in my log. 
[image]

```
(isaaclab) G:\2024_Tennis_Robot\IsaacLab-main>isaaclab.bat -p source\standalone\tutorials\00_sim\create_empty.py
[INFO] Using python from: E:\anaconda\envs\isaaclab\python.exe
[WARN][AppLauncher]: There are no arguments attached to the ArgumentParser object. If you have your own arguments, please load your own arguments before calling the `AppLauncher.add_app_launcher_args` method. This allows the method to check the validity of the arguments and perform checks for argument names.
[INFO][AppLauncher]: Loading experience file: G:\2024_Tennis_Robot\IsaacLab-main\source\apps\isaaclab.python.kit
[Warning] [omni.isaac.kit.simulation_app] Modules: ['omni.kit_app'] were loaded before SimulationApp was started and might not be loaded correctly.
[Warning] [omni.isaac.kit.simulation_app] Please check to make sure no extra omniverse or pxr modules are imported before the call to SimulationApp(...)
Loading user config located at: 'e:/anaconda/envs/isaaclab/lib/site-packages/omni/data/Kit/Isaac-Sim/4.2/user.config.json'
[Info] [carb] Logging to file: e:/anaconda/envs/isaaclab/lib/site-packages/omni/logs/Kit/Isaac-Sim/4.2/kit_20241019_172004.log
2024-10-19 21:20:04 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.

|---------------------------------------------------------------------------------------------|
| Driver Version: 565.90        | Graphics API: Vulkan
|=============================================================================================|
| GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
|     |                                  |        |     |            | Device-ID | UUID       |
|     |                                  |        |     |            | Bus-ID    |            |
|---------------------------------------------------------------------------------------------|
| 0   | NVIDIA GeForce RTX 2070 Super .. | Yes: 0 |     | 8203    MB | 10de      | 6aa60100.. |
|     |                                  |        |     |            | 1ed1      | 75a9c30a.. |
|     |                                  |        |     |            | 1         |            |
|=============================================================================================|
| OS: Windows 10 Home China, Version: 10.0 (22H2), Build: 19045, Kernel: 10.0.19041.5007
| Processor: Intel(R) Core(TM) i7-10750H CPU @ 2.60GHz | Cores: 6 | Logical: 12
|---------------------------------------------------------------------------------------------|
| Total Memory (MB): 16222 | Free Memory: 7733
| Total Page/Swap (MB): 86702 | Free Page/Swap: 75501
|---------------------------------------------------------------------------------------------|
2024-10-19 21:20:11 [7,642ms] [Warning] [omni.replicator.core.scripts.annotators] Annotator PostProcessDispatch is already registered, overwriting annotator template
2024-10-19 21:20:12 [8,221ms] [Warning] [omni.kit.widget.cache_indicator.cache_state_menu] Unable to detect Omniverse Cache Server. File C:/Users/hp/.nvidia-omniverse/config/omniverse.toml is not found. Consider installing it for better IO performance.
[INFO]: Setup complete...
```

YuHoChau, 2024-10-23:
Same error here

xkq1qkx, 2024-11-02:
almost the same error here：
```
IsaacLab$ python source/standalone/tutorials/00_sim/create_empty.py
[WARN][AppLauncher]: There are no arguments attached to the ArgumentParser object. If you have your own arguments, please load your own arguments before calling the `AppLauncher.add_app_launcher_args` method. This allows the method to check the validity of the arguments and perform checks for argument names.
[INFO][AppLauncher]: Loading experience file: /home/yfwang/Desktop/IsaacLab/source/apps/isaaclab.python.kit
Loading user config located at: '/home/yfwang/.local/share/ov/pkg/isaac-sim-4.2.0/kit/data/Kit/Isaac-Sim/4.2/user.config.json'
[Info] [carb] Logging to file: /home/yfwang/.local/share/ov/pkg/isaac-sim-4.2.0/kit/logs/Kit/Isaac-Sim/4.2/kit_20241103_020553.log
2024-11-02 18:05:53 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.

|---------------------------------------------------------------------------------------------|
| Driver Version: 545.29.06     | Graphics API: Vulkan
|=============================================================================================|
| GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
|     |                                  |        |     |            | Device-ID | UUID       |
|     |                                  |        |     |            | Bus-ID    |            |
|---------------------------------------------------------------------------------------------|
| 0   | NVIDIA GeForce RTX 4090          | Yes: 0 |     | 24564   MB | 10de      | 0          |
|     |                                  |        |     |            | 2684      | 4f2df5a2.. |
|     |                                  |        |     |            | 1         |            |
|=============================================================================================|
| OS: 22.04.4 LTS (Jammy Jellyfish) ubuntu, Version: 22.04.4, Kernel: 6.5.0-44-generic
| XServer Vendor: The X.Org Foundation, XServer Version: 12101004 (1.21.1.4)
| Processor: Intel(R) Core(TM) i7-14700KF | Cores: 20 | Logical: 40
|---------------------------------------------------------------------------------------------|
| Total Memory (MB): 64102 | Free Memory: 59631
| Total Page/Swap (MB): 0 | Free Page/Swap: 0
|---------------------------------------------------------------------------------------------|
2024-11-02 18:05:57 [3,265ms] [Warning] [omni.replicator.core.scripts.annotators] Annotator PostProcessDispatch is already registered, overwriting annotator template
[INFO]: Setup complete...

```
And the render window is black.

xkq1qkx, 2024-11-02:
> almost the same error here：
>
> ```
> IsaacLab$ python source/standalone/tutorials/00_sim/create_empty.py
> [WARN][AppLauncher]: There are no arguments attached to the ArgumentParser object. If you have [...]

maybe there is nothing wrong. try this, and i find it works well :
./isaaclab.sh -p source/standalone/tutorials/02_scene/create_scene.py

louislelay, 2024-11-02:
Alright @xkq1qkx, I'll give it a shot later.

For anyone else facing the error `UnicodeDecodeError: 'cp932' codec can't decode byte`, it seems that this happens when the app tries to read a file or string with characters that don't work with the cp932 encoding (often used in Japanese Windows envs, as in my case).

A likely fix is to change the encoding used by Isaac Sim. I resolved it by entering `set PYTHONUTF8=1` in the terminal before launching the application.

That said, I was still running into a black screen in the render view.

louislelay, 2024-11-13:
> maybe there is nothing wrong. try this, and i find it works well : ./isaaclab.sh -p source/standalone/tutorials/02_scene/create_scene.py

Ok it was because of this, so it wasn't related to the other errors I had.

closed 2025-02-26 by louislelay