isaac-sim/IsaacLab #1270: [Question] How to grasp and lift an object with a Franka robot in a Direct RL approach?

pmfaustino, 2024-10-18:
I have seen and tried out the [Isaac-Lift-Cube-Franka-v0 Environment](https://github.com/isaac-sim/IsaacLab/blob/main/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/manager_based/manipulation/lift/config/franka/joint_pos_env_cfg.py), where the Franka robot is able to pick up and lift a cube following a Manager Based RL approach and I was trying to obtain similar results following a Direct RL approach instead. Using the [Isaac-Franka-Cabinet-Direct-v0 Environment](https://github.com/isaac-sim/IsaacLab/blob/main/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/direct/franka_cabinet/franka_cabinet_env.py) as a basis, a few changes in the code have been made to be able to reach the cube and lift it. The issue is that the robot gets close to the cube, but doesn't grasp it or lift it and I can't understand what I'm missing/doing wrong. You may find the code attached below. Any help would be appreciated.

[franka_cuboid_env.txt](https://github.com/user-attachments/files/17437839/franka_cuboid_env.txt)

mpgussert, 2024-10-18:
Hello @pmfaustino!

Unfortunately, there's not a lot we can do to help without more information.  Can you provide a code snippet or example?

Also, How are you controlling the robot in both cases?  Are you using the "cube grabbing policy" from the first environment, in the second?  

thanks!
Gus

pmfaustino, 2024-10-18:
> Hello @pmfaustino!
> 
> Unfortunately, there's not a lot we can do to help without more information. Can you provide a code snippet or example?
> 
> Also, How are you controlling the robot in both cases? Are you using the "cube grabbing policy" from the first environment, in the second?
> 
> thanks! Gus

I apologize for the misunderstanding. The code that I have been running is the one from the attached file. It is a python file, but I was unable to attach it with the proper extension due to GitHub's policy for file attachments.

But that's basically the gist of it, I was trying to achieve similar results to the first environment (Manager Based RL approach) by altering the code from the second environment (Direct RL approach). Have not been able to successfully do so though.

mpgussert, 2024-10-24:
hello again!

Sorry for the delayed response.  I've been taking a look at your code and I'm curious about this reward function

`    def _compute_rewards(
        self,
        actions,
        franka_lfinger_pos,
        franka_rfinger_pos,
        end_effector_pos,
        cuboid_pos,
        dist_reward_scale,
        lift_reward_scale,
        action_penalty_scale,
        finger_reward_scale,
        num_envs,
    ):

        # Check if the object is lifted above a minimal height
        minimal_height = 0.04
        lift_reward = torch.where(cuboid_pos[:, 2] > minimal_height, 1.0, 0.0)

        # Calculate the distance between end-effector and the cube
        object_distance = torch.norm(cuboid_pos - end_effector_pos, dim=1)
        distance_reward = 1.0 - torch.tanh(object_distance / 0.1)

        # Penalty for excessive action usage
        action_penalty = torch.sum(actions ** 2, dim=-1)

        # Encourage smaller finger distances to grasp the object
        lfinger_dist = torch.norm(franka_lfinger_pos - cuboid_pos, dim=1)
        rfinger_dist = torch.norm(franka_rfinger_pos - cuboid_pos, dim=1)
        finger_dist_penalty = (lfinger_dist + rfinger_dist) * 0.5

        # Total reward calculation
        rewards = (
            lift_reward * lift_reward_scale
            + distance_reward * dist_reward_scale
            - action_penalty * action_penalty_scale
            - finger_dist_penalty * finger_reward_scale
        )

        self.extras["log"] = {
            "distance_reward": (dist_reward_scale * distance_reward).mean(),
            "lift_reward": (lift_reward_scale * lift_reward).mean(),
            "action_penalty": (-action_penalty_scale * action_penalty).mean(),
            "finger_penalty": (finger_reward_scale * finger_dist_penalty).mean(),
        }

        return rewards
`
specifically, the grasp portion 

`        # Encourage smaller finger distances to grasp the object
        lfinger_dist = torch.norm(franka_lfinger_pos - cuboid_pos, dim=1)
        rfinger_dist = torch.norm(franka_rfinger_pos - cuboid_pos, dim=1)
        finger_dist_penalty = (lfinger_dist + rfinger_dist) * 0.5`

so you want the left and right fingers to be as close to the cuboid COM as possible, but there are TWO possible solutions to this (two minima).  One is with the cube in between the fingers, but the other is with the gripper closed and both fingers on one side (they would both have the same distance to the COM as the case when it's in between them).

I think the issue here is you need to modify the reward function to explicitly care about how much of the cuboid is BETWEEN the pinchers, and not just the distance to the COM.  You may be able to do this with some simple modifications in what you currently have, 

Imagine drawing two vectors: from the cuboid COM to each finger.  When both fingers are on one side, those vectors are pointing in the same direction, but when it's between them, they point in opposite directions.

What if instead of caring about the norm, what if instead you just computed those to vectors, and penalized the inner product between them as well as their individual lengths?

linden713, 2024-10-24:
Hi @pmfaustino, I am tring to lift a cube using manager based approach. But I failed to lift it. Could you tell me where did you change to achieve that?

pmfaustino, 2024-10-25:
> hello again!
>
> Sorry for the delayed response. I've been taking a look at your code and I'm curious about this reward function
>
> ```
> def _compute_rewards(
>         self,
>         actions, [...]

Thank you @mpgussert  for your feedback. I'm relatively new to this, but I think I understand what you're saying and will try to make the changes you mentioned to see whether the results improve.

pmfaustino, 2024-10-25:
> Hi @pmfaustino, I am tring to lift a cube using manager based approach. But I failed to lift it. Could you tell me where did you change to achieve that?

Hi @linden713 . There is one available environment for Isaac Lab, called [Isaac-Lift-Cube-Franka-v0](https://github.com/isaac-sim/IsaacLab/blob/main/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/manager_based/manipulation/lift/config/franka/joint_pos_env_cfg.py), which follows a Manager-Based approach and uses a Franka robot to lift and cube and take it to a target position, might be worth checking it out if you haven't done it yet. It might be similar to what you're trying to achieve.

pmfaustino, 2024-10-25:
> Hi @pmfaustino. Thank you for your reply. I tried this environment several days ago, but using the default parameters did not yield the expected results (I used skrl). Afterwards, I added some additional reward functions, such as grasp_object_reward, and increased the number of training steps. However, the results were still not satisfactory. So, I’d like to ask how you achieved it? Did you use the default parameters, and which reinforcement learning library did you use?

I was able to achieve good results without making any code or parameter changes. I merely followed the [Training with an RL Agent](https://isaac-sim.github.io/IsaacLab/main/source/tutorials/03_envs/run_rl_training.html) tutorial to train and play the trained agent and it worked fine (also using skrl). so I'm afraid I don't know how to help you.

RandomOakForest, 2024-11-16:
Thanks for posting this and sharing this discussions. Will close this issue for now, feel free to reopen may we be able to provide further help.

closed 2024-11-16 by RandomOakForest

itsmehere, 2025-01-17:
@pmfaustino When you said "I have seen and tried out the [Isaac-Lift-Cube-Franka-v0 Environment](https://github.com/isaac-sim/IsaacLab/blob/main/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/manager_based/manipulation/lift/config/franka/joint_pos_env_cfg.py), where the Franka robot is able to pick up and lift a cube following a Manager Based RL approach..." I was wondering how you were able to do that. I am trying to get just the example environment to work and I've outlined how I'm running that below, but am I doing anything incorrectly?

**My steps:**
I’m running the command below to try and train the Franka arm to lift a cube and bring it to the sampled target position:

./isaaclab.sh -p source/standalone/workflows/rsl_rl/train.py --task Isaac-Lift-Cube-Franka-v0 --num_envs 64 --headless

When I play this policy using

./isaaclab.sh -p source/standalone/workflows/rsl_rl/play.py --task=Isaac-Lift-Cube-Franka-v0 --num_envs=1

The franka arm just moves around slightly but doesn’t go anywhere near the cube, nor does it pick it up. Am I doing something incorrect when training, or is there some value I have to set? All I’ve done is setup IsaacLab and IsaacSim by following the installation instructions, and I haven’t changed any code. I’m in the process of learning so I’m running a few examples right off the bat, and I’m just trying to get Isaac-Lift-Cube-Franka-v0 as listed [here](https://isaac-sim.github.io/IsaacLab/main/source/overview/environments.html) to work.

itsmehere, 2025-01-17:
@linden713 Were you able to figure out the issue? I am also not able to get the [Isaac-Lift-Cube-Franka-v0](https://github.com/isaac-sim/IsaacLab/blob/main/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/manager_based/manipulation/lift/config/franka/joint_pos_env_cfg.py) env working. I trained a policy to 10k iterations with 64 env using this command: `./isaaclab.sh -p source/standalone/workflows/rsl_rl/train.py --task=Isaac-Lift-Cube-Franka-v0 --num_envs=64`, and the result is shown below. The arm moves but doesn't pick up the cube.

https://github.com/user-attachments/assets/ecd3ba89-a87b-4030-8aad-1da8d8e10e9f

xiaolijz, 2025-02-19:
> I was able to achieve good results without making any code or parameter changes. I merely followed the [Training with an RL [...]

Hello, I'm encountering a problem with picking up a cube. When I close the gripper, I can't lift the cube. It seems that the relationship between the cube and the robot has not been established.
```
           if current_goal_idx == 2:
                for _ in range(20):
                    sim.step()
                robot.set_joint_position_target(gripper_close_pos, joint_ids=gripper_joint_ids)
                scene.write_data_to_sim()
                for _ in range(50):
                    sim.step()
```

https://github.com/user-attachments/assets/acbea0bf-0d01-4ad4-8b0f-1a754a82a9c5