isaac-sim/IsaacLab #1280: [Question] The task of the robot's learning policy used in the PreTraindicated Policy Action.

Hokite, 2024-10-21:
I would like to know the task of the robot's learning policy used in the PreTraindicated Policy Action.
policy_path=f"{ISAACLAB_NUCLEUS_DIR}/Policies/anymal-C/Blind/policy.pt " 
What kind of learning task did the learning policy in that path do?

[image]

RandomOakForest, 2024-12-16:
That seems to refer to the [navigation environment here](https://isaac-sim.github.io/IsaacLab/main/source/overview/environments.html#navigation). Our apologies we don't yet have this clearly documented.

cikufa, 2025-03-04:
Hi @Hokite 
Were you able to figure out the details of blind and heightscan policies?

StafaH, 2026-09-16:
Thanks @Hokite for asking what the `Policies/ANYmal-C/Blind/policy.pt` policy was trained to do.

This is the low-level ANYmal-C locomotion controller used by the navigation environment, rather than the navigation policy itself. It tracks a three-dimensional base-velocity command—forward velocity, lateral velocity, and yaw rate—and outputs 12 joint-position actions. The high-level navigation policy supplies those velocity commands while learning to reach a target position and heading.

“Blind” refers to the flat-terrain variant, which uses proprioceptive observations without a terrain height scan. `HeightScan/policy.pt` is the rough-terrain variant and additionally consumes the height-scanner grid.

The current wiring is in `source/isaaclab_tasks/isaaclab_tasks/contrib/navigation/config/anymal_c/navigation_env_cfg.py`, while the underlying velocity-tracking observations and rewards are defined under `source/isaaclab_tasks/isaaclab_tasks/core/velocity/`.

Closing this as answered. Please reopen if you need the original training run metadata rather than the policy’s task and interface.

closed 2026-09-16 by StafaH