isaac-sim/IsaacLab #1284: [Bug Report] Bug title Manipulator robot arm velocity_limit gets ingored

hanlin-ga, 2024-10-21:
I tested the following command `./isaaclab.sh -p source/standalone/workflows/rsl_rl/play.py --task Isaac-Lift-Cube-Franka-v0 --num_envs` for Franka lifting cube task, I modified the velocity_limit of franka.py under ~/IsaacLab/source/extensions/omni.isaac.lab_assets/omni/isaac/lab_assets. I set the velocity_limit to be a very small value 0.0001, but when I run the command, franka still moves as fast as it can. It seems the velocity_limit is totally ignored. 

`    actuators={
        "panda_shoulder": ImplicitActuatorCfg(
            joint_names_expr=["panda_joint[1-4]"],
            effort_limit=87.0,
            velocity_limit=0.0001,
            stiffness=80.0,
            damping=4.0,
        ),
        "panda_forearm": ImplicitActuatorCfg(
            joint_names_expr=["panda_joint[5-7]"],
            effort_limit=12.0,
            velocity_limit=0.0001,
            stiffness=80.0,
            damping=4.0,
        ),
        "panda_hand": ImplicitActuatorCfg(
            joint_names_expr=["panda_finger_joint.*"],
            effort_limit=200.0,
            velocity_limit=0.2,
            stiffness=2e3,
            damping=1e2,
        ),
    },`

jsmith-bdai, 2024-10-28:
Yes, this is correct. The `ImplicitActuator` doesn't utilize the `velocity_limit` parameter. It is however used by other actuator types. 

What are you trying to achieve? You can potential use the effort_limit to slow robot motion down if that's what you're after

RandomOakForest, 2024-11-22:
Thanks for posting this. Will close this for now as you may have found a workaround. Please reopen when you have more details for us to investigate. Thanks.

closed 2024-11-22 by RandomOakForest