isaac-sim/IsaacLab #1289: [Question] Assistance Required for Configuring Environment dt in SKRL

XinyuSong123, 2024-10-22:
Hi,

I attempted to run the SKRL examples, and the output information is as follows:

[INFO]: Base environment:
Environment device: cuda:0
Environment seed: None
Physics step-size: 0.005
Rendering step-size: 0.02
Environment step-size: 0.02

However, upon reviewing the simulation configuration, I noticed that the physics dt is set to the default value of 1.0 / 60.0, which differs from the output information above.

Could you kindly provide guidance on how to correctly configure the environment dt when using SKRL? Thank you in advance!

Mayankm96, 2024-10-22:
Which environment are you running?

XinyuSong123, 2024-10-22:
> Which environment are you running?

I downloaded [torch_humanoid_ppo.py](https://skrl.readthedocs.io/en/latest/_downloads/51f7463da2cb989ae449368604e53543/torch_humanoid_ppo.py) and changed the task to Isaac-Velocity-Flat-G1-v0

T-K-233, 2024-11-22:
Hello @XinyuSong123 , could you give a pointer on how to correctly configure the physics / environment frequency? Thx!

XinyuSong123, 2024-11-22:
Hi,

For my G1 environment, the configuration file is located at the path:
```
/IsaacLab/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/manager_based/locomotion/velocity/velocity_env_cfg.py.
```
Within this file, the class LocomotionVelocityRoughEnvCfg contains the basic environment settings. You can modify these settings by adjusting its attributes:

```

@configclass
class LocomotionVelocityRoughEnvCfg(ManagerBasedRLEnvCfg):
    """Configuration for the locomotion velocity-tracking environment."""

    # Scene settings
    scene: MySceneCfg = MySceneCfg(num_envs=4096, env_spacing=2.5)
    # Basic settings
    observations: ObservationsCfg = ObservationsCfg()
    actions: ActionsCfg = ActionsCfg()
    commands: CommandsCfg = CommandsCfg()
    # MDP settings
    rewards: RewardsCfg = RewardsCfg()
    terminations: TerminationsCfg = TerminationsCfg()
    events: EventCfg = EventCfg()
    curriculum: CurriculumCfg = CurriculumCfg()

    def __post_init__(self):
        """Post initialization."""
        # general settings
        self.decimation = 1
        self.episode_length_s = 20.0
        # simulation settings
        self.sim.dt = 1/60
        self.sim.render_interval = self.decimation
        self.sim.disable_contact_processing = True
        self.sim.physics_material = self.scene.terrain.physics_material
        # update sensor update periods
        # we tick all the sensors based on the smallest update period (physics update period)
        if self.scene.height_scanner is not None:
            self.scene.height_scanner.update_period = self.decimation * self.sim.dt
        if self.scene.contact_forces is not None:
            self.scene.contact_forces.update_period = self.sim.dt

        # check if terrain levels curriculum is enabled - if so, enable curriculum for terrain generator
        # this generates terrains with increasing difficulty and is useful for training
        if getattr(self.curriculum, "terrain_levels", None) is not None:
            if self.scene.terrain.terrain_generator is not None:
                self.scene.terrain.terrain_generator.curriculum = True
        else:
            if self.scene.terrain.terrain_generator is not None:
                self.scene.terrain.terrain_generator.curriculum = False
```