isaac-sim/IsaacLab #1290: [Bug Report] Hydra's OmegaConfig is not compatible with MultiAssetSpawnerCfg

zoctipus, 2024-10-22:
If you are submitting a bug report, please fill in the following details and use the tag [bug].

### Describe the bug

Using MultiAssetSpawnerCfg as spawn will make Hydra's ConfigStore unhappy when 
`ConfigStore.instance().store(name=task_name, node=cfg_dict)`
and produces error such as 
```
Exception has occurred: ConfigKeyError
module 'omni.isaac.lab.sim.schemas.schemas' has no attribute 'MassPropertiesCfg'
    full_key: 0
    object_type=None
    full_key: env.scene.object.spawn.assets_cfg[0]
    object_type=list
    full_key: assets_cfg
    object_type=None
    full_key: env.scene.object.spawn.assets_cfg
    object_type=dict
```

### Steps to reproduce

To provide an minimal example that reproduce this behavior, consider editing below example

say for example in `source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/manager_based/manipulation/lift/config/franka/joint_pos_env_cfg.py`

```
# Set Cube as object
        self.scene.object = RigidObjectCfg(
            prim_path="{ENV_REGEX_NS}/Object",
            init_state=RigidObjectCfg.InitialStateCfg(pos=[0.5, 0, 0.055], rot=[1, 0, 0, 0]),
            spawn=UsdFileCfg(
                usd_path=f"{ISAAC_NUCLEUS_DIR}/Props/Blocks/DexCube/dex_cube_instanceable.usd",
                scale=(0.8, 0.8, 0.8),
                rigid_props=RigidBodyPropertiesCfg(
                    solver_position_iteration_count=16,
                    solver_velocity_iteration_count=1,
                    max_angular_velocity=1000.0,
                    max_linear_velocity=1000.0,
                    max_depenetration_velocity=5.0,
                    disable_gravity=False,
                ),
            ),
        )
```

changing above code with below to use MultiAssetSpawnerCfg

```
# Set Cube as object
        self.scene.object = RigidObjectCfg(
            prim_path="{ENV_REGEX_NS}/Object",
            init_state=RigidObjectCfg.InitialStateCfg(pos=[0.5, 0, 0.055], rot=[1, 0, 0, 0]),
            spawn=sim_utils.MultiAssetSpawnerCfg(
                assets_cfg=[
                    UsdFileCfg(
                    usd_path=f"{ISAAC_NUCLEUS_DIR}/Props/Blocks/DexCube/dex_cube_instanceable.usd",
                    scale=(0.8, 0.8, 0.8),
                    rigid_props=RigidBodyPropertiesCfg(
                        solver_position_iteration_count=16,
                        solver_velocity_iteration_count=1,
                        max_angular_velocity=1000.0,
                        max_linear_velocity=1000.0,
                        max_depenetration_velocity=5.0,
                        disable_gravity=False,
                    ),),
                ],
            ),
        )
```

Then Launching Training with task name "Isaac-Lift-Cube-Franka-v0" will error with
```
Exception has occurred: ConfigKeyError
module 'omni.isaac.lab.sim.schemas.schemas' has no attribute 'MassPropertiesCfg'
    full_key: 0
    object_type=None
    full_key: env.scene.object.spawn.assets_cfg[0]
    object_type=list
    full_key: assets_cfg
    object_type=None
    full_key: env.scene.object.spawn.assets_cfg
    object_type=dict
```
-->

### System Info

Describe the characteristic of your environment:

- Commit: [6bc4d0a](https://github.com/isaac-sim/IsaacLab/commit/6bc4d0a01a800c015d33a1b0dc78220216664332)
- Isaac Sim Version: 4.2
- OS: Ubuntu 20.04
- GPU: 4090
- CUDA: 12.5
- GPU Driver: 555.42.02

### Additional context

I tried to fix the issue on my own, and discovered some broken import at 
`source/extensions/omni.isaac.lab/omni/isaac/lab/sim/spawners/spawner_cfg.py`

However correcting these import doesn't make the Hydra config store happy and more type hint issue still exists 

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

- [ ] Criteria 1 A normal training script can be run with SceneCfg containing MultiAsset Sprawner

Mayankm96, 2024-10-22:
Wondering if https://github.com/isaac-sim/IsaacLab/pull/1227 resolves the issue.

zoctipus, 2024-10-23:
it resolves!

closed 2024-10-23 by zoctipus