isaac-sim/IsaacLab #1297: [Bug Report] Can't link the omniverse assets in training process inside docker container.

Privilger, 2024-10-24:
### Describe the bug

Following the [docker deployment tutorial](https://isaac-sim.github.io/IsaacLab/main/source/deployment/docker.html), I was able to run isaacsim inside the docker container with headless mode on a **remote server workstation computer**.

However, when I try to train a G1 walking policy using the following command:
`/isaac-sim/python.sh source/standalone/workflows/rsl_rl/train.py --task Isaac-Velocity-Flat-G1-v0 --num_env 4096 --headless`

I receive this warning and the python program is terminated.
[image]

### Steps to reproduce

1. Enter the docker container with this [tutorial](https://isaac-sim.github.io/IsaacLab/main/source/deployment/docker.html)

3. run command:
 `/isaac-sim/python.sh source/standalone/workflows/rsl_rl/train.py --task Isaac-Velocity-Flat-G1-v0 --num_env 4096 --headless`

 
 
8. Could not open asset @http://omniverse-content-production....
 [image]

### System Info

Describe the characteristic of your environment:

- Commit: a56291d07a28548784f46aed16a4b219c9b57e39
- Isaac Sim Version: 4.2.0
- OS: Ubuntu 20.04
- GPU: RTX L20
- CUDA: 12.2
- GPU Driver: 535

### Additional context

I check my **remote server computer**'s network to see if it is network problem.
I was able to download the asset using 'wget http://omniverse-content-production....' command to download the usd asset successful.

I also try to run the training python script in my **local computer** with docker container, it work well. 

### Checklist

- [x ] I have checked that there is no similar issue in the repo (**required**)
- [ x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

jsmith-bdai, 2024-10-28:
Thanks for posting - this is strange. Can you attach a full stack trace please?

xiaojuntime, 2024-10-29:
# On host side
1. Determine local assets directory by  `$ docker inspect isaac-lab-base`
For instance, using following volume mapping
    host: "/var/lib/docker/volumes/docker_isaac-lab-data/_data"
    container: "/workspace/isaaclab/data_storage"
Then create the local directory structure 
$ sudo mkdir -p /var/lib/docker/volumes/docker_isaac-lab-data/_data/Assets/Isaac/4.2/

2. Download assets to local directory
You can pre-copy/download any assets you want to the local directory.
For 'Isaac-Velocity-Flat-G1-v0' task, following assets are needed.
[image]

# In isaac-lab-base container
Change the assets_path in /workspace/isaaclab/source/extensions/omni.isaac.lab/omni/isaac/lab/app/app_launcher.py from external website to local asset directory.
        #assets_path = "http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/4.2"
        assets_path = "/workspace/isaaclab/data_storage/Assets/Isaac/4.2"

Following command should be ok.
`# /isaac-sim/python.sh source/standalone/workflows/rsl_rl/train.py --task Isaac-Velocity-Flat-G1-v0 --num_env 4096 --headless`

Privilger, 2024-11-05:
> Thanks for posting - this is strange. Can you attach a full stack trace please?

Hi, thanks for reply, this the the log of the progress
[image]

Privilger, 2024-11-05:
> # On host side
> 1. Determine local assets directory by  `$ docker inspect isaac-lab-base`
>    For instance, using following volume mapping
>    host: [...]

Thanks for reply. It works well!

RandomOakForest, 2024-11-22:
It seems you have resolved this issue. Will close it for now. Thanks.

closed 2024-11-22 by RandomOakForest